OpenCPN/OpenCPN #2521: CM93 - Quilting problem and offset issue

rgleason, 2022-01-18:
In tracker, https://www.cruisersforum.com/forums/f134/weather-routing-100060.html#post3557098
Gerhard has identified the commit.
Sherry reports in detail in CF.

filochard, 2022-01-23:
Hi
I don't know if it is the right place for this : 
in flatpak, OpênCPN 5.6.0 doesn't display anymore the cm93 charts
[image]
when OpenCPN 5.2.4 (linux version) can  and when the 5.2.4 flatpak previous version could
[image]

bdbcat, 2022-01-23:
A known problem.  Cm93 does not display with quilting turned OFF.  Will be corrected by next update.

rgleason, 2022-01-23:
Good that you posted filochard, but normally you should post separately for a separate topic. As you know this is not the OP problem. The OP problem is CM93 Offset adjustments do not work.for SaltyPaws, Sherry. He was not able to post in Issues due to bandwidth. Please see the link in the first post.

filochard, 2022-01-24:
Hi 
Sorry not to have created a new issue but this one was about CM93 and I thought my problem was linked to this 

Thanks to bdbcat
I turned quilting ON and got a correct display with OpenCPN 5.6.0
 (I didn't know it was necessary since it was OFF inside OpenCPN 5.2.4 and there was not such a problem... I could not imagine it was so simple to workaround)
Nevertheless it's a bad surprise for a user updating from 5.2.4 to 5.6.0 
Thanks for correcting this in the next update  (I will test it turning quilting OFF...)

rgleason, 2022-01-24:
Filochard, I think you should create a separate Issue about this, to provide a reminder. Thanks very much!

rgleason, 2022-01-24:
Thanks. Filochard.
I think SaltyPaws (Sherry"s) issue is related to 
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:charts:cm93_offsets
but see the link above in the first post please.

bdbcat, 2022-01-24:
Rick..
This is all the same issue.  cm93 does not display with quilting off.
For Sherry, the twist is:  cm93 offsets can only be set with quilting OFF.  It is thus impossible to adjust cm93 offsets, as she posts.

rgleason, 2022-01-24:
Lord this can get "twisted" sometimes! Should I just close this?

bdbcat, 2022-01-25:
Nope.  Leave it open.  It is an open Issue, though the solution is known and will wait until the first maintenance release.
Also duplicated by FS#2814.

CarCode, 2022-01-26:
Don't know why bdbcat don't want to revert his faulty code. It is very easy:
Revert the commit from 22.07.2021 or from 04.08.2021 "Harden cm93 cell quilting logic, avoiding pointer deref faults" in cm93.cpp of the git source code and it works again as before.
As there are also other problems with cm93 he don't want to fix I guess for unknown reasons he don't support cm93 anymore.

rgleason, 2022-03-18:
Wasn't this fixed?  I see Solution Pending in Tracker https://opencpn.org/flyspray/index.php?do=details&task_id=2814
Can I close it?

bdbcat, 2022-03-18:
It will be fully fixed in 5.6.2, starting Beta soon.
Dave

closed 2022-05-04 by nohal