OpenCPN/OpenCPN #2524: Downloading NZ charts fail

deanfourie1, 2022-01-25:
As the title says, attempting to download NZ charts fail. This has been happening for about 3 months now and I've had to revert to manual import.

Just a heads up

transmitterdan, 2022-01-25:
I just tried it an it worked ok for me.  If you click on the "New Zealand NZMariner RNC Charts" in the "Select Catalog..." tab of the chart downloader then click the "Edit" button what is the URL shown?

Also, exactly what happens when you try to download these charts?

transmitterdan, 2022-01-25:
The URL in my NZ chart catalog is: 
https://raw.githubusercontent.com/chartcatalogs/catalogs/master/NZ_RNC_Catalog.xml

mgrouch, 2022-01-25:
> The URL in my NZ chart catalog is:
> https://raw.githubusercontent.com/chartcatalogs/catalogs/master/NZ_RNC_Catalog.xml

Fails for me too. In OpenCPN 5.2.4. 
catalog URL is the same. 

Thanks

mgrouch, 2022-01-25:
Is that certificate issue?

mgrouch, 2022-01-25:
```

wget --no-check-certificate https://www.linz.govt.nz/sites/default/files/media/ntm/bsb/BSB_BASE.zip
wget --no-check-certificate https://www.linz.govt.nz/sites/default/files/media/ntm/bsb/BSB_UPDATE.zip
unzip BSB_BASE.zip
unzip -o BSB_UPDATE.zip
rm  BSB_BASE.zip  BSB_UPDATE.zip

```

nohal, 2022-01-25:
Yes, it is a misconfigured certificate on the server and the workaround is implemented in 5.6.0

rgleason, 2022-03-25:
Can't this be closed now? @deanfourie1

rgleason, 2022-03-25:
Similar issue https://github.com/OpenCPN/OpenCPN/issues/2497

deanfourie1, 2022-03-25:
I'm not sure if it's resolved but I'll close it anyway

closed 2022-03-25 by deanfourie1