OpenCPN/OpenCPN #2525: CPU usage is way higher than qtVlm doing about the same

mgrouch, 2022-01-27:
OpenCPN CPU usage is way higher than qtVlm doing about the same.

I think OpenCPN needs some profiling and performance improvements.
I've added number of instruments to Dashboard Tactics and its seriously hogging CPU on raspberry pi.
(I don't even have that many sensors in that test just IMU and GPS)

Thanks

nohal, 2022-01-27:
`perf` is your friend.

mgrouch, 2022-01-27:
@nohal 

What’s the command to compile OpenCPN 5.6.0 on raspberry pi for
perf. 
or some other (if better) profiler?

thanks

nohal, 2022-01-27:
You don't need anything special, simply run `perf top $(pidof opencpn)` and watch, then act according to what you see.

mgrouch, 2022-01-27:
@nohal 
> You don't need anything special, simply run `perf top $(pidof opencpn)` and watch, then act according to what you see.
```

$perf top $(pidof opencpn)
/usr/bin/perf: line 13: exec: perf_5.10: not found
E: linux-perf-5.10 is not installed.
```

mgrouch, 2022-01-27:
@nohal 

Will it give me a hotspot function name and line number in the source code with CPU% like in other languages' profilers?
I'm not an expert in C++

Thanks

nohal, 2022-01-27:
It will not give you line number, it will give you the hotspot(s). I suggest you just run it, it will probably be faster than asking these questions.

leamas, 2022-01-27:
> /usr/bin/perf: line 13: exec: perf_5.10: not found

Seems that you need to install  linux-perf-5.10 (`sudo apt install linux-perf-5.10`).  You need the linux-perf package which matches your current kernel version as displayed by  `uname -r` after stripping any  -\*  suffix.

CarCode, 2022-01-27:
For sure the CPU usage is completely different between qtVlm and OpenCPN because qtVlm depends on Qt as the name says already while OpenCPN depends on wxWidgets libraries. Qt is much faster.

nohal, 2022-01-27:
@CarCode perhaps you could attach a link to source of this information and explain how exactly is it relevant to this particular situation.

mgrouch, 2022-01-27:
[image]

wxPen::GetWidth is taking ridiculously much CPU

nohal, 2022-01-27:
Not our code. So now you need to find what in our code calls it too often that it makes it eat all your CPU and the fix will probably be to trivially cache the result.
Something like `perf record -a --call-graph dwarf -p $(pidof opencpn) sleep 1` and `perf report` should get you somewhere.

CarCode, 2022-01-27:
> @CarCode perhaps you could attach a link to source of this information and explain how exactly is it relevant to this particular situation.

Simply rewrite OCPN with Qt and you have the answer.

nohal, 2022-01-27:
> > @CarCode perhaps you could attach a link to source of this information and explain how exactly is it relevant to this particular situation.
> 
> Simply rewrite OCPN with Qt and you have the answer.

What answer exactly?

mgrouch, 2022-01-27:
> 

perf report segmentation faults

nohal, 2022-01-27:
Then you will probably have to either fix your system or find the problematic spot the good old way isolating the problematic functionality and reading the relevant code. I do not see any excessive CPU usage and `wxPen::GetWidth` does not seem to be causing any issues on any of my systems, so really can't  help you much more here.

mgrouch, 2022-01-27:
found  these in wxWidgets forums

https://forums.wxwidgets.org/viewtopic.php?t=39983

it talking about antialiasing

```
One thing that makes the GDI+ implementation of wxGraphicsContext slow is its subpixel accuracy (aka anti-aliasing). But for some drawing operations you may not need it, so turning it off might help:

For "normal" drawing operations: wxGraphicsContext::SetAntialiasMode()
For bitmap drawing: wxGraphicsContext::SetInterpolationQuality()

Try experimenting with these values and check if it changes the performance.

```

nohal, 2022-01-27:
I thought you are running Linux on a RPi, GDI+ is Win32 API.

leamas, 2022-01-27:
> perf report segmentation faults

Also when running without plugins?

mgrouch, 2022-01-27:
ok.

setting
`
gtk-xft-antialias=0

`

in gtk config files (for GTK2 and GTK3) fixed issue with wxPen::GetWidth

It's no longer a hotspot.

However overall CPU usage of OpenCPN still much higher than qtVlm

Thanks

nohal, 2022-01-27:
And if you want to do something about it, repeat the excercise.
FWIW CPU usage of qtVLM has been roughly 200% higher than OpenCPN's on my main Mac system during a trivial basemap panning and zooming test I just did, so it probably depends on what exactly you do.

mgrouch, 2022-01-27:
On my system qtVlm updates vessel position once in 3 sec.
OpenCPN seems does it every second. So I guess OpenCPN does 3 times more job.

mgrouch, 2022-01-27:
Is this configurable in OpenCPN? Can I set this refresh rate to 3 seconds?
Or it just updates on all Nmea and SignalK messages?
In that case a flood of messages would slow it down a lot.

nohal, 2022-01-27:
An intense flow of messages will definitely cost computing power. OpenCPN core updating the vessel position is definitely not the place where it will cause any problem though.

mgrouch, 2022-01-27:
I checked "Show FPS" in settings but I can't see where it shows FPS.

nohal, 2022-01-27:
Me neither, probably it is broken. But I don't see any way it could help you in this investigation.

transmitterdan, 2022-01-27:
> I checked "Show FPS" in settings but I can't see where it shows FPS.

This hasn't worked for a long while. Probably needs to be removed. 

But that isn't going to help with your issue. It could be inefficient drawing of the new dashboard items or some other plugin.

I test on Raspberry Pi 4 using lots of dashboard items and high speed NMEA stream with dual chart canvas (RNC and ENC) and it uses about 50% of one core. No real issues that I can see with O itself.

closed 2022-04-12 by mgrouch