OpenCPN/OpenCPN #2528: Serial ports not recognized.

leamas, 2022-01-27:
When trying to create a new connection, available serial ports are not visible in the UI. 

This is a tracker bug for [original issue](https://github.com/flathub/org.opencpn.OpenCPN/issues/2) filed against the  Flatpak packaging. However, after digging into this it seems to be about a test which is applied on all platforms.  

Further work will be done in the original issue, filing this for visibility

leamas, 2022-01-27:
Status after triaging [original issue](https://github.com/flathub/org.opencpn.OpenCPN/issues/2):

  - Things works as expected on kernel 5.12
  - Bug exists on kernels 5.13 and 5.15
  - Work  around is to create a device symlink as described in LINUX-DEVICES.md, the chapter on "Fixed Device Names".
  - There is no documented way to know for sure if there is anything connected to a serial port. What we used may very well have been a bug.

leamas, 2022-01-27:
Possible strategies to handle this:
- Look for asserted DCD or CTS signals. Such are a sure sign a device is connected (the absence does not mean anything). They  can often be easily patched on the hardware level by connecting pins 4 <> 6 and/or  7<-> 8.
- Accept all devices configured by udev i. e., `/sys/bus/usb-serial/devices/*`
- Accept serial devices matching ttyS[0-3]. It is possible to check for irq and port, but this requires root privileges. ttyS[0-3] corresponds to the common PC com1..com4. Assuming Raspberry uses the same ports (not all of them).
- If need be: accept all devices matching a hidden configuration parameter.

I intend to try to fix this, but needs time to consider the situation before coding. All sorts of input welcome!

Hakansv, 2022-01-27:
@leamas  Yes both direct via OCPN and Openplotter "serial".
Shot below from my RPi4 Buster.
The RPi400 bullseye shows a similar list. 
But.. Flatpack I've never touched.
[image]

Hakansv, 2022-01-27:
pi@RPi4:~ $ ls /sys/bus/usb-serial/devices/
ttyUSB0

leamas, 2022-01-27:
Dammit. ttyACM1 is not visible. Assuming it is still connected (if you have time): output from `find /sys -name ttyACM1` ?

Hakansv, 2022-01-27:
pi@RPi4:~ $ find /sys -name ttyACM1
find: ‘/sys/kernel/debug’: Permission denied
/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/tty/ttyACM1
/sys/class/tty/ttyACM1
find: ‘/sys/fs/bpf’: Permission denied

Hakansv, 2022-01-27:
pi@RPi4:~ $ sudo find /sys -name ttyACM1
/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0/tty/ttyACM1
/sys/class/tty/ttyACM1

leamas, 2022-01-27:
OK, that should be enough. Thanks. Perhaps best to keep it simple: Accept known patterns like /ttyACM, ttyUSB, /dev/arduino... and use a comboBox so user can enter other filenames if requires.

Hakansv, 2022-01-27:
What we have now works fine. "Never touch a working system" :)

leamas, 2022-01-27:
But not when you upgrade the kernel, which is inevitable. Better to fix the bug in advance. Actually, it is not in advance, since the first user reports already are upon us.

Hakansv, 2022-01-27:
> when you upgrade the kernel

Excuse my limited Linux know how. How do I do that apart from the apt full-upgrade?

leamas, 2022-01-27:
>  How do I do that apart from the apt full-upgrade?

Wait for next release.

That is, all open-source software are streams. One such stream is the kernel, others are distributions like Debian, Ubuntu, Fedora and what not.

All distributions update their kernel version over time. Almost all are faster in this process that Debian, which movel very slowly. As a consequence, the new kernels > 5.12 has now arrived to Fedora and also Ubuntu (but not Ubuntu LTS). 

The reported errors are from Ubuntu 21.10 and Fedora 33, both of which OS:es which only can be used using Flatpak. Next Debian release is Bookworm, and next Ubuntu LTS release is Jammy which is due in April. All of these OS:es will use kernels > 5.12, so the bug will hit us, Flatpak or native .deb packages all the same. 

We could of course sit and wait until our code breaks. But besides that it is already broken, it's IMHO a bad strategy in a case like this.

Hakansv, 2022-01-27:
OK, good info. 
You're playing Churchill here? Very good you take the foresight role.
And I suppose it worthless to discuss backward compatibility philosophy?

leamas, 2022-01-27:
> And I suppose it worthless to discuss backward compatibility philosophy?

Definitely not. Any fix here should be backward compatible. Actually more than so, since a combobox should handle possible cases which does not work today. The complete test matrix is yet to be defined...

leamas, 2022-01-27:
> You're playing Churchill here?

Which quote do you have in mind here?

Hakansv, 2022-01-27:
Not so very symptomatic but earlier today I run onto [this](https://www.inspirationalstories.com/quotes/winston-churchill-want-of-foresight-unwillingness-to-act-when/) so it was in my head.

leamas, 2022-01-27:
Well, to compare the upcoming kernel version with the rise of Third Reich is perhaps to exaggerate a little ;)

leamas, 2022-01-28:
Kernel bug hunt, sigh. Offending commit : https://github.com/torvalds/linux/commit/f64d74a59c476

leamas, 2022-01-29:
No need to add a wxComboBox : It's already used. Which means that  user which knows the device path can just print it in the box.

rgleason, 2022-01-29:
https://www.cruisersforum.com/forums/f134/dataport-connection-problems-o-v5-6-0-a-259859.html

PR #2537 (2022-01-30): Serial ports detection update (#2528)
As described in #2528, no serial ports are displayed to user when creating a new connection on kernels > 5.12.

In addition to ports presented on kernels <= 5.12 also present ttyS0-ttysS3 and ttyACM? +  ttyUSB?. Also present all ports with hardware DSR or CTS set.  This should fix immediate problems on users running on kernels > 5.12.

For Flatpak, use the sysfs-based port enumeration. This is required since Flatpak does not have  libudev available.

EDIT: Closes: #2528
closed 2022-01-30 by bdbcat