OpenCPN/OpenCPN #2529: OpenCPN hangs when Filter NMEA course data is enabled in connection settings

mgrouch, 2022-01-27:
OpenCPN hangs when Filter NMEA course data is enabled in connection settings

Thanks

mgrouch, 2022-01-27:
Does same type of filter exist for SignalK data?

Hakansv, 2022-01-27:
> re: Does same type of filter exist for SignalK data?

No, you've to do that in SignalK.

Hakansv, 2022-01-27:
> OpenCPN hangs when Filter NMEA course data is enabled in connection settings

All filter I've tried is working on Win10, Rpi and Android.
I think you've to be more specific about how and when and what system.

mgrouch, 2022-01-27:
On raspberry pi4

Hakansv, 2022-01-27:
I've two of these RPi4. What specific filter are you using when crashing? What, if any, filters are working?
I've for example filtered out all headings. No problem.

mgrouch, 2022-01-27:
Open connection settings, 

Check 'Filter NMEA Course and Speed Data' checkbox
put 2 sec into Filter Period (sec).

Restart OpenCPN

Hakansv, 2022-01-27:
That was a another task. Sorry for my misunderstanding.
Well, I tried both 1, 2 and 3 seconds "Filter NMEA Course and Speed Data" on a RPi4 buster without being able to reproduce your issue. Also after a O restart.

mgrouch, 2022-01-27:
Hangs in my setup. Every time. Consistently.

I have SignalK (baro, temp), NMEA from pypilot (IMU heading, rate of turn, etc), GPS from GPSD
coming into OpenCPN.

mgrouch, 2022-01-27:
Locks one CPU at 100%

mgrouch, 2022-01-27:
ok. dashboard tactics plugin must be enabled for it to break.
I see in profiler that dashboard tactics .so library is taking almost 100%.

So it's a combination of filter and dashboard tactics plugin, what breaks OpenCPN

nohal, 2022-01-27:
It would probably be helpful if you discarded 3rd party plugins not being part of OpenCPN before opening issues on this project.

mgrouch, 2022-01-27:
Now I was able to reproduce it without having any third party plugins. Simply with regular Dashboard.
I think this should be re-open.

mgrouch, 2022-01-27:
I was able to reproduce hanging even in OpenCPN safe mode. No plugins.

FilterNMEA_Avg=1
FilterNMEA_Sec=2

in opencpn.conf

This issue should really be re-opened

transmitterdan, 2022-01-28:
I cannot reproduce this.  Can you give some hint about dashboard.  Does it hang if you do not display any dashboard for example?

If you set FilterNMEA_Sec to something more than 2 does it hang?

mgrouch, 2022-01-28:
I do not think it's dashboard related. I had all dashboards disabled.
I think it's related to NMEA feed. I have several from pypilot, from GPSd and from SignalK

mgrouch, 2022-01-28:
And it goes into endless loop. CPU 100% on one core

mgrouch, 2022-01-28:
Does the code handle all errors from system calls?

transmitterdan, 2022-01-28:
> Does the code handle all errors from system calls?

The code makes no system calls within the filter function.

```
void MyFrame::FilterCogSog(void) {
```

It has been tested and has worked well for many years.

mgrouch, 2022-01-28:
How can I help you? 
It's reproducible in my env 100%. with no plugins.
I can run or install anything you ask to get to the bottom of it.
It's not on my boat computer it's in my home set up.

nohal, 2022-01-28:
Keep simplyfying the configuration (or better, start from defaults and keep complicating your configuration until it starts to be problematic) until you find out what exactly is the problem, then depending on what you actually find try to reproduce in a way we can replicate (= replaying recorded traffic, not having the actual sensors you have connected).
You state there is an endless loop in the code, how did you find that and what evidence you have for it exactly?
It certainly is possible to create a self amplifying message storm by configuring the connections to repeat the same messages, are you absolutely sure it is not your case?

transmitterdan, 2022-01-28:
> How can I help you? It's reproducible in my env 100%. with no plugins. I can run or install anything you ask to get to the bottom of it. It's not on my boat computer it's in my home set up.

If you can send a so-called sample file that can be played by SignalK server I can run that and try to reproduce the issue.  If you can record your instruments to a file using SignalK, play that file through the SignalK server and reproduce the issue then I should be able to do the same.

mgrouch, 2022-01-28:
I've created 3 files from 3 sources that feed OpenCPN
```

{
  echo "nc 127.0.0.1 10110 > 10110.txt"
  echo "nc 127.0.0.1 20220 > 20220.txt"
  echo "gpspipe -R 127.0.0.1:2947 > 2947.txt"
} | xargs -L 1 -I CMD -P 3 bash -c CMD

[2947.txt](https://github.com/OpenCPN/OpenCPN/files/7960064/2947.txt)
[10110.txt](https://github.com/OpenCPN/OpenCPN/files/7960065/10110.txt)
[20220.txt](https://github.com/OpenCPN/OpenCPN/files/7960066/20220.txt)

```

mgrouch, 2022-01-28:
I think it's related somehow to GPS. When I unplug GPS I have no issue

nohal, 2022-01-28:
Also post your configuration file or at least the connection section of it, please.

mgrouch, 2022-01-28:
Config is this:

https://raw.githubusercontent.com/bareboat-necessities/lysmarine_gen/buster/install-scripts/6-desktop/files/opencpn.conf

For breaking it use

FilterNMEA_Avg=1

transmitterdan, 2022-01-29:
I am able to reproduce this problem using SignalK.  Investigating...

transmitterdan, 2022-01-29:
Findings so far:

SignalK is sending impossible values for heading (specifically course over ground).  For example: it sends a CoG = -6.2774385622041925e+66 which is nonsense.  There isn't enough error checking on this in the SignalK message handler so OpenCPN sends these nonsense angles to everywhere and that causes near infinite loops in some places.

I am still trying to figure out why SignalK is sending these nonsensical angles.

transmitterdan, 2022-01-29:
Here is the root of the problem. SignalK sometimes sends course angles as NULL meaning the data is not available I presume.  In most SignalK messages OpenCPN checks for null JSON values but in a couple places it does not (e.g. CoG).  This results in unintialized angular values passed on to the rest of O modules such as Dashboard.  I have no idea why SignalK is sending messages with CoG values marked as NULL.

[image]

I will submit a PR to fix this.  There is no good workaround for V5.6.0 except perhaps to find a way that SignalK does not ever send NULL course data.

closed 2022-02-18 by bdbcat