OpenCPN/OpenCPN #2534: In show NMEA Debug window copy and pause buttons are not working

mgrouch, 2022-01-28:
In show NMEA Debug window copy and pause buttons are not working.
They can't be pressed.

raspberry pi4

O 5.6.0

Thanks

Hakansv, 2022-01-28:
On my Pis I first have to exit Options Window. Then it works

mgrouch, 2022-01-28:
Thanks. I couldn't figure it out.

mgrouch, 2022-01-28:
I kind of don't understand why does show window on check box click if that window is not yet usable.

Hakansv, 2022-01-28:
It's still usable since you can see what's flowing or not when you edit connections.
The button functions are as they are on Debian at least, not so on Windows. Easy to learn. Why care?
Not as issue to my mind. Future request?

bcn2, 2022-01-29:
This is valid for macOS as well. In the Linux systems this issue is around since ever.
The debug window does not receive the focus. It will therefore stay behind the options window as well.
IIRC a weird bug in wxWidgets.
As Håkan pointed out: work around is closing the Option dialog.

leamas, 2022-01-29:
Window ownership probably needs a review here. Test sequence
  - Open Options | Connections
  - Select a connection
  - Check the Show NMEA Debug Window
  - Close the options window (windows controls)
  - Re-open Options window.

After this, both the Debug and the Options window are active. So, it is sort of possible.

The NMEA debug window can be moved outside the main OpenCPN window. While not really a bug, I'm not aware of any other OpenCPN windows which are detached in this sense.

This is debugging functions, so we could live with it. That said, there are probably bugs lurking around here.

closed 2022-05-17 by mgrouch