OpenCPN/OpenCPN #2535: track

Ambrozius, 2022-01-28:
The purple track behind the ship is not showing in version 5.6.0

[image]

Hakansv, 2022-01-28:
They are on my Win10 and RPi. What's your system? 
(Silly Q: You have clicked "Apply" or "OK" in the Options windows after selection?)

Hakansv, 2022-01-28:
Hmm..  On my Win10 the tracks are not shown if I switch OpenGl off. 
Would you mind check if you've OpenGl activated and switch it on if not?

Ambrozius, 2022-01-29:
> They are on my Win10 and RPi. What's your system? (Silly Q: You have clicked "Apply" or "OK" in the Options windows after selection?)

the operation system is win 10
I have clicked "Apply" or "OK"
the previous version 5.0 worked fine. i upgraded to 5.6

Ambrozius, 2022-01-29:
> Hmm.. On my Win10 the tracks are not shown if I switch OpenGl off. Would you mind check if you've OpenGl activated and switch it on if not?

when I enable it the track becomes visible. but the program crashes after that

[image]

rgleason, 2022-01-29:
This sounds to me like an opengl driver issue associate with your hardware. I believe this discussion really belongs on Cruiser Forum  for OpenCPN. 

Have you tried running Realtek OpenGL Viewer https://www.realtech-vr.com/home/glview  for checking your hardware?

Please continue this discussion by creating a new thread in Cruiser's Forum OpenCPN where we can help you sort the problem out. This Issues area is not for helping users.  Thanks.

CarCode, 2022-01-30:
It would be nice if people could read what someone writes. This user said it had worked for him with the older version of OCPN but after update to 5.6.0 it does not anymore. Because it doesn't work for other people also without OpenGL it clearly is another bug of 5.6.0.

> This sounds to me like an opengl driver issue associate with your hardware. I believe this discussion really belongs on Cruiser Forum for OpenCPN.
>
> Have you tried running Realtek OpenGL Viewer [...]

rgleason, 2022-01-30:
Well, perhaps you are right Carcode, so thank you @Ambrozius  for posting here.

bdbcat, 2022-01-31:
I'm confused.
Is the track problem observed with GL On, or Off?  Or both?  And only on Windows?

Hakansv, 2022-01-31:
@bdbcat 
On two of my Win10 and OCPN 5.6.0 (Incl. present master..) 
For both raster and vector and single or multi canvas:
If I switch OpenGL off -> OK; the AIS tracks are not shown.
If I switch OpenGL on -> OK; the AIS tracks are back again.

Ambrozius, 2022-01-31:
> @bdbcat On two of my Win10 and OCPN 5.6.0 (Incl. present master..) For both raster and vector and single or multi canvas: If I switch OpenGL off -> OK; the AIS tracks are not shown. If I switch OpenGL on -> OK; the AIS tracks are back again.

that matches my problem.
only my pc crashes with open GL on.
in version 5.0 everything works fine.

Ambrozius, 2022-01-31:
in version 5.0 and open GL on the program is slow.
the old PC :
[image]
the new PC i dont know

Ambrozius, 2022-01-31:
only the old pc crast with open GL on

bdbcat, 2022-01-31:
Thanks for the info.
Confirm, please:  We are talking _**only**_ about AIS tracks.
Not ownship tracks.
Correct?

Ambrozius, 2022-01-31:
Correct, only AIS tracks

Hakansv, 2022-01-31:
@bdbcat 
Ownship tracks are working both with and without OpenGl. Correct.

Hakansv, 2022-01-31:
@Ambrozius 
My view.. The issue here is the AIS tracks malfunction without OpenGl.
Your PC's OpenGl function may be you need another driver but it's not a discussion to solve here. There are a lot earlier discussions and knowledge on the Forum, please.

rgleason, 2022-01-31:
Thanks. That is what I said at the beginning.  I left a suggestion to download and run ViewGL but he did not choose to do that. 

 Can this be closed now?

bdbcat, 2022-02-01:
Nope.
AIS tracks without GL still not working.  That is the OP's issue.

```
If I switch OpenGL off -> OK; the AIS tracks are not shown.
If I switch OpenGL on -> OK; the AIS tracks are back again.

```
All the rest is not relevant to this issue.

transmitterdan, 2022-02-01:
BTW, I can confirm this issue on Windows 10 machine.

bdbcat, 2022-02-01:
This has been broken for a long time.  I guess not many users do not use OpenGL.
Anyway, fixed by a9db82f37e3cd9cbedadee2655c3dbb28bc010f2, in github master now.

closed 2022-02-01 by bdbcat

rgleason, 2022-02-01:
Thanks, Dave.  I use openGL a lot, but never noticed the problem.

Hakansv, 2022-02-01:
@bdbcat 
For info nothing else:
Fix confirmed AND the tracks in OpenGl mode are more clear and bright than before. Not bad!
Thanks /Håkan