OpenCPN/OpenCPN #2536: helicopter icon

Ambrozius, 2022-01-28:
the helicopter icon is too small compared to the ship icon

[image]

Hakansv, 2022-01-28:
The helicopter icon is programmed that in way so I wouldn't call this a bug or an issue.
If you've dissenting opinion about the size would you mind raise a future request instead. Possible after a Forum discussion?
Please close, thanks.
Håkan

rgleason, 2022-01-29:
@Ambrozius   The Original Poster OP needs to close this please. You can discuss on Cruiser's Forum and post it in tracker after that discussion. Thanks.

Ambrozius, 2022-01-30:
> @Ambrozius The Original Poster OP needs to close this please. You can discuss on Cruiser's Forum and post it in tracker after that discussion. Thanks.

how do i close it ?

rgleason, 2022-01-30:
I think you will have a close button at the bottom.  Ah. Sorry. I see I am able to close it too.

closed 2022-01-30 by rgleason