OpenCPN/OpenCPN #2538: Linux: Options: Connections: Cannot read info

leamas, 2022-01-29:
As of attached screenshot: Important data, notably the device, are truncated due to too narrow columns.

Stretching the window makes no difference. I see the same problem both on a native Fedora build and on Flatpak.

[image]

leamas, 2022-01-29:
hm... seems to be about the device. If using a short device name like _foo_ everything looks fine. However, on linux there are no such short names (unless the /dev/ prefix is removed)

leamas, 2022-01-29:
No. something more complicated. If I define a new device it's displayed like this
[image]

However, after restarting OpenCPN it's displayed like this:

[image]

bdbcat, 2022-04-07:
Close?

leamas, 2022-04-07:
No, bug is still present on master. Could be closed after merging #2586

PR #2586 (2022-04-28): Connection parameters: unreadable data fix (#2538)
The culprit is that wxStaticText for some reason does not respect the font when calculating the minimum width, at least on GTK. Issue is present also in recent GTK 3.1.5.

Work around using a new bold label class.

Tested on Windows, Flatpak, Focal, and MacOS.

Closes: #2538
closed 2022-04-28 by nohal