OpenCPN/OpenCPN #2540: Route and Mark Manager ETA bug

rgleason, 2022-01-29:
Please see https://www.cruisersforum.com/forums/f134/bug-in-the-route-and-mark-manager-259704.html#post3551888

rgleason, 2022-03-19:
<html><body>
<div style="margin:20px; margin-top:5px; ">
<div class="smallfont" style="margin-bottom:2px">Quote:</div>

``In order for us to make this work now, we have to save > export > delete all > import again after each change  to make the changes come through for the entire route (we found this out by mere luck).
`

```
...This is where things start to get strange. The time stamps showing in the ETA column remain unchanged except for the 1st row in the table. Editing the ‘Plan Speed’ also messes up the ETA times stamps succeeding the first row.
See screenshots for clarification (here I tried to change the speed, left Input Departure time unchanged): 
```

bdbcat, 2022-03-19:
This is fixed in O561 Beta, coming soon now...
Leave it open for now.

bdbcat, 2022-04-07:
Close?

closed 2022-04-07 by rgleason