OpenCPN/OpenCPN #2542: Raspberry Pi 3 - openplotter v2 flatpak install fails

jongough, 2022-01-30:
Following the instructions to install the flatpak version I issued:
```
flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/pi/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.
```
Then tried and got the error below:
```
 flatpak install --user  https://flathub.org/repo/appstream/org.opencpn.OpenCPN.flatpakref

Note that the directories 

'/var/lib/flatpak/exports/share'
'/home/pi/.local/share/flatpak/exports/share'

are not in the search path set by the XDG_DATA_DIRS environment variable, so
applications installed by Flatpak may not appear on your desktop until the
session is restarted.
```

leamas, 2022-01-30:
What does `uname -m`say?

leamas, 2022-01-30:
> Following the instructions to install the flatpak version [...]

Are you referring to this link: https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak ?

jongough, 2022-01-30:
armv7l

jongough, 2022-01-30:
I followed this:
[https://opencpn.org/OpenCPN/info/downloadopencpn.html](https://opencpn.org/OpenCPN/info/downloadopencpn.html)

leamas, 2022-01-30:
> armv7l

armv7l is dubbed armhf in Debian parlance. We don't support Flatpak on armhf, only aarch64

> I followed this: [...]

Hm... I don't really see anything about Flatpak there when I follow the Raspberry link. When looking into the Flatpak link I find this sentence in the first paragraph

>  It is also the the supported way to install OpenCPN on RPi-4 and other high-end aarch64 system. 

However, you have a armv7l/armhf system, not a aarch64 one so it just doesn't work.

Technically, all is as it is intended to be here. Since you made a mistake there might be room to improve the documentation, but it's not immediately clear to me if or how.

jongough, 2022-01-30:
Read `This packaging system is supported on all major Linux distributions.` . I was having some difficulties with 5.6.0 on the Pi where some of the depths were showing and some not. I wanted a different installation so tried flatpak. The statement suggests that it should work. A standard user would assume it would work. I saw nothing to say it wouldn't. I installed flatpak OK, it was just the OCPN part that failed.

The message `error: No such ref 'app/org.opencpn.OpenCPN/arm/stable' in remote opencpn-origin` is not helpful and, to me at least, flatpack is a 'black box', so I just followed the bouncing ball and it failed. Your statement only says  that it is the supported way to install on Pi4's, but does not say anything about not working on <Pi 4's.

I did not make a mistake, I followed the documentation. Change the documentation to be clearer, or make it available on armhf systems. Don't expect users to understand any of this.

leamas, 2022-01-30:
Well, Rick owns the user docs. 

The actual statements are clear. One could perhaps add change that sentence to something like  `This packaging system is supported on all major Linux distributions. One exception in 32-bit (armhf) Raspbian system.`, dunno. But that is all there is to it, I guess.

We have in other contexts agreed that the typical armhf system don't have enough resources, in particular RAM, to successfully run Flatpak.

rgleason, 2022-01-31:
Note there is a Raspberry Pie 4 Page now https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:rpi4

That is empty and can be edited/improved.

Also I have fixed the top of the Raspberry Pie 2/3 Page and it can be improved.
Many users and developers have edit rights and I can add editors.

rgleason, 2022-01-31:
Also added to RPI 2/3 " the typical armhf system don't have enough resources, in particular RAM, to successfully run Flatpak."
Anything else needed?

leamas, 2022-01-31:
The very title of that page might be wrong. As has been clarified here, there is no such thing as Raspberry Pie 4 installation-wise.  

What we have is aarch64-based platforms like vanilla Debian, Raspbian 64-bit beta, Arch and what not.  And beasts like Jon's which is running a 32-bit armv7l/armhf OS, usually Raspian. These have nothing at all in common, and having a common place for these is IMHO plain wrong

leamas, 2022-01-31:
> Also added to RPI 2/3 " the typical armhf system don't have enough resources, in particular RAM, to 

Why not be clear instead: we don't support Flatpak on armhf/armv7l. Full stop.

rgleason, 2022-01-31:
I added  (not supported on RPI 2/3)  here
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak

rgleason, 2022-01-31:
I suggested using Flatpak installation on RPI 4 and provided a link

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:rpi4

rgleason, 2022-01-31:
Is this paragraph still applicable?  I don't think so.

https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:raspberrypi_rpi2#installing_raspbian

If you rewrite or correct, I will amend it.

leamas, 2022-01-31:
> I suggested using Flatpak installation on RPI 4 and provided a link

Which is plain wrong. Again. There is no such thing as RPI 4 .

rgleason, 2022-01-31:
Or you can log in and edit.  Perhaps that is better.   
Do you want that page deleted now?

leamas, 2022-01-31:
Just remove what you have done, make it blank again. No time now.

rgleason, 2022-01-31:
If I remove or leave that blank, then what you wrote at the top of 
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:raspberrypi_rpi2

Is wrong too.

leamas, 2022-01-31:
Perhaps, but facing a blank side is better than facing plain wrong info. I can handle that later, just leave it in place for now.

rgleason, 2022-01-31:
Ok, I removed the ref to flatpak for RPI4,  do you want RPI page removed now?

You just have to edit the page and remove all text and then hit enter to save the empty page.

leamas, 2022-01-31:
Edited, but cannot  parse  the documentation on page links. Someone needs to fox that page.

rgleason, 2022-01-31:
Have fixed it. Thanks.

You take the URL for the page and remove the first part for "internal links"`

[[opencpn:opencpn_user_manual:getting_started:opencpn_installation:raspberrypi_rpi2|Raspberry Pi 2/3]]`

For external links it is 

`[[https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:raspberrypi_rpi2| Raspberry Pi 2/3]]`

nohal, 2024-02-15:
Every reasonably usable ARM system (Equivalent to RPi 3B or better) not past it's end of life nowadays is `aarch64`. Closing.

closed 2024-02-15 by nohal