OpenCPN/OpenCPN #2552: Fault when following routes - RPI

andyrbarrow, 2022-02-07:
I'm using an RPI4 - 4GB with OpenPlotter. Data input to OCPN is SignalK. I've also tried using an NMEA0183 TCP input with the same result. Using a CAN/RS422 hat for input to the PI.

When I follow a route, after a minute or so I get a fault. If I don't clear it immediately, OCPN freezes and I have to restart the program. If I don't follow a route, I have no problems. 

Here is a screenshot:
[image]

leamas, 2022-02-07:
Just a drive-by comment: The dialog seems to have a backtrace available. Is there anything meaningful there?

andyrbarrow, 2022-02-15:
I finally got a stack trace (and finally got within cell coverage). 
ASSERT INFO:
../src/common/datetimefmt.cpp(776): assert "Assert failure" failed in Format(): unknown format specifier

BACKTRACE:
[1] wxDateTime::Format(wxString const&, wxDateTime::TimeZone const&) const
[2] ConsoleCanvas::UpdateRouteData()
[3] ConsoleCanvas::RefreshConsoleData()
[4] MyFrame::OnFrameTimer1(wxTimerEvent&)
[5] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[6] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[7] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[8] wxEvtHandler::TryHereOnly(wxEvent&)
[9] wxEvtHandler::ProcessEventLocally(wxEvent&)
[10] wxEvtHandler::ProcessEvent(wxEvent&)
[11] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[12] wxTimerImpl::SendEvent()

leamas, 2022-02-15:
Some digging shows that this probably is about these lines in src/concanv.cpp around line 400:
```
        if (speed > 0.) {
          // Show date, e.g. Feb 15, if TTG > 24 h
          seta = tttg_sec > SECONDS_PER_DAY ? eta.Format(_T("%e/%m %H:%M"))
                                            : eta.Format(_T("%H:%M"));
        } else {
```

Looking in the wxWidgets sources  I just cannot make things fit. The file *src/common/datetimefmt.cpp* is there and hasn't been changed since 2019.  However, line 779 is not in the vicinity of `wxDateTime::Format()`. 

As I read current code's `wxDateTime::Format()`  all format specifiers used are supported. Furthermore, I find no assert or wxAssert at all. Sigh

However, here is a comment around line 1166:

        case 'e':           // day of a month (1-31) (GNU extension)

referring to %e.  Perhaps meaningful, perhaps not.  But anyway, this does not seem to be the source file actually used on Android (?).  Not from master branch?  No idea.

bdbcat, 2022-02-16:
@leamas...
IIUC, the OP's issue  is linux/gtk on rPI4.  Not Android.  So likely wx3.0.
In that source, I see that 'e' is only used when parsing a format.  It is not present when formatting a string Format(..).
So, on some platforms, and some wx versions, 'e' is accepted in Format().  On others, apparently not so.
Test it.  Create and activate a route, with some GPS data active.  Move the last point dynamically so that the TTG becomes greater than one day.  Either the output in console switches, or it asserts.

nohal, 2023-02-27:
Are we happy here?

leamas, 2023-03-17:
@andyrbarrow : ping? ^

leamas, 2023-03-19:
Yes. Closing.

closed 2023-03-19 by leamas