OpenCPN/OpenCPN #2562: [Android] Warning when installing 5.2.8 on Samsung Galaxy Tab S4

JNavas2, 2022-02-12:
Warning that device is _not_ supported.
Full specifications [here](https://www.gsmarena.com/samsung_galaxy_tab_s4_10_5-9262.php).

bdbcat, 2022-02-12:
What warning text, where printed?

JNavas2, 2022-02-12:
I should have done a screenshot, but the app is now installed, and I'll be testing it in a race tomorrow (Saturday), so further troubleshooting will have to wait until afterward. 

I think the warming was something like: `This app may not be optimized for your device`

Once the race is behind me, I'll uninstall, and then reinstall, and see if I get the same warning.

In the meantime, see [Large screens app quality  |  Android Developers](https://developer.android.com/docs/quality-guidelines/large-screens-app-quality)

JNavas2, 2022-02-13:
The boat skipper now has the tablet, so it may be some time before I can try to recreate this issue.