OpenCPN/OpenCPN #2563: [Android] Display corruption when scrolling ENC charts

JNavas2, 2022-02-13:
Display of NOAA ENC charts is corrupted with scrolling, can be recovered with zoom in and out.
[YouTube video](https://youtu.be/n8VuszPrtxg)
OpenCPN Android 5.2.8
Device model: Google Pixel 5a (barbet)
Android version: 12 ( SQ1A.220205.002 )

JNavas2, 2022-02-14:
These are the loaded NOAA ENC chart files: [SFBA.zip](https://github.com/OpenCPN/OpenCPN/files/8057555/SFBA.zip)

JNavas2, 2023-08-21:
SEVERITY: HIGH/CRITICAL
Version 5.6.2 on Google Pixel 6a
In trying to build a route from Pittsburg CA to Delta Yacht Club on Tule Island
with all the necessary NOAA ENC charts loaded
the display constantly jumps back to the unhelpful basemap almost every time I scroll.
I can sometimes recover by zooming in and out with  + and - 
but it's too painful to be usable.

bdbcat, 2023-08-22:
You might try updating to the current v5.8.4, available on the Playstore.

JNavas2, 2023-08-22:
Duplicate of what Issue?
The Issue number should always be provided when closing as a duplicate.

JNavas2, 2023-08-22:
@rgleason 
Duplicate of which Issue?
The _Issue number_ should always be provided when closing as a duplicate.
Discussions are not Issues, and I'm not looking for help. I'm reporting a _serious_ bug.
If Discussion #3403 is what you meant, then _please reopen this Issue_.

@bdbcat 
Thank you, but this ENC display issue persists in 5.8.4
See #3403