OpenCPN/OpenCPN #2564: [Windows] Route & Mark Manager, Waypoints Usability Issues

JNavas2, 2022-02-15:
1. Route & Mark Manager position and size are not saved between program executions. They should be, at least as an option.
2. Waypoints tab column widths are not saved between program executions. They should be, at least as an option. 
3. Waypoints tab does not display position, making it painful to check positions. Ideally, columns would be selectable, as in Windows Explorer Details View, and position should be shown by default.
4. Waypoints Properties window size is not retained, always comes up too small to be useful.
5. Waypoint Icons are too difficult to change: the long list must be painfully navigated each time.
6. Waypoints Scale is too easy to change accidentally.
7. Ideally, it would be possible to select multiple Waypoints, and for all of them set an Icon and/or Scale.
8. Waypoints Description tab is redundant.

OpenCPN 5.6.0 on Windows 8.1 Pro x64

[image]

Hakansv, 2022-02-15:
@JNavas2 
An issue(bug) here is meant as a report when something in the code does not work. 
Your list is more of ideas and personal thoughts to change behavior. Personally I can agree of some and others not.
It's very good you've suggestions to make a better program but this is not the forum to achieve that. 
For OpenCPN there is a [tracker ](https://opencpn.org/flyspray/index.php?project=6&do=index&switch=1). There you can suggest changes and maybe a note on the [OpenCPN forum ](https://www.cruisersforum.com/forums/f134/) to get others to support it by there votes.

So please close this issue and instead use the tools for future suggestions.

JNavas2, 2022-02-15:
@Hakansv 
As a developer myself (personal and professional), I do appreciate the challenges of FOSS.
I sincerely appreciate the existence of this software, and I don't mean to sound ungrateful or harsh.

That said, and with respect, "issue" is not necessarily synonymous with (or limited to) "bug"
(not working as designed). It's not just a matter of functionality, it's also a matter of usability.

These are neither "feature requests" for new functionality or suggestions for improvements.
They are basic usability Issues (to me at least).

As a user I need to manage waypoints, and the current implementation has serious usability issues.
I have wasted way too much time trying to do basic tasks that should be simple to accomplish.

Take, for example, verifying waypoint positions, an essential task.
Unless I'm missing something, there's simply no good way to do that in OpenCPN.
I had to Export them and load them in another program in order to display and check them.
I should not have to do that.

Hope that helps,
John

p.s. 

1. I'm in the process of evaluating and reviewing OpenCPN for Android 
(running on a Samsung Galaxy Tab S4 at the helm of a racing sailboat
and on my Google Pixel 5a smartphone).
I'm only using the Windows version because I was advised to use it
because things like waypoint management were so difficult on the Android version.

2. In my expert, professional opinion, "working as designed" is too often an excuse
for poor design, and poor design is just as much an "issue" as poor implementation.
The more important standards are "suitable to the task" and "meets expectations."

3. On the Forum, which also has usability issues, I was blocked for "suspicious
activity" with no explanation or recourse. I don't have time for that kind of foolishness.

4. I apologize for not having a public email address.
(It's on my public website, but that's not obvious.)
I've now made it public in case you wish to have a more appropriate offline discussion.

bdbcat, 2022-02-16:
John...
OpenCPN has been in wide release for many, many years.  We have many tens-of-thousand of users, worldwide, on a large variety of platforms.  So as you can imagine, we receive plenty of feedback, from both new users, as well as experienced sailors and developers.

What we have here is an issue of nomenclature.

In OpenCPN, we consider an "issue" to be a bug.  That is, the function as promised simply does not work, or leads to instability or another system failure.
Questions about "usability", particularly from new users, are considered to be "Feature Requests", in OCPN-speak.  Generally, these points are handled at relatively low priority, as compared to actual "issues' and other functional enhancements.  But they are all considered on their own merits, as limited volunteer resources permit.

Thanks for your comments.
Dave

JNavas2, 2022-02-16:
> What we have here is an issue of nomenclature.
> In OpenCPN, we consider an "issue" to be a bug. That is, the function as promised simply does not work, or leads to instability or another system [...]

Thank you, but I think what we actually have here is a difference in perspectives and priorities.
I see these things as Issues, or I would not have taken the tine to document and post them here.
Since they will not be taken seriously, I won't invest any more time on it.
It's a shame, because I do like the ENC display (when it's not corrupted),
and I was willing to invest both time and expertise to help make it work properly.
John

closed 2022-02-16 by JNavas2

bdbcat, 2022-02-16:
John...
Sorry you feel that way.  We would have appreciated your comments and observations.
However, In an environment of limited volunteer developer resources, as most FOSS projects are, we must formulate and stick to a set of hard priorities, or else we would never reach a release point for a stable application.  Feature-creep is a very real factor in our thinking.

Here is an example of the thinking, for one platform only:
"Is the Android RouteProps dialog hard to use on portrait mode phones? Yes it is"

"Which is a better use of limited resources, best serving the user?"
(a) Improve the RouteProp dialog for one sub-platform, or 
(b) Implement support for for Android Scoped Storage in A11/12
(c) Or should we require both, taking as long as it takes before a release?
Answer is clear:  (b).
Basic functionality wins out every time.

Anyway, you are welcome back if you like.
Dave

JNavas2, 2022-02-16:
Dave,
1. You've already received a good deal of feedback from me.
2. Unfortunately, your priorities do not work for our use case. Functionality that's not usable has no value to us.
3. Showstoppers from trying OpenCPN Android in a race last weekend:
   1. A critical waypoint was moved accidentally with no way to recover rendering it useless.
   2. Despite several tries I was unable to initiate Navigate to Waypoint. In a race there is no time to keep screwing around.
   3. The Chart Panel was accidentally popped up covering much of the display, so the driver abandoned it.
4. For us, as it is now, OpenCPN Android is at best a nice ENC display (when not corrupted), but is not usable for waypoint or route navigation.
5. We have no. interest in the desktop version other than as a (poor) work-around for Android issues.
6. OpenCPN Android seems more suitable for casual boating or cruising than for racing.
7. Since our use case does not match your priorities, and since our issues are deprecated to low-priority feature requests, I think it unlikely that I will be back.

Thank you nonetheless,
John

nohal, 2022-02-16:
As some of your issues seem to be pretty simple work items, maybe, being a programmer yourself, a more productive approach would be to actually implement them instead of getting pissed that people did not drop anything they were doing and did it for you immediately.
And, as a programmer, both professionaly and in my free time on OpenCPN, please, pretty please, try not to open issues, bug reports or feature requests with multiple work items in them, it is a PITA to handle.

JNavas2, 2022-02-16:
Thank you, but I'm neither pissed nor interested in OpenCPN politics, and I've now Closed all my Issues.

rgleason, 2022-02-16:
You could purchase Expedition, that might be the safest route. Don't know if it is on android however.

JNavas2, 2022-02-17:
@rgleason Thank you for the suggestion.
Unfortunately, Expedition only runs on Windows, and does not work well on a tablet.
I won't go into more detail because I don't think this is a good place to discuss alternatives.

rgleason, 2022-02-17:
Probably an appropriate decision. I just want to say that all of us share your interest in improving OpenCPN for Android, but as Dave said earlier, we are somewhat restricted by programming resources, although there has been remarkable change and continuous improvements over many years. Good Luck on your search.

JNavas2, 2022-02-17:
@rgleason Thank you again.
If you wish to continue the conversation, send me an email (through my Profile here)
or suggest a good open forum (other than OpenCPN - Cruisers & Sailing Forums).