OpenCPN/OpenCPN #2567: logbookkonni crashes O ion pi4

mgrouch, 2022-02-17:
logbookkonni crashes O ion pi4 (buster) at startup

```

ASSERT INFO:
/usr/include/wx-3.0/wx/arrstr.h(178): assert "nIndex < m_nCount" failed in Item(): wxArrayString: index out of bounds

BACKTRACE:
[1] logbookkonni_pi::LoadConfig()
[2] logbookkonni_pi::Init()
[3] PlugInManager::LoadPlugInDirectory(wxString const&, bool, bool)
[4] PlugInManager::LoadAllPlugIns(bool, bool)
[5] MyFrame::OnInitTimer(wxTimerEvent&)
[6] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[7] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[8] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[9] wxEvtHandler::TryHereOnly(wxEvent&)
[10] wxEvtHandler::ProcessEventLocally(wxEvent&)
[11] wxEvtHandler::ProcessEvent(wxEvent&)
[12] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[13] wxTimerImpl::SendEvent()

```

leamas, 2022-02-17:
This seems to be about LogbookKonni. As such, it should be filed against the plugin fork at  https://github.com/rgleason/LogbookKonni_pi.

rgleason, 2022-03-01:
Closing,. Mike added to logbook.

rgleason, 2022-03-01:
Alec this is about Buster which is no longer supported. Therefore shouldn't it be dropped.

closed 2022-03-01 by rgleason