OpenCPN/OpenCPN #2568: 5.6.1 Current master, Crash on loading alpha plugin list

RooieDirk, 2022-02-17:
System: Ubuntu 20.04 
I get a crash if I try to load the alpha plugin list. It looks like a problem with an invalid ocpn-plugin.xml file.
If I remove the last part of the file, O is happy again.
The problem part:
_<plugin version="1">
    <name> Sean Depagnier Alpha Plugins meta-url </name>
    <version> 1.0 </version>
    <summary> meta-url to Sean's Alpha Plugins by rgleason </summary>
    <meta-url> https://raw.githubusercontent.com/rgleason/plugins/rg-alpha/ocpn-plugins.xml </meta-url>
  </plugin>_
I'm not familiar with this part of the code, but it seems as if we need a sanity check on the file before using it.

leamas, 2022-02-20:
Some notes on this:
  - The sanity checks is against the xsd files, but the xsd format does not include the meta-url element. So Alpha catalogs does indeed fail.
  - At a glance, we have this which "looks" wrong to me;  should the meta-url be part of the plugin element?
```
 <plugin version="1">
    <name> o-charts </name>
[...]
    <tarball-url> https://dl.cloudsmith.io/public/david-register/ocpn-plugins-unstable/raw/names/o-charts-android-arm64-16-tarball/versions/0.9.0.6.0.4505b90/o-charts_pi-0.9.0.6-0_android-arm64-16.tar.gz </tarball-url>
    <meta-url> https://raw.githubusercontent.com/rgleason/plugins/rg-alpha/ocpn-plugins.xml </meta-url>
  </plugin>
```
  - The meta-url thing is a kludge we IMHO should get rid of. Developers **do** have a need to publish their own catalogs/plugins fopr test purposes. However, publishing a separate catalog and use that as "Custom URL" is a much simpler and cleaner solution. Requires documentation to be useful, though.

leamas, 2022-02-20:
Filed https://github.com/OpenCPN/plugins/pull/609, which documents a Custom URL usecase which can be used instead of a meta-url to publish plugins for testing.

leamas, 2022-02-22:
https://github.com/OpenCPN/plugins/pull/611 which is supposed to fix this is merged.

@RooieDirk : can you verify that the actual problem here is fixed?

RooieDirk, 2022-02-23:
Fixed

closed 2022-02-23 by RooieDirk