OpenCPN/OpenCPN #2569: New charts list dialog - vertical slider

Hakansv, 2022-02-19:
Dave..
Win10.
The nice new chart folder dialog in options->Charts needs an "update" after having used the "Add directory" function.
In my example the folder list was close to filling up the available space so no vertical slide bar was needed.
After having added a new directory it was not visible in the list. Hmm.. So I did it once more. No change. 
Now for a test I resized the options window and then the slide bar was created and both new directories was shown.

I also recognized the directory window need a "redraw" to possible create the vertical slider when entering "Chart Files" the first time. All directories was not shown from start. See first picture.  After my dialog resize it was clear there are more. See second picture the "ENC_DK" folder. That one was not added now.

So, a redraw after the initial view of the charts list and after the "Add directory"  function may do?

The test was initially to test possible long and UTF8 directory names and it's handled very nice. See pict two. Great!

Before:
[image]

After resize:
[image]

bdbcat, 2022-02-19:
Hakan...
Thanks for the test.
This issue corrected by 3d21d28ab29269e9d4b4c36c14988f330485330e.  I needed to "workaround" what seems to be a persistent wxWidgets bug with wxScrolledWindow.

We will see the biggest win with this new chart list dialog is on Android phone devices, which are very narrow in size.

Hakansv, 2022-02-19:
Fix confirmed.
I really look forward to see this on the A. I do agree it's needed.
Thanks.

closed 2022-02-19 by Hakansv