OpenCPN/OpenCPN #2570: Current master hangs at startup

RooieDirk, 2022-02-19:
On linux (ubuntu 20-04) The current master hangs at startup.
Last bit of logfile:
`17:44:58.001 MESSAGE glChartCanvas.cpp:1505 OpenGL-> Minimum cartographic line width:  1.0
17:44:58.001 MESSAGE glChartCanvas.cpp:1508 OpenGL-> Minimum symbol line width:  1.0
17:44:58.003 MESSAGE gshhs.cpp:1338 Loading World Chart Q=0 in 0 ms.
17:44:58.003 MESSAGE gshhs.cpp:164 Background world map loaded from GSHHS datafiles found in: /usr/local/share/opencpn/gshhs/
17:44:58.213 MESSAGE chartdb.cpp:214 ChartDB Cache policy:  Max open chart limit is 20.
17:44:58.213 MESSAGE chartdbs.cpp:154 Loading chart db version: V018
17:44:58.213 MESSAGE chartdbs.cpp:1208 Chartdb: Chart directory list follows
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #0: /home/dirk/Charts/Hyd_Encdata_dgas30_RD
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #1: /home/dirk/Charts/WorldMapENC
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #2: /home/dirk/.opencpn/Waddenzee_HD_Relief
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #3: /home/dirk/.opencpn/Waddenzee_Laseraltimetrie_Diepte
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #4: /home/dirk/Charts/Norway_R
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #5: /home/dirk/Charts-test/o-charts
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #6: /home/dirk/Charts/ENC/FR_INLAND
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #7: /home/dirk/Charts/VMH
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #8: /home/dirk/Charts/ENCi
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #9: /home/dirk/Charts/CM93
17:44:58.213 MESSAGE chartdbs.cpp:1227   Chart directory #10: /home/dirk/Charts/US/RNC
17:44:58.214 MESSAGE chartdbs.cpp:1227   Chart directory #11: /home/dirk/.opencpn/Waddenzee_Infrarood
17:44:58.214 MESSAGE chartdbs.cpp:1227   Chart directory #12: /home/dirk/Charts/ANWB
17:44:58.214 MESSAGE chartdbs.cpp:1227   Chart directory #13: /home/dirk/Charts/US/ENC
17:44:58.214 MESSAGE chartdbs.cpp:1227   Chart directory #14: /home/dirk/.opencpn/Waddenzee_Achtergrond
17:44:58.496 MESSAGE chart1.cpp:2475 GPS Watchdog Timeout is: 6 sec.
17:44:58.508 MESSAGE chart1.cpp:2562 OpenCPN Initialized in 6336 ms.
17:44:59.025 MESSAGE glChartCanvas.cpp:732 BuildFBO 3
17:44:59.026 MESSAGE glChartCanvas.cpp:1217 OpenGL Framebuffer OK, size = 1278
17:44:59.049 MESSAGE chart1.cpp:6723 OnInitTimer...0`

If started from commandline:
`LSA lib pcm_dsnoop.c:641:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_dmix.c:1089:(snd_pcm_dmix_open) unable to open slave
17:39:09: Debug: Device: 0: HDA ATI HDMI: 0 (hw:0,3)
17:39:09: Debug: Device: 1: HDA ATI HDMI: 1 (hw:0,7)
17:39:09: Debug: Device: 2: HD-Audio Generic: ALC892 Analog (hw:1,0)
17:39:09: Debug: Device: 3: HD-Audio Generic: ALC892 Digital (hw:1,1)
17:39:09: Debug: Device: 4: HD-Audio Generic: ALC892 Alt Analog (hw:1,2)
17:39:09: Debug: Device: 5: hdmi
17:39:09: Debug: Device: 6: pulse
17:39:09: Debug: Device: 7: default
17:39:09: Debug: Device: 0: HDA ATI HDMI: 0 (hw:0,3)
17:39:09: Debug: Device: 1: HDA ATI HDMI: 1 (hw:0,7)
17:39:09: Debug: Device: 2: HD-Audio Generic: ALC892 Analog (hw:1,0)
17:39:09: Debug: Device: 3: HD-Audio Generic: ALC892 Digital (hw:1,1)
17:39:09: Debug: Device: 4: HD-Audio Generic: ALC892 Alt Analog (hw:1,2)
17:39:09: Debug: Device: 5: hdmi
17:39:09: Debug: Device: 6: pulse
17:39:09: Debug: Device: 7: default
17:39:09: Debug: Device: 0: HDA ATI HDMI: 0 (hw:0,3)
17:39:09: Debug: Device: 1: HDA ATI HDMI: 1 (hw:0,7)
17:39:09: Debug: Device: 2: HD-Audio Generic: ALC892 Analog (hw:1,0)
17:39:09: Debug: Device: 3: HD-Audio Generic: ALC892 Digital (hw:1,1)
17:39:09: Debug: Device: 4: HD-Audio Generic: ALC892 Alt Analog (hw:1,2)
17:39:09: Debug: Device: 5: hdmi
17:39:09: Debug: Device: 6: pulse
17:39:09: Debug: Device: 7: default
17:39:09: Debug: Device: 0: HDA ATI HDMI: 0 (hw:0,3)
17:39:09: Debug: Device: 1: HDA ATI HDMI: 1 (hw:0,7)
17:39:09: Debug: Device: 2: HD-Audio Generic: ALC892 Analog (hw:1,0)
17:39:09: Debug: Device: 3: HD-Audio Generic: ALC892 Digital (hw:1,1)
17:39:09: Debug: Device: 4: HD-Audio Generic: ALC892 Alt Analog (hw:1,2)
17:39:09: Debug: Device: 5: hdmi
17:39:09: Debug: Device: 6: pulse
17:39:09: Debug: Device: 7: default
Killed
`

RooieDirk, 2022-02-19:
Hm getting unreadable. Same info as attached files
[opencpn.zip](https://github.com/OpenCPN/OpenCPN/files/8102887/opencpn.zip)

leamas, 2022-02-19:
You might want to read up on how to format posts here: [Github Markup](https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax)

The question is if this indeed is about the master branch and not your test environment. Do this by  doing  ` git checkout Release_5.6.0` and then rebuild OpenCPN.

If it fails the same way, this is about your test setup (since the released version don't hang).

If not, something has happened between Release_5.6.0 and the tip of current master causing the hang. The do a ` git bisect` to determine the commit which introduced this problem. If this is new to you, there are multiple howtos out there. One can be found [here](https://www.metaltoad.com/blog/beginners-guide-git-bisect-process-elimination), there are many others. Google is your friend.

transmitterdan, 2022-02-19:
One important thing it does not long after this log message is to potentially rebuild the chart database.  If you are building this, you could sprinkle some more log messages through the 0th leg of the startup switch statement.  That might narrow down where it gets stuck.  Possibly your charts or the existing database are unique and exposes the problem.

RooieDirk, 2022-02-20:
Found the commit.
```
commit 62811df4f53b58497153966649f4d246b48823b6
Author: dsr <bdbcat@yahoo.com>
Date:   Wed Feb 16 22:18:25 2022 -0500

    Add test code to detect duplicate waypoint GUID

 src/routeman.cpp | 27 +++++++++++++++++++++++++++
 1 file changed, 27 insertions(+)
```
The problem is that I have some large layers, and this peace of code takes a very long time to check them out. Long time as in multiple hours.
I'm not sure if the fact that the layers don't have an GUI does make it worse.

bdbcat, 2022-02-20:
OK, I'll take that out.
It is some very crude coded added to investigate FS#2831, since replaced by a JS script doing the same thing.  Never meant for production....

bdbcat, 2022-02-20:
Done, removed in master branch

RooieDirk, 2022-02-20:
Fixed

closed 2022-02-20 by RooieDirk