OpenCPN/OpenCPN #2583: Debian scaling issue

Hakansv, 2022-03-12:
On RPi4 Debian.
O build from source present master.
I can run O from clean, new, config file.  I can add connections and a chart but..:
As soon I change for example the ship scale factor from 0 to -1 O crashes. This scenario is repeated a couple of times: 
The primitive gdb debug says:
```
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0xb63227e4 in wxFormatString::AsWChar() () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
(gdb) kill
Kill the program being debugged? (y or n) y
[Inferior 1 (process 19635) killed]
(gdb) q
pi@RPi4:~/builds/OpenCPN/build $ uname -a
Linux RPi4 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
```
If relevant or not I've also seen "
`(opencpn:27649): Gtk-WARNING **: 11:48:51.999: gtk_widget_size_allocate(): attempt to allocate widget with width -5 and height 15`

Håkan

transmitterdan, 2022-03-12:
The GTK warnings are not relevant.

I can't reproduce on any other RPi OS configuration. Can you build a debug version and post the gdb back trace (bt)?

Hakansv, 2022-03-12:
Dan..
This is RPi4 Buster (32)
No extra cmake despite "Debug".
Here I just run with the same config file that caused the seg. fault. That's after my scaling changes. (The same config runs fine on OpenPlotter OCPN: "Ubuntu 18.04", now uninstalled. 
Home brewed O has worked fine before but I have not yet traced back in commits.
I'm not so used to gdb but here's a try:

```
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0xb63227e4 in wxFormatString::AsWChar() () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
(gdb) backtrace
#0  0xb63227e4 in wxFormatString::AsWChar() () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#1  0x00215510 in wxFormatString::operator wchar_t const*() const (this=0xbeffd220)
    at /usr/include/wx-3.0/wx/strvararg.h:225
#2  0x0025f950 in wxString::Printf<int>(wxFormatString const&, int) (this=0xbeffd178, f1=..., a1=141)
    at /usr/include/wx-3.0/wx/string.h:2302
#3  0x0035b7dc in ChartCanvas::SetCursorStatus(double, double)
    (this=0xdde070, cursor_lat=57.672284845595776, cursor_lon=11.741395202688718)
    at /home/pi/builds/OpenCPN/src/chcanv.cpp:4155
#4  0x0035b5e4 in ChartCanvas::OnCursorTrackTimerEvent(wxTimerEvent&) (this=0xdde070, event=...)
    at /home/pi/builds/OpenCPN/src/chcanv.cpp:4129
#5  0xb625c784 in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const ()
    at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#6  0xb63cb7a8 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
    at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#7  0xb63cb8e4 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) ()
    at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#8  0xb63cbd94 in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#9  0xb63cbe30 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#10 0xb63cbefc in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#11 0xb63cbb64 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#12 0xb633a404 in wxTimerImpl::SendEvent() () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#13 0xb65da444 in  () at /usr/lib/arm-linux-gnueabihf/libwx_gtk3u_core-3.0.so.0
#14 0xb53df778 in  () at /usr/lib/arm-linux-gnueabihf/libglib-2.0.so.0
(gdb) 
```

Hakansv, 2022-03-13:
@bdbcat 
This seems to be about the change of degree symbol from "°" to \u00B0?  (commit dd50dbd2)
```
-    s.Printf(wxString("%03dÂ°  ", wxConvUTF8), (int)brg);
+    s.Printf("%03d\u00B0  ", (int)brg);
 
```
Isn't the hex method compatible with wx 3.0?  or do I have trouble with my libwx_baseu-3.0.so?
```
chcanv.cpp:4155

else
    s.Printf("%03d\u00B0  ", (int)brg);
```

Hakansv, 2022-03-13:
Dave..
For a test on my RPi Buster, wx 3.0,  I took out the commit mentioned above, Then O is running as normal.

bdbcat, 2022-03-14:
Strange.  No one else is seeing this.
Is your machine set to SE language?  Could that be the difference?

leamas, 2022-03-14:
hm...

> O build from source present master.

```
0x0035b7dc in ChartCanvas::SetCursorStatus(double, double)
    (this=0xdde070, cursor_lat=57.672284845595776, cursor_lon=11.741395202688718)
    at /home/pi/builds/OpenCPN/src/chcanv.cpp:4155
```

However, when I look into current master, line 4155 is not involved at all in https://github.com/OpenCPN/OpenCPN/commit/dd50dbd2383af8c0d2a9775fd6ba3143af34bcdf. On the other hand, lines around 4132 are:

```
  DistanceBearingMercator(cursor_lat, cursor_lon, gLat, gLon, &brg, &dist);
  if (g_bShowMag)
    s.Printf("%03d\u00B0(M)  ", (int)gFrame->GetMag(brg));
  else
    s.Printf("%03d\u00B0  ", (int)brg);

```

These lines are the only ones affected by that commit in chcanv.cpp, Seems to hold both for Release_5.6.0 and current master at 987a27db079

Something fishy is going on here. Some kind of build problem?

leamas, 2022-03-14:
> Line 4155 is probably not valid anymore since new commits arrived afterwards.

If you checkout the latest release i . e., `git checkout Release_5.6.0`and build using that: is the bug still around?

If so, it's most likely about your build environment. If not, you should be able to make a `git bisect` to  find the offending commit.

leamas, 2022-03-14:
> I may have to check my use of dependencies on the Buster?

First, make sure you don't have a packaged installation in place if you are using a local build (`apt remove opencpn`). And, while there are shortcuts, the safe route is to always do `make install` before making any tests.

Besides that, as long as it builds it should be fine (assuming you are using the "official" way to install deps as of the manual.).

Hakansv, 2022-03-18:
Now I've tried with another image where I know built has succeeded before.
So no old package installed. "Official" dependencies used what I'm aware of.
The wxWidgets cmake found would be suitable? 
Could you Linux gurus find any obvious cause?
 
Using present master:

```
gdb openpcn
.......
.......
Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.
0xb630e7e4 in wxFormatString::AsWChar() () from /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
(gdb) backtrace
#0  0xb630e7e4 in wxFormatString::AsWChar() () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#1  0x00207ae0 in ChartCanvas::SetCursorStatus(double, double) ()
#2  0x0020c7e4 in ChartCanvas::OnCursorTrackTimerEvent(wxTimerEvent&) ()
#3  0xb6248784 in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const ()
    at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#4  0xb63b77a8 in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
    at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#5  0xb63b78e4 in wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*) () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#6  0xb63b7d94 in wxEvtHandler::TryHereOnly(wxEvent&) () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#7  0xb63b7e30 in wxEvtHandler::ProcessEventLocally(wxEvent&) () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#8  0xb63b7efc in wxEvtHandler::ProcessEvent(wxEvent&) () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#9  0xb63b7b64 in wxEvtHandler::SafelyProcessEvent(wxEvent&) () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#10 0xb6326404 in wxTimerImpl::SendEvent() () at /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0
#11 0xb65c910c in  () at /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0
#12 0xb5de8778 in  () at /usr/lib/arm-linux-gnueabihf/libglib-2.0.so.0
(gdb) kill
Kill the program being debugged? (y or n) y
[Inferior 1 (process 16676) killed]
(gdb) q

pi@RPi4:~/builds/OpenCPN/build $ uname -a
Linux RPi4 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

pi@RPi4:~/builds/OpenCPN/build $ cmake -DCMAKE_BUILD_TYPE=Release -DOCPN_TARGET_TUPLE:STRING="ubuntu-armhf;18.04;armhf" ../
-- cmake version: 3.16.3
-- *** Host Build Architecture is armhf
-- Setting C++11 standard via cmake standard mechanisma
-- Default compiler options:
-- CMAKE_CXX_FLAGS: 
-- CMAKE_CXX_FLAGS_DEBUG: -g
-- CMAKE_CXX_FLAGS_MINSIZEREL: -Os -DNDEBUG
-- CMAKE_CXX_FLAGS_RELEASE: -O3 -DNDEBUG
-- CMAKE_CXX_FLAGS_RELWITHDEBINFO: -O2 -g -DNDEBUG
-- CMAKE_INSTALL_LIBDIR1 
-- CMAKE_INSTALL_LIBDIR2 lib
-- CMAKE_INSTALL_LIBDIR3  lib
-- Building version 5.6.16
-- *** Staging to build opencpn 5.6.16 ***
-- *** Package will NOT include documentation ***
-- Checking for module 'gtk+-3.0'
--   No package 'gtk+-3.0' found
-- Could NOT find GTK3 (missing: GTK3_INCLUDE_DIRS GTK3_LIBRARIES) 
CMake Warning (dev) in CMakeLists.txt:
  A logical block opening on the line

    /home/pi/builds/OpenCPN/CMakeLists.txt:586 (if)

  closes on the line

    /home/pi/builds/OpenCPN/CMakeLists.txt:588 (endif)

  with mis-matching arguments.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL....
--     GL Lib: /usr/lib/arm-linux-gnueabihf/libGL.so/usr/lib/arm-linux-gnueabihf/libGLU.so
[... 63 lines cut ...]
-- chartdldr_pi: /usr/lib/arm-linux-gnueabihf/libarchive.so
-- *** Added plugin: /home/pi/builds/OpenCPN/plugins/chartdldr_pi
-- 
-- *** Staging to build dashboard_pi ***
-- *** Added plugin: /home/pi/builds/OpenCPN/plugins/dashboard_pi
-- *** Staging to build grib_pi ***
-- Could NOT find JPEG (missing: JPEG_LIBRARY JPEG_INCLUDE_DIR) 
-- Could NOT find Jasper (missing: JASPER_LIBRARIES JASPER_INCLUDE_DIR JPEG_LIBRARIES) 
-- Using bundled jasper library...
-- *** Added plugin: /home/pi/builds/OpenCPN/plugins/grib_pi
-- Build type: Release
-- *** Staging to build wmm_pi ***
-- Found OpenGL...
--     Lib: /usr/lib/arm-linux-gnueabihf/libGL.so/usr/lib/arm-linux-gnueabihf/libGLU.so
--     Include: /usr/include
-- Found OpenGL...
--     Lib: /usr/lib/arm-linux-gnueabihf/libGL.so/usr/lib/arm-linux-gnueabihf/libGLU.so
--     Include: /usr/include
-- *** Added plugin: /home/pi/builds/OpenCPN/plugins/wmm_pi
-- *** Package will include tide and current data ***
-- Git status: clean
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/builds/OpenCPN/build
```

leamas, 2022-03-18:
> Thread 1 "opencpn" received signal SIGSEGV, Segmentation fault.

```
 0x00207ae0 in ChartCanvas::SetCursorStatus(double, double) ()
#2  0x0020c7e4 in ChartCanvas::OnCursorTrackTimerEvent(wxTimerEvent&) ()
```

So this happens in  ChartCanvas::SetCursorStatus, probably when attempting to to wxFormat, wxPrintf or something similar using the dreaded printf-style templates.

You need line numbers. Unfortunately, your current image seems to be built without debugging symbols i. e., without the -g compile flags. You can check using the  command `file`, for example `file /usr/local/bin/opencpn` if this is the file you are running. For me, this shows:

    $ file  /usr/local/bin/opencpn
    /usr/local/bin/opencpn: ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=d50af394f16a0c0e198b03750ab016c26a9989aa, for GNU/Linux 3.2.0, with debug_info, not stripped

If your file is stripped, you need to rebuild it with -g. I **think** setting CMAKE_BUILD_TYPE=DEBUG should do this, not sure though -- the handling of build flags in the main OpenCPN is a complete mess.

Another useful command is ldd, it shows the runtime linkage. You can use it to reveal the wxWidgets libraries actually used. For me, it goes
```
$ ldd   /usr/local/bin/opencpn | grep wx
	libwx_gtk3u_gl-3.0.so.0 => /lib64/libwx_gtk3u_gl-3.0.so.0 (0x00007fb432db5000)
	libwx_baseu_net-3.0.so.0 => /lib64/libwx_baseu_net-3.0.so.0 (0x00007fb432d71000)
	libwx_baseu_xml-3.0.so.0 => /lib64/libwx_baseu_xml-3.0.so.0 (0x00007fb432d5b000)
	libwx_gtk3u_html-3.0.so.0 => /lib64/libwx_gtk3u_html-3.0.so.0 (0x00007fb432c88000)
	libwx_gtk3u_adv-3.0.so.0 => /lib64/libwx_gtk3u_adv-3.0.so.0 (0x00007fb432ab9000)
	libwx_gtk3u_aui-3.0.so.0 => /lib64/libwx_gtk3u_aui-3.0.so.0 (0x00007fb432a20000)
	libwx_gtk3u_core-3.0.so.0 => /lib64/libwx_gtk3u_core-3.0.so.0 (0x00007fb43241e000)
	libwx_baseu-3.0.so.0 => /lib64/libwx_baseu-3.0.so.0 (0x00007fb4321ee000)
	libwx_gtk3u_webview-3.0.so.0 => /lib64/libwx_gtk3u_webview-3.0.so.0 (0x00007fb4321c8000)
```
You could do the same on your host to make sure the correct wxWidgets libraries are used

leamas, 2022-03-18:
Looking closer at SetCursorStatus I see several wxString::Printf() and wxString::Format() which most likely are the culprit. Personally, I would redefine wxString s to std::stringstream and use the overloaded, typesafe  << operator instead of the ::Format and ::Printf. Using standard, compile-time checked stuff is basically always better (and, FWIW, also faster). 

The need to use types like wxString is basically  just in the GUI interface. This is also what the wxWidgets docs says.

But then again, that's me.

Hakansv, 2022-03-18:
re: 

> wxString::Printf() and wxString::Format() which most likely are the culprit.

If so what has happened the last weeks? This have been use for years.  
And it don't explain the strange canvas [position mentioned before](https://github.com/OpenCPN/OpenCPN/issues/2583#issuecomment-1066891490)

Anyone else have successfully built on Rasbian(Debian), Buster using present master?

transmitterdan, 2022-03-18:
i build on Buster. No issues.
[image]

leamas, 2022-03-18:
If I understand this correctly, the crash occurs when running OpenCPN, not when building it. Right?

Also, without going into details I assume bugs could depend on the input data used here (?)  For example, is `g_defaultBoatSpeed`  sane? Not pretending that this is the issue, just that this code is not really robust.

Checks include resetting opencpn.conf, and/or building from the Release_5.6.0 tag. 

>  cmake -DCMAKE_BUILD_TYPE=Release

If you instead use cmake -DCMAKE_BUILD_TYPE=DEBUG you should get the debugging symbols, notably line numbers, which makes the backtrace more useful.

And, please submit
  - The exact command used to run opencpn
  - The output from `which opencpn`
  - The output from `ldd %OPENCPN% | wx`, where %OPENCPN% is the actual opencpn program you run.
  - Likewise, the output from `file %OPENCPN%`

leamas, 2022-03-18:
Also: have you tried the `-safe-mode` option which disables all plugins?

Hakansv, 2022-03-18:
Thanks for your views on this.
Here's what I get so I suppose despite having used the "new official" dependencies something look strange, or?
```
pi@RPi4:~ $ ldd   /usr/local/bin/opencpn | grep wx
	libwx_gtk2u_gl-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_gl-3.0.so.0 (0xb6bea000)
	libwx_baseu_net-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_baseu_net-3.0.so.0 (0xb6ba2000)
	libwx_baseu_xml-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_baseu_xml-3.0.so.0 (0xb6b85000)
	libwx_gtk2u_html-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_html-3.0.so.0 (0xb6ad6000)
	libwx_gtk2u_adv-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_adv-3.0.so.0 (0xb6967000)
	libwx_gtk2u_aui-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_aui-3.0.so.0 (0xb68e5000)
	libwx_gtk2u_core-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0 (0xb643b000)
	libwx_baseu-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0 (0xb61fb000)
pi@RPi4:~ $ 
```

leamas, 2022-03-18:
NO, nothing strange here what I can see. It looks perfectly sane.

But that was just one  item out of four. In particular:
  - Bug still here with an empty opencpn.conf. Or?
  - Bug still here suing -safe-mode. Or?
  - Bug still here if you build from the Release_5.6.0 tag. Or?

transmitterdan, 2022-03-19:
FYI:
```
ldd /usr/local/bin/opencpn | grep wx
	libwx_gtk3u_gl-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk3u_gl-3.0.so.0 (0xb6b66000)
	libwx_baseu_net-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_baseu_net-3.0.so.0 (0xb6b1e000)
	libwx_baseu_xml-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_baseu_xml-3.0.so.0 (0xb6b01000)
	libwx_gtk3u_html-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk3u_html-3.0.so.0 (0xb6a52000)
	libwx_gtk3u_adv-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk3u_adv-3.0.so.0 (0xb68e7000)
	libwx_gtk3u_aui-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk3u_aui-3.0.so.0 (0xb6867000)
	libwx_gtk3u_core-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk3u_core-3.0.so.0 (0xb63cb000)
	libwx_baseu-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0 (0xb618b000)
	libwx_gtk3u_webview-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk3u_webview-3.0.so.0 (0xb6163000)
```

bdbcat, 2022-03-19:
Hakan...
You may like to try a build from latest github master.
I did make some changes to "degree" symbol rendering pattern, for Android.
6eaeec1dbb7cc6916f754c026dc5b03783c3cdc0

Dave

Hakansv, 2022-03-19:
@transmitterdan 
Thanks for the ldd compare.
There's a significant difference to my system.  yours: libwx_gtk3u_gl-3.0.so  mine: libwx_gtk2u_gl-3.0.so.0 
My system is 32 Rasbian:  Linux RPi4 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
I think you've something else?

@leamas 

    Bug still here with an empty opencpn.conf. Or?  >>> Yes
    Bug still here suing -safe-mode. Or?  >>>> Yes
    Bug still here if you build from the Release_5.6.0 tag. Or?  >>>> Will try again . Builds now

@bdbcat 
No difference using latest master. 
And I don't think this error is really relevant. It's more a symptom of some basic wx failure. Not less since the screen position is a mess positioning O collapsed and outside the screen edge.

transmitterdan, 2022-03-19:
@Hakansv
Mine seems a little older:
```
$ uname -a
Linux raspberrypi4 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux
```

leamas, 2022-03-19:
> My system is 32 Rasbian: Linux RPi4 5.10.103-v7l+ https://github.com/OpenCPN/OpenCPN/pull/1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

Just to handle the timezones: That version is the kernel version. More needed is the debian version as displayed by for example `lsb_release -a` or `cat /etc/debian_version`

leamas, 2022-03-19:
Seems indeed that the \U00B0 might be part of this: https://stackoverflow.com/questions/6796157.

transmitterdan, 2022-03-19:
Is it possible this is a locale specific problem with wxWidgets?  I think the code takes different paths interpreting these characters depending on locale.

Hakansv, 2022-03-20:
lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

Could all this bee about missing dependencies for Buster?  (On another RPi4 I've Rasbian 11 and it's working fine)
Dan has like: libwx_gtk3u_gl-3.0.so.0 =>    and I like: libwx_gtk2u_gl-3.0.so.0 =>
I now used the "official" method to install. When I did it again there are some obstacles. See the result of the second step below. 
I may see if the "old" dependencies would fit better here?
```
pi@RPi4:~/builds/OpenCPN $ sudo mk-build-deps -i -r ci/control
dh_testdir
dh_testroot
dh_prep
dh_testdir
dh_testroot
dh_install
dh_install: warning: Compatibility levels before 10 are deprecated (level 9 in use)
dh_installdocs
dh_installdocs: warning: Compatibility levels before 10 are deprecated (level 9 in use)
dh_installchangelogs
dh_compress
dh_compress: warning: Compatibility levels before 10 are deprecated (level 9 in use)
dh_fixperms
dh_installdeb
dh_installdeb: warning: Compatibility levels before 10 are deprecated (level 9 in use)
dh_gencontrol
dh_md5sums
dh_builddeb
dpkg-deb: building package 'opencpn-build-deps' in '../opencpn-build-deps_1.0_all.deb'.

The package has been created.
Attention, the package has been created in the current directory,
not in ".." as indicated by the message above!
Selecting previously unselected package opencpn-build-deps.
(Reading database ... 200611 files and directories currently installed.)
Preparing to unpack opencpn-build-deps_1.0_all.deb ...
Unpacking opencpn-build-deps (1.0) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies...Starting pkgProblemResolver with broken count: 1
Starting 2 pkgProblemResolver with broken count: 1
Investigating (0) opencpn-build-deps:armhf < 1.0 @iU mK Nb Ib >
Broken opencpn-build-deps:armhf Depends on base-files:armhf < 10.3+rpi1+deb10u11 @ii mK > (>= 11)
  Removing opencpn-build-deps:armhf because I can't find base-files:armhf
Broken opencpn-build-deps:armhf Depends on libwxgtk-webview3.0-dev:armhf < none @un H >
  Removing opencpn-build-deps:armhf because I can't find libwxgtk-webview3.0-dev:armhf
  Or group remove for opencpn-build-deps:armhf
Done
 Done
Starting pkgProblemResolver with broken count: 0
Starting 2 pkgProblemResolver with broken count: 0
Done
The following packages will be REMOVED:
  opencpn-build-deps
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 10,2 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 200614 files and directories currently installed.)
Removing opencpn-build-deps (1.0) ...
mk-build-deps: Unable to install opencpn-build-deps at /usr/bin/mk-build-deps line 416.
mk-build-deps: Unable to install all build-dep packages

```

leamas, 2022-03-20:
> Release: 10

In this context this basically means that you are on Debian 10 a.  k. a. Buster

>Dan has like: libwx_gtk3u_gl-3.0.so.0 => and I like: libwx_gtk2u_gl-3.0.so.0 =>

Timezones makes me make assumptions; in this case, that transmitterdan is on Bullseye i. e., Debian 11.

Buster is the only release supporting both wxwidgets/gtk2 and wxwidgets/gtk3, so it's possible to build OpenCPN on Buster using wxWidgets/gtk3. This would give you the same deps as transmitterdan, but OTOH make you incompatible with all plugins -- the build chains only build for wxWidgets/gtk2 on buster/armhf.  Assuming that this is not really what you want.

Furthermore 

Looking further into the `\U00B0` mess this is probably a compiler issue rather than about wxWidgets. You might want to try  changing lines like 
 
    s.Printf("%03d\u00B0  ", (int)brg);

to 

     s.Printf(u8"%03d\u00B0  ", (int)brg);

according to the SO question [above](https://github.com/OpenCPN/OpenCPN/issues/2583#issuecomment-1072995994). Another solution is the one Dave has applied to master. The difference is about sticking to standard C++ libraries vs using wxWidgets.

Hakansv, 2022-03-20:
> Looking further into the \U00B0 mess this is probably a compiler issue rather than about wxWidgets. You might want to try changing lines like

If so why is present master working on my Rasbian 11 and all other system?

leamas, 2022-03-20:
The difference could be the compiler. No time to check now,  but Debian 10 is quite old. Add to that that the expression "some text\u00B0" is actually underspecified according to the stackoverflow above, and there is a chance that the handling of such an expression has been changed over time.

Anyway, using `u8"some text\u00B0"` is as I understand it well defined C++11 and could be used to disambiguate expressions like  `"some text\u00B0"`.

Just 5 öre, and not tested...

leamas, 2022-03-20:
I have been able to reproduce this. It is  as I see it a wxWidgets bug. Facts in case, on Fedora 34, gcc 11.2.1:

Test program is named foo.cpp and compiled using

    $ make CPPFLAGS="$(wx-config --cppflags)" LDFLAGS="$(wx-config --libs)" foo

Following program segfaults ( changing `u8"Degree` to `"Degree`makes no difference)
```
#include <wx/string.h>
#include <iostream>

int main (int argc, char** argv) {
  wxString s;
  s.Printf(u8"Degree sign: \u00B0  ");
  std::cout << s;
}
```

Following program works:
```
#include <wx/string.h>
#include <iostream>

int main (int argc, char** argv) {
  std::cout << "Degree sign:\u00B0 \n";
}
```

I should have paid more attention to you first remark about dd50dbd2. Sorry for that.  It happened partly because I was too quick (again!), partly because I was mislead by not being able construct your history.

As I see it, the lesson learned her should be to avoid wxString if possible (here, it certainly is). The alternative is the quick fix done in 6eaeec1db, available in current master at c8292172a

Lessons learnedAvoid wxString if possible, and in code like this it's definitely possible

leamas, 2022-03-20:
@transmitterdan:

Is this a wxWidgets bug we shoudl report, or something which follows the docs? Have problems to wrap my head around the strings and UTF docs...

bdbcat, 2022-03-20:
Alec...
You understand, of course, that converting OCPN from wxString to std::string is a Herculean task, however laudable.
We should:
1.  Use std::string in new code, where sensible.
2. Fix compiler variations as they occur, as with https://github.com/OpenCPN/OpenCPN/commit/6eaeec1dbb7cc6916f754c026dc5b03783c3cdc0

transmitterdan, 2022-03-21:
> 

@bdbcat :

I'll take it up and see if the bug is compiler or library related.

transmitterdan, 2022-03-21:
Just so I don't go to wxWidgets with something "stupid" please help me understand this a little since I am not a wchar expert.  As I understand, before C++20 u8"string" has type char* . Why do we use char * instead of wchar_t*? To wit, the test program:
```
#include <wx/string.h>
#include <iostream>

int main (int argc, char** argv) {
  wxString s;
  s.Printf(u8"Degree sign: \u00B0 \n");
}
```
does seg fault as @leamas shows, however:
```
#include <wx/string.h>
#include <iostream>

int main (int argc, char** argv) {
  wxString s;
  s.Printf(L"Degree sign: \u00B0 \n");
}
```
does not and it correctly encodes the degree symbol.  Is u8 really the correct string specifier?

Hakansv, 2022-03-21:
A test on a clean Rasbian image 
The official installation of dependencies gives the same kind of error as described [here](https://github.com/OpenCPN/OpenCPN/issues/2583#issuecomment-1073200269):

Broken opencpn-build-deps:armhf Depends on base-files:armhf < 10.3+rpi1+deb10u11 @ii mK > (>= 11)
  Removing opencpn-build-deps:armhf because I can't find base-files:armhf
..........
Removing opencpn-build-deps (1.0) ...
mk-build-deps: Unable to install opencpn-build-deps at /usr/bin/mk-build-deps line 416.
mk-build-deps: Unable to install all build-dep packages

leamas, 2022-03-21:
> A test on a clean Rasbian image

WTF?  What does `dpkg -l | grep base-files` say?

leamas, 2022-03-21:
@transmitterdan:  After your input (thanks!)  I revisited https://docs.wxwidgets.org/3.0/overview_unicode.html. It actually sorts things out (?)

First, the wxWidgets builds on both Debian/ubuntu and Fedora are all using wxUSE_UNICODE_WCHAR == 1 i. e., they are indeed using wide strings internally. Also, https://docs.wxwidgets.org/3.0/overview_unicode.html gives an example using `L"Salut \u00E0 toi!"`. Somehow, it seems you are on the right track here: using wide characters in strings with non-ascii characters is the correct path.

However, when I compile the test program on Fedora using `L"Degree sign: \u00B0 \n"` it does not segfault, but it doesn't print anything. In the debugger I can see that while the internal string looks sane, ToStdString() returns an empty string.

And when I compile the test program on Debian/Ubuntu linkage fails with unresolved symbols. This is insane, but a fact· Seems to be about wide characters. Will look into it, but no ENOTIME rigth now.

leamas, 2022-03-21:
@bdbcat:

Of course we cannot start replacing all wxString with std::string. While laudable, life is just too short for that. Agreed.

But we could certainly consider replacing it in situations where they are actually problematic, and still not that complicated to get rid of. IMHO, chcanv::SetCursorStatus is such a case.

The statement that we should use std::string where  possible  in new code is important; I can nothing but concur.

All this said, we need to come to some conclusion in the @transmitterdan  subthread first: what is *really* going on here?

Hakansv, 2022-03-21:
Now It's working on a Buster..........
> WTF? What does dpkg -l | grep base-files say?

I think that's so to say overdue, see below.  But I'll check later. On another PC now.
For Info:..

On the Rasbian Buster from a SSD card with new image:
As said I tried the "official" dependency installer. It failed.
Then I used the (old) setup of "compiling Linux".
There I'd had to manually install wxGTK3....  and due to a make error also libusb-1.0
After that it builds and installs fine.
I've added charts and used ship scaling up and down, where it before crashed.
All data using degree symbols as of latest master works fine.
o-charts from PM is not available for the default build using cmake ../ so I'll build me a home brewed tarball for that. I'll be back

BTW: Is there any way I can clean my other system from all (strange) dependencies I may have introduced and make a functional build also there. (I'm just lazy and would spare me to set this system up from scratch with all other applications?)

Thanks
Håkan

Hakansv, 2022-03-21:
More info..
Also o-charts_pi using SG-lock dongle works fine.
All scaling and degree symbols works still fine.

leamas, 2022-03-21:
This is really as expected. This issue has morphed into two others:

  - There are no working compilation  instructions for Raspbian/Buster.  Although not officially supported (official builds uses ubuntu/armhf), we night need to handle it.
  - We need to sort out the handling of utf literal strings. Current fix works, but there are other aspects.

transmitterdan, 2022-03-21:
In several places the auto-pasting macro _T(), aka wxT(), is not used.  This seems at least inconsistent and quite possibly a bug.  Is it possible this is the root of the problem?  One side effect of wxT() is that it pastes on the L to a string literal.  IOW, in a Unicode build:
```wxString(_T("Some string \U000B \n"))```
after preprocessing becomes:
```wxString(L"Some string \U000B \n")```

leamas, 2022-03-21:
> In several places the auto-pasting macro _T(), aka wxT(), is not used. This seems at least inconsistent and quite possibly a bug

No: https://docs.wxwidgets.org/3.0/group__group__funcmacro__string.html#ga7dfc2888539861afe6c4337ef315472b

Hakansv, 2022-03-21:
> This is really as expected. This issue has morphed into two others:> 
>     There are no working compilation instructions for Raspbian/Buster. Although not officially supported (official builds uses ubuntu/armhf), we night need to handle it.

If it was that clear we may have have had a long discussion about absent bugs. Sorry I didn't understand any possible info about that.

Now I'll try to clean my old system from dependencies creating the build. Maybe remove all about wx and redo it?
(I've a influxdb + grafana there I don't want to create from scratch if possible :))
Anyhow.. this issue would not be relevant any more. 
Thanks for all input.

closed 2022-03-21 by Hakansv