OpenCPN/OpenCPN #2587: Random AIS 2bells.wav sound on 5.6.0 on macOS

jeronimo, 2022-03-18:
[image]
[image]
[image]

Everything seems to be disabled but it still alerts randomly...

bdbcat, 2022-03-18:
Are there AIS targets that might alert, if sounds were enabled?

jeronimo, 2022-03-18:
> Are there AIS targets that might alert, if sounds were enabled?

None when it made a sound recently.

rgleason, 2022-03-18:
Options > User Interface  
is Play Ships Bells checked?

For watchdog you can select the provided sound files for various alerts so that you can determine which alarm it is, or if it is Watchdog at all.

nohal, 2022-03-18:
https://www.clockguy.com/SiteRelated/SiteReferencePages/ShipsBellsExplained.html

closed 2022-03-18 by nohal