OpenCPN/OpenCPN #2591: Android armhf circleci local build not generating opencpn executable

jongough, 2022-03-20:
To try and debug a problem in a plugin on Android armhf it is necessary to build a debug version of OCPN. OCPN armhf is being built using circleci and there is a job, build-android-armhf, using circleci/android:api-28-ndk docker image. When attempting to build OCPN master branch with this using this command:
```
circleci local execute --job build-android-armhf
```
The build completes but does not appear to generate an OCPN executable. The following is a listing of what is in 'build' before the docker image is stopped:
```
+ mv build-android-hf build
+ ls -la build
total 117980
drwxr-xr-x 12 circleci circleci      4096 Mar 20 22:04 .
drwxr-xr-x 23 circleci circleci      4096 Mar 20 22:08 ..
-rw-r--r--  1 circleci circleci     32535 Mar 20 21:54 CMakeCache.txt
drwxr-xr-x 15 circleci circleci      4096 Mar 20 22:08 CMakeFiles
-rw-r--r--  1 circleci circleci      3571 Mar 20 21:54 CPackConfig.cmake
-rw-r--r--  1 circleci circleci      3883 Mar 20 21:54 CPackSourceConfig.cmake
-rw-r--r--  1 circleci circleci    146982 Mar 20 21:54 Makefile
-rwxr-xr-x  1 circleci circleci    144052 Mar 20 21:54 NSIS.template.in
-rw-r--r--  1 circleci circleci     17473 Mar 20 21:54 cmake_install.cmake
drwxr-xr-x  3 circleci circleci      4096 Mar 20 21:54 fpu_neon
-rw-r--r--  1 circleci circleci       164 Mar 20 21:54 gitcheck.sh
drwxr-xr-x  2 circleci circleci      4096 Mar 20 21:54 include
drwxr-xr-x  2 circleci circleci      4096 Mar 20 21:56 lib
-rw-r--r--  1 circleci circleci   1567128 Mar 20 21:56 libS57ENC.a
-rw-r--r--  1 circleci circleci    126178 Mar 20 21:54 libSYMBOLS.a
-rwxr-xr-x  1 circleci circleci 118689824 Mar 20 22:04 libgorp.so
drwxr-xr-x 20 circleci circleci      4096 Mar 20 21:54 libs
drwxr-xr-x  3 circleci circleci      4096 Mar 20 21:54 mavx2
drwxr-xr-x  3 circleci circleci      4096 Mar 20 21:54 msse
drwxr-xr-x  3 circleci circleci      4096 Mar 20 21:54 msse2
drwxr-xr-x  3 circleci circleci      4096 Mar 20 21:54 msse3
-rwxr-xr-x  1 circleci circleci      1874 Mar 20 21:54 opencpn.appdata.xml
drwxr-xr-x  7 circleci circleci      4096 Mar 20 21:54 plugins
-rw-r--r--  1 circleci circleci       143 Mar 20 21:54 repack.sh
-rwxr-xr-x  1 circleci circleci       958 Mar 20 21:54 uninstall.cmake
```
The cmake output is as follows:
```
+ cd build-android-hf
+ rm -rf *
+ cmake -DCMAKE_BUILD_TYPE=Release -DOCPN_TARGET_TUPLE:STRING=Android-armhf;16;armhf -Dtool_base=/opt/android/android-ndk-r21e/toolchains/llvm/prebuilt/linux-x86_64 ..
-- Building for wxQt-Android
-- The C compiler identification is Clang 9.0.9
-- The CXX compiler identification is Clang 9.0.9
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Detecting C compile features
-- Detecting C compile features - failed
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Detecting CXX compile features
-- Detecting CXX compile features - failed
-- cmake version: 3.13.4
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- *** Host Build Architecture is armhf
-- Setting C++11 standard via cmake standard mechanism
-- Default compiler options:
-- CMAKE_CXX_FLAGS: 
-- CMAKE_CXX_FLAGS_DEBUG: -g
-- CMAKE_CXX_FLAGS_MINSIZEREL: -Os -DNDEBUG
-- CMAKE_CXX_FLAGS_RELEASE: -O3 -DNDEBUG
-- CMAKE_CXX_FLAGS_RELWITHDEBINFO: -O2 -g -DNDEBUG
-- CMAKE_INSTALL_LIBDIR1 
-- CMAKE_INSTALL_LIBDIR2 lib
-- CMAKE_INSTALL_LIBDIR3  lib
-- Building version 5.6.1
-- *** Staging to build gorp 5.6.1 ***
-- *** Package will NOT include documentation ***
-- Forcing wxWidgets_VERSION_STRING: 
-- 
-- *** Staging to build gorp  ***
-- *** Build type: Release
-- *** Will install to /usr/local  ***
-- GarminHost is enabled.
-- Android Build support file base:  /home/circleci/project/cache/OCPNAndroidCoreBuildSupport
-- PackageNem debug  gorp /home/circleci/project
-- Found Gettext: /usr/bin/msgmerge (found version "0.19.8.1") 
-- Could NOT find LIBUDEV (missing: UDEV_LIBRARY UDEV_INCLUDE_DIR) 
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- Could NOT find LIBUSB (missing: LIBUSB_INCLUDE_DIR LIBUSB_LIBRARIES) 
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- Could NOT find PANGO (missing: PANGO_INCLUDE_DIR PANGO_LIBRARIES) 
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- Could NOT find Sqlite (missing: SQLITE_INCLUDE_DIR SQLITE_LIBRARIES) 
-- Using bundled sqlite3 library.
-- SVG support enabled...
--     Adding local GLU
--     Revised GL Lib (with local): GLU_static
--     Adding local LIBTESS2
-- OCPN_SOUND_CMD selected : 
-- OCPN_ENABLE_SYSTEM_CMD_SOUND is set but I cannot find aplay(1) or afplay(1) or omxplayer(1)
-- HAVE_SYSTEM_CMD_SOUND = 
-- HAVE_PORTAUDIO = 
-- Could NOT find LibSndFile (missing: LIBSNDFILE_LIBRARY LIBSNDFILE_INCLUDE_DIR) 
-- ENABLE_LIBSNDFILE is set but I cannot find libsndfile
-- HAVE_SNDFILE = 
-- Building with bundled tinyxml
-- S57 ENC support: enabled
-- Garmin Host Mode support: enabled
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
-- Probing for compiler -msse support -- Fail
-- Probing for compiler -msse2 support -- Fail
-- Probing for compiler -msse3 support -- Fail
-- Probing for compiler -mavx2 support -- Fail
-- Probing for compiler -mfpu=neon support -- Fail
-- Checking for one of the modules 'liblz4;lz4'
-- Building with bundled lz4
-- Checking for one of the modules 'liblz4;lz4'
-- Building with bundled lz4
-- Building for target (Platform; version; arch): android-armhf; 16; armhf
-- *** Staging to build PlugIns ***
-- Build type: Release
-- *** Staging to build chartdldr_pi ***
-- chartdldr_pi: Building without wxDATAVIEWLISTCTRL chart listing
-- Checking for one of the modules 'libunarr;unarr'
-- chartdldr_pi: Building with bundled unarr
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY) (found version "1.2.7")
-- Could NOT find BZip2 (missing: BZIP2_LIBRARIES BZIP2_INCLUDE_DIR) 
-- *** Added plugin: /home/circleci/project/plugins/chartdldr_pi
-- 
-- *** Staging to build dashboard_pi ***
-- *** Added plugin: /home/circleci/project/plugins/dashboard_pi
-- *** Staging to build grib_pi ***
-- src/grib_pi.hsrc/grib_pi.cppsrc/GribOverlayFactory.cppsrc/GribOverlayFactory.hsrc/GribUIDialogBase.cppsrc/GribUIDialogBase.hsrc/GrabberWin.cppsrc/GrabberWin.hsrc/GribUIDialog.cppsrc/GribUIDialog.hsrc/CursorData.cppsrc/CursorData.hsrc/GribSettingsDialog.cppsrc/GribSettingsDialog.hsrc/GribRequestDialog.cppsrc/GribRequestDialog.hsrc/icons.hsrc/msg.hsrc/email.hsrc/email.cppsrc/pi_TexFont.hsrc/pi_TexFont.cppsrc/GribTable.hsrc/GribTable.cppsrc/CustomGrid.hsrc/CustomGrid.cppsrc/icons.cppsrc/GribReader.cppsrc/GribRecord.cppsrc/GribV1Record.cppsrc/GribV2Record.cppsrc/zuFile.cppsrc/IsoLine.cppsrc/pi_ocpndc.cppsrc/pi_ocpndc.hsrc/pi_shaders.hsrc/pi_shaders.cppsrc/bzip2/bzlib.csrc/bzip2/blocksort.csrc/bzip2/compress.csrc/bzip2/crctable.csrc/bzip2/decompress.csrc/bzip2/huffman.csrc/bzip2/randtable.c
-- Could NOT find JPEG (missing: JPEG_LIBRARY JPEG_INCLUDE_DIR) 
-- Could NOT find Jasper (missing: JASPER_LIBRARIES JASPER_INCLUDE_DIR JPEG_LIBRARIES) 
-- Using bundled jasper library...
-- *** Added plugin: /home/circleci/project/plugins/grib_pi
-- Build type: Release
-- *** Staging to build wmm_pi ***
-- Using GLESv1 for Android
-- *** Added plugin: /home/circleci/project/plugins/wmm_pi
-- *** Package will include tide and current data ***
-- Git status: clean
-- Configuring done
-- Generating done
-- Build files have been written to: /home/circleci/project/build-android-hf
```
How do I generate an android armhf runnable version of opencpn that I can use for debugging plugins under Android Studio?

leamas, 2022-03-20:
This is basically a duplicate of #2081

jongough, 2022-03-20:
No this is not a duplicate of that issue. The build works using the supplied circleci and completes cleanly. It does use a 653.9MB support.zip downloaded with all the android gui stuff. I am using the process that should be used by circleci in the cloud but running it locally. This is how I use Frontend2 when testing all builds that use docker images rather than machine images. This is what OCPN does. The default plugins are generated correctly into the lib directory.

bdbcat, 2022-03-20:
jon....
There is no such thing as an Android executable.  One builds an APK file, installs it, and then does a remote debug session over USB cable. There is a lot more involved in building an installable OpenCPN APK file for Android than the OCPN master c++ source.  The CCI scripts build only the native code module (libgorp.so).  This is combined with lots of Android Java modules, built using maven, to generate an APK.  I do all this manually for Android releases, today.
There is a plan to fully automate the building of Android APK files.  That is what is referred to in #2081.  It is not done yet, mainly due to the press of more universally important bugs and issues.  But someday, maybe soon after O562 releases....

jongough, 2022-03-21:
Is there a debug install available to help with debugging plugins on android? If not then it is not possible to really support android as it is all guesses, if yes where do developers find it and what tool is used to run the debugger? Currently the builds seem to default to Release so remote debugging is not possible. I only have Android Studio and its virtual android devices to do the work on.

bdbcat, 2022-03-21:
Jon...
I wish the Android development environment was as clean as linux, or even Windows.  But, for heavily native library dependent projects,  it is not.  Debugging native (.so) plugins by stepping through the code, line by line, is all but impossible.  I've spent literally years struggling with this.

So, how do I do it?
1.  Sprinkle qDebug() calls around suspicious points, like old fashioned "printf" debugging.  qDebug() has lots of overloads, so one can do things like:
```
double pi = 3.14;
qDebug() << pi;
// or
qDebug() << "This is point A"

```
2.  You may monitor the qDebug() output by connecting a target device by USB cable to PC host, and running "adb logcat".  To filter the logcat output, use "adb logcat -s "OpenCPN", "libgorp.so""
3. If using Android studio emulators, the adb logcat output is also available in the IDE.
4. adb logcat will also give a pretty good stack trace on crash.

I acknowledge that supporting Android with managed plugins is not easy.  But it is possible with time and patience.

jongough, 2022-03-21:
Thanks. I am trying to sort out a crash when displaying OCPN/Settings/Plugins after installing OD. OD works OK, but trying to get to the plugins page again instantly stops OCPN. There is no information in the log or anything else that I can see. The problem is I do not know what has changed that could affect the plugin manager, although I note, testing on ubuntu, that plugin manager 'runs' the plugin then deletes it when it displays installed plugins. Not quite what I was expecting, but....

jongough, 2022-03-21:
I have a logcat around about the time of displaying the Plugins tab in OCPN settings:
```
03-21 17:26:49.651  6434  6656 I chatty  : uid=10154(org.opencpn.opencpn) qtMainLoopThrea identical 1683 lines
03-21 17:26:49.651  6434  6656 D emuglGLESv2_enc: sendVertexAttributes: bad offset / len!!!!!
03-21 17:27:00.004   830   830 D KeyguardClockSwitch: Updating clock: 527
03-21 17:27:00.011   830   830 D KeyguardClockSwitch: Updating clock: 527
03-21 17:27:00.732  6434  6656 I OpenCPN :   getAndroidLocaleString language ID is: en_US
03-21 17:27:00.865  6434  6656 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x8 in tid 6656 (qtMainLoopThrea), pid 6434 (opencpn.opencpn)
03-21 17:27:00.749  6434  6656 I OpenCPN :   getAndroidLocaleString language ID is: en_US
03-21 17:27:00.922  6696  6696 I crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
03-21 17:27:00.922   285   285 I tombstoned: received crash request for pid 6656
03-21 17:27:00.923  6696  6696 I crash_dump32: performing dump of process 6434 (target tid = 6656)
03-21 17:27:00.927  6696  6696 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
03-21 17:27:00.927  6696  6696 F DEBUG   : Build fingerprint: 'google/sdk_gphone_x86_arm/generic_x86_arm:11/RSR1.201013.001/6903271:userdebug/dev-keys'
03-21 17:27:00.927  6696  6696 F DEBUG   : Revision: '0'
03-21 17:27:00.927  6696  6696 F DEBUG   : ABI: 'x86'
03-21 17:27:00.928  6696  6696 F DEBUG   : Timestamp: 2022-03-21 17:27:00+1100
03-21 17:27:00.928  6696  6696 F DEBUG   : pid: 6434, tid: 6656, name: qtMainLoopThrea  >>> org.opencpn.opencpn <<<
03-21 17:27:00.928  6696  6696 F DEBUG   : uid: 10154
03-21 17:27:00.928  6696  6696 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x8
03-21 17:27:00.928  6696  6696 F DEBUG   : Cause: null pointer dereference
03-21 17:27:00.928  6696  6696 F DEBUG   :     eax c0293cc4  ebx c36ef33c  ecx f44559c0  edx 00000000
03-21 17:27:00.928  6696  6696 F DEBUG   :     edi c36ef320  esi c36ef380
03-21 17:27:00.928  6696  6696 F DEBUG   :     ebp f1a3f030  esp c37f5650  eip c61405a7
03-21 17:27:00.968  6696  6696 F DEBUG   : backtrace:
03-21 17:27:00.968  6696  6696 F DEBUG   :       #00 pc 003385a7  <anonymous:c5e08000>
03-21 17:27:01.105     0     0 D logd    : logdr: UID=10154 GID=10154 PID=6696 n tail=50 logMask=8 pid=6434 start=0ns timeout=0ns
03-21 17:27:01.111     0     0 D logd    : logdr: UID=10154 GID=10154 PID=6696 n tail=50 logMask=1 pid=6434 start=0ns timeout=0ns
03-21 17:27:01.239     0     0 D logd    : logdr: UID=10154 GID=10154 PID=6696 n tail=0 logMask=8 pid=6434 start=0ns timeout=0ns
03-21 17:27:01.240     0     0 D logd    : logdr: UID=10154 GID=10154 PID=6696 n tail=0 logMask=1 pid=6434 start=0ns timeout=0ns
03-21 17:27:01.204   285   285 E tombstoned: Tombstone written to: /data/tombstones/tombstone_08
03-21 17:27:01.205   622  6704 I DropBoxManagerService: add tag=data_app_native_crash isTagEnabled=true flags=0x2
03-21 17:27:01.206   622  6703 W ActivityTaskManager:   Force finishing activity org.opencpn.opencpn/org.qtproject.qt5.android.bindings.QtActivity
03-21 17:27:01.206   622   664 I BootReceiver: Copying /data/tombstones/tombstone_08 to DropBox (SYSTEM_TOMBSTONE)
03-21 17:27:01.207   622   664 I DropBoxManagerService: add tag=SYSTEM_TOMBSTONE isTagEnabled=true flags=0x2
03-21 17:27:01.209   305   408 D goldfish-address-space: claimShared: Ask to claim region [0x3f298a000 0x3f338a000]
03-21 17:27:01.216   622  5353 D HostConnection: HostConnection::get() New Host Connection established 0xf4439820, tid 5353
03-21 17:27:01.219   622   656 W BroadcastQueue: Background execution not allowed: receiving Intent { act=android.intent.action.DROPBOX_ENTRY_ADDED flg=0x10 (has extras) } to com.google.android.gms/.stats.service.DropBoxEntryAddedReceiver
03-21 17:27:01.271     0     0 I init    : Untracked pid 6696 exited with status 0
03-21 17:27:01.230   622   656 W BroadcastQueue: Background execution not allowed: receiving Intent { act=android.intent.action.DROPBOX_ENTRY_ADDED flg=0x10 (has extras) } to com.google.android.gms/.chimera.GmsIntentOperationService$PersistentTrustedReceiver
03-21 17:27:01.278     0     0 I init    : Untracked pid 6698 exited with status 0
03-21 17:27:01.250   622   742 W InputDispatcher: channel 'd452c14 org.opencpn.opencpn/org.qtproject.qt5.android.bindings.QtActivity (server)' ~ Consumer closed input channel or an error occurred.  events=0x9
03-21 17:27:01.250   622   742 E InputDispatcher: channel 'd452c14 org.opencpn.opencpn/org.qtproject.qt5.android.bindings.QtActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
03-21 17:27:01.254   292   292 I Zygote  : Process 6434 exited due to signal 11 (Segmentation fault)
03-21 17:27:01.301     0     0 I binder  : undelivered transaction 893205, process died.
03-21 17:27:01.254   622   656 W BroadcastQueue: Background execution not allowed: receiving Intent { act=android.intent.action.DROPBOX_ENTRY_ADDED flg=0x10 (has extras) } to com.google.android.gms/.stats.service.DropBoxEntryAddedReceiver
03-21 17:27:01.255   622   656 W BroadcastQueue: Background execution not allowed: receiving Intent { act=android.intent.action.DROPBOX_ENTRY_ADDED flg=0x10 (has extras) } to com.google.android.gms/.chimera.GmsIntentOperationService$PersistentTrustedReceiver
[... cut ...] rces_with_requirements ANDROID_EMU_YUV_Cache ANDROID_EMU_vulkan_ignored_handles ANDROID_EMU_has_shared_slots_host_memory_allocator ANDROID_EMU_vulkan_free_memory_sync ANDROID_EMU_vulkan_shader_float16_int8 ANDROID_EMU_vulkan_async_queue_submit ANDROID_EMU_sync_buffer_data ANDROID_EMU_read_color_buffer_dma GL_OES_EGL_image_external_essl3 GL_OES_vertex_array_object GL_KHR_texture_compression_astc_ldr ANDROID_EMU_host_side_tracing ANDROID_EMU_gles_max_version_3_0 
```
But I cannot get much from it.

bdbcat, 2022-03-22:
Jon..
x86 builds of OpenCPN are not very thoroughly tested.  They are not shipped in Production.  So the crash in qt:
`
03-21 17:27:00.865  6434  6656 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x8 in tid 6656 (qtMainLoopThrea), pid 6434 (opencpn.opencpn) 

`could be from anything, really.
Do you have an armhf build of ODraw with a problem that I could import to a real device, and test?

jongough, 2022-03-22:
As far as I am aware the OD build is armhf using docker to provide the environment:
```
    build-android-armhf:
        docker:
        - image: cimg/android:2021.10.2-ndk
        resource_class: large
        environment:
        - OCPN_TARGET:  android-armhf
        - DEPLOY_USE_ORB: true
        steps:
        - checkout
        - run: chmod a+x ci/*.sh
        - run: bash ci/circleci-build-android-armhf.sh
        - deploy-code:
            install-python: true
```
with cmake:
```
cmake  \
  -D_wx_selected_config=androideabi-qt-armhf \
  -DwxQt_Build=build_android_release_19_static_O3 \
  -DQt_Build=build_arm32_19_O3/qtbase \
  -DCMAKE_AR=$COMPDIR/toolchains/llvm/prebuilt/linux-x86_64/bin/arm-linux-androideabi-ar \
  -DCMAKE_CXX_COMPILER=$COMPDIR/toolchains/llvm/prebuilt/linux-x86_64/bin/armv7a-linux-androideabi21-clang++ \
  -DCMAKE_C_COMPILER=$COMPDIR/toolchains/llvm/prebuilt/linux-x86_64/bin/armv7a-linux-androideabi21-clang \
  -DOCPN_Android_Common=OCPNAndroidCommon-master \
  -DCMAKE_INSTALL_PREFIX=/ \
  ..
```
and using your `master.zip`:
```
https://github.com/bdbcat/OCPNAndroidCommon/archive/master.zip
```
The fact that I am building it locally using `circleci local` should not make any difference.

There is an alpha build in cloudsmith:
[https://cloudsmith.io/~opencpn/repos/ocpn_draw-alpha/packages/?q=*1.8.14.2*armhf*](https://cloudsmith.io/~opencpn/repos/ocpn_draw-alpha/packages/?q=*1.8.14.2*armhf*)
built by circleci.

jongough, 2022-03-24:
I have just done a test run on android and on ubuntu and got the following from the code generated using circleci local and using qDegug for android and std::cout for ubuntu (this is done automatically via defines). I imported OD from the local drive using the tz file, I then clicked the install button. On ubuntu it installed and on clicking the Enable checkbox it all worked. On andriod clicking the install button caused OCPN to crash with the following output in logcat:
```
03-24 18:57:13.656  7069  7105 D libgorp.so: Thu Mar 24 18:57:13 2022  ::  create_pi 
03-24 18:57:13.656  7069  7105 D libgorp.so: Thu Mar 24 18:57:13 2022  ::  ocpn_draw_pi 
03-24 18:57:13.689  7069  7105 D libgorp.so: Thu Mar 24 18:57:13 2022  ::  ocpn_draw_pi return 
03-24 18:57:13.713  7069  7105 I OpenCPN :   getAndroidLocaleString language ID is: en_US
03-24 18:57:13.721  7069  7069 I OpenCPN : invokeDelegateMethod onWindowFocusChanged
03-24 18:57:13.735  7069  7105 I OpenCPN :   getAndroidLocaleString language ID is: en_US
03-24 18:57:13.760  7069  7105 F libc    : Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x8 in tid 7105 (qtMainLoopThrea), pid 7069 (opencpn.opencpn)
03-24 18:57:13.762   602  4799 I system_server: oneway function results will be dropped but finished with status OK and parcel size 4
03-24 18:57:13.827  7154  7154 I crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
03-24 18:57:13.828   284   284 I tombstoned: received crash request for pid 7105
03-24 18:57:13.829  7154  7154 I crash_dump32: performing dump of process 7069 (target tid = 7105)
```
The same process on ubuntu generated this output:
```
Thu Mar 24 19:09:07 2022 :: create_pi
Thu Mar 24 19:09:07 2022 :: ocpn_draw_pi
Thu Mar 24 19:09:16 2022 :: ocpn_draw_pi return
Thu Mar 24 19:09:16 2022 :: destroy_pi
Thu Mar 24 19:09:16 2022 :: ~ocpn_draw_pi
Thu Mar 24 19:09:30 2022 :: create_pi
Thu Mar 24 19:09:30 2022 :: ocpn_draw_pi
Thu Mar 24 19:09:35 2022 :: ocpn_draw_pi return
Thu Mar 24 19:09:35 2022 :: destroy_pi
Thu Mar 24 19:09:35 2022 :: ~ocpn_draw_pi
```
At this point in ubuntu I get the message that OD is installed successfully. Clicking OK gives:
```
Thu Mar 24 19:20:29 2022 :: create_pi
Thu Mar 24 19:20:29 2022 :: ocpn_draw_pi
Thu Mar 24 19:21:05 2022 :: ocpn_draw_pi return
Thu Mar 24 19:21:05 2022 :: destroy_pi
Thu Mar 24 19:21:05 2022 :: ~ocpn_draw_pi
```
The plugin is only installed and not enabled. If the enabled button is clicked I get:
```
Thu Mar 24 19:21:44 2022 :: create_pi
Thu Mar 24 19:21:44 2022 :: ocpn_draw_pi
Thu Mar 24 19:22:06 2022 :: ocpn_draw_pi return
Thu Mar 24 19:22:06 2022 :: Init
Thu Mar 24 19:22:06 2022 :: AddLocaleCatalog
Thu Mar 24 19:22:06 2022 :: PluginGetFocusCanvas
Thu Mar 24 19:22:06 2022 :: LoadConfig
Thu Mar 24 19:22:06 2022 :: Width: 22, Height: 22
Thu Mar 24 19:22:06 2022 :: Init return
Thu Mar 24 19:22:06 2022 :: LateInit
```
Not sure why ubuntu OCPN is creating and destroying the plugin so many times, but.....