OpenCPN/OpenCPN #2594: COG predictor deviation bug when zooming in.

Hakansv, 2022-03-26:
A strange bug in master code.
Win10. Both oeuSENC and raster charts. (It's probably also present on the RPi but there I've problem building from source so it's not verified.)
When zooming in the course predictor line for AIS targets is changed about 90°. It happens at different chart scales dependent on the SOG of the ship. For 20 knots around 1:8000, 10 knots 1:4000 and corresponding bigger scale at lower speed.
See two shots for a AIS target speed about 8 knots. The first at about 1:5000 chart scale and the next at ~1:3400. As seen from the target query the course is not changed.

When own ship is involved O on the chart O halted instead of a predictor line change but the break point logic is more or less the same as for when only AIS targets are present on screen.
My VS debug shows nothing when O freezes. 

[image]

[image]

Hakansv, 2022-04-04:
More info.
I've reproduced this behavior on other Win10, a RPi400 and also Android (v87). 
It though seems to be dependent on some setting(s) since a friend of mine couldn't reproduce running the same code. I'll check more for differences.

Hakansv, 2022-04-04:
ais.cpp
```
spherical_ll_gc_ll(td->Lat, td->Lon, td->COG,
                     target_sog * g_ShowCOG_Mins / 60., &pred_lat, &pred_lon);
```
From own tests: It could be that when the sum of target_sog * g_ShowCOG_Mins equals >= 1.0 this issue is valid. Not else.

bdbcat, 2022-04-05:
Corrected by ff9b2848d4bde719f414a15d9a27f9ed972fe6d7

Hakansv, 2022-04-05:
@bdbcat 
Yes it helped but not enough. For high speed targets the arrow still gets wrong angle.
If I increase the validity test limits another ten times (< 10^6) it solved also for 27 knots 60 minutes length and 1:800 scale.
But If the validity test then still fulfills its purpose I don't know.

bdbcat, 2022-04-05:
I'll switch to 10^6.  The test is intended to catch non-sense from the lat/lon to pixel calculations.  Of course, a million pixels would require a very large screen..  But is still a valid number.
Thanks
-d

Hakansv, 2022-04-05:
Very good.
Thanks

closed 2022-04-05 by Hakansv