OpenCPN/OpenCPN #2595: Image tearing in mbtiles when tile data is not 256 x 256

IJMacD, 2022-03-29:
OpenCPN assumes the tiles in mbtiles files will always be 256x256 pixels. This isn't always the case. "Hi-res" versions are available in the wild at 2x resolution i.e. 512x512.

The section of code below copies image data from a buffer that might not be 256x256 into one that is. This results in tearing if the assumption proved false.

https://github.com/OpenCPN/OpenCPN/blob/af0baad318cfadd85ae6241b5573daf6e24dd125/src/mbtiles.cpp#L826-L837

There is a row in the metadata table which hints at the scaling (`name = 'scaling', value = '2.00000'` for example), but the image size has already been read so should be taken into account.

https://github.com/OpenCPN/OpenCPN/blob/af0baad318cfadd85ae6241b5573daf6e24dd125/src/mbtiles.cpp#L775-L776

bdbcat, 2022-03-31:
IJMacD...
Can you post a link to an MBTiles fileset that contains 512x512 tiles?  Would be useful for testing and confirmation of the fix.
Thanks
Dave

IJMacD, 2022-04-01:
I have made the file available in a gist.

https://gist.github.com/IJMacD/e7eacd6d6f3e8d2fe72fbc52257837a8

The file is available free (as in beer) on the internet but not free (as in speech) and does not permit redistribution. I've included it here for demonstration purposes but will need to take it down in a month or so.

If you need some truly free tiles I could make a sample.

MapTiler Desktop calls them "Retina" or "HiDPI" tiles.

https://documentation.maptiler.com/hc/en-us/articles/4410674590609-Retina-HiDPI-tiles

bdbcat, 2022-04-01:
OK, got it.
No redist.  Demonstration only.  Agreed.

bdbcat, 2022-04-02:
Somewhat crude, but safe correction for HiDPI tiles made in 7798a4ad89050a50700837dedd3d8f258f5afa15
It may be possible in future version to realize the full resolution contained in a HiDPI tileset.  This will require some significant refactoring and testing.   Best deferred for O5.8.

closed 2024-02-15 by nohal