OpenCPN/OpenCPN #2603: No debugging capabiltiy for OpenCPN on android

jongough, 2022-04-09:
Building and supporting plugins on all the different platforms requires that there is some method of debugging and allowing the developer to build versions of OCPN for the test platform with debugging enabled. It also requires the use of a suitable environment for testing the debug builds.

Currently the only debugging method available for developers is to use qDebug in the android builds as a "printf" equivalent to put messages out to the adb logcat. However, this only allows the plugin to have qDebug statements and there is no capability to step through the logic to help understand where the problem is occurring.

Without a suitable debugging environment production problems in OCPN and associated plugins on android are not supportable. ODraw 1.8.14.x is suffering from this issue. For example after the install of ODraw and the 'succeeded' message, OCPN crashes. There is nothing in the log or logcat to say what is happening.

opencpn.log 
```
15:12:43.576 MESSAGE pluginmanager.cpp:1174 Checking plugin candidate: /data/user/0/org.opencpn.opencpn/libchartdldr_pi.so
15:12:43.576 MESSAGE pluginmanager.cpp:1174 Checking plugin candidate: /data/user/0/org.opencpn.opencpn/libdashboard_pi.so
15:12:43.577 MESSAGE pluginmanager.cpp:1174 Checking plugin candidate: /data/user/0/org.opencpn.opencpn/libgrib_pi.so
15:12:43.578 MESSAGE pluginmanager.cpp:1174 Checking plugin candidate: /data/user/0/org.opencpn.opencpn/libwmm_pi.so
15:12:43.579 MESSAGE OCPNPlatform.cpp:1016 ChangeLocale: Language load for: 
```
logcat:
```
04-09 15:12:22.118  6233  6233 I OpenCPN : FileChooserDialog selected: ocpn_draw_pi-1.8.14.5-android-armhf-16-android-armhf.tar.gz
04-09 15:12:22.184  6233  6233 I OpenCPN : invokeDelegateMethod onWindowFocusChanged
04-09 15:12:22.898  6233  6268 D libgorp.so: onTimerEvent 3
04-09 15:12:22.898  6233  6268 I OpenCPN : isFileChooserFinished:  returning file:/storage/emulated/0/Download/ocpn_draw_pi-1.8.14.5-android-armhf-16-android-armhf.tar.gz
04-09 15:12:23.168  6233  6268 I OpenCPN : SecureFileCopy: /storage/emulated/0/Download/ocpn_draw_pi-1.8.14.5-android-armhf-16-android-armhf.tar.gz to: /storage/emulated/0/Android/data/org.opencpn.opencpn/files/plugins/cache/tarballs/ocpn_draw_pi-1.8.14.5-android-armhf-16-android-armhf.tar.gz
04-09 15:12:23.234  6233  6268 I OpenCPN :   getAndroidLocaleString language ID is: en_US
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_9 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_10 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_7 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_6 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_3 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_8 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_4 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_11 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_5 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.435   570  1619 W InputReader: Device virtio_input_multi_touch_2 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:23.495   308  1286 D goldfish-address-space: claimShared: Ask to claim region [0x3ea338000 0x3ea514000]
04-09 15:12:23.504   308  1286 D goldfish-address-space: claimShared: Ask to claim region [0x3ea514000 0x3ea6f0000]
04-09 15:12:23.505   308  1286 D goldfish-address-space: claimShared: Ask to claim region [0x3ea6f0000 0x3ea8cc000]
04-09 15:12:23.516  6233  6233 I OpenCPN : invokeDelegateMethod onWindowFocusChanged
04-09 15:12:23.573  6233  6250 W System  : A resource failed to call close. 
04-09 15:12:23.573  6233  6250 W System  : A resource failed to call close. 
04-09 15:12:24.616  6233  6268 D libgorp.so: OCPNMB-1 ret 5100
04-09 15:12:24.617   570   934 W InputReader: Device virtio_input_multi_touch_9 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.617   570   934 W InputReader: Device virtio_input_multi_touch_10 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.617   570   934 W InputReader: Device virtio_input_multi_touch_7 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.617   570   934 W InputReader: Device virtio_input_multi_touch_6 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.617   570   934 W InputReader: Device virtio_input_multi_touch_3 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.617   570   934 W InputReader: Device virtio_input_multi_touch_8 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.617   570   934 W InputReader: Device virtio_input_multi_touch_4 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.617   570   934 W InputReader: Device virtio_input_multi_touch_11 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.618   570   934 W InputReader: Device virtio_input_multi_touch_5 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.618   570   934 W InputReader: Device virtio_input_multi_touch_2 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:24.618  6233  6268 D libgorp.so: OCPNMB-2 ret 5100
04-09 15:12:24.696  6233  6233 I OpenCPN : invokeDelegateMethod onWindowFocusChanged
04-09 15:12:26.110  6233  6268 D libgorp.so: 1
04-09 15:12:26.257   308  1286 D goldfish-address-space: claimShared: Ask to claim region [0x3ea338000 0x3eb058000]
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_9 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_10 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_7 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_6 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_3 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_8 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_4 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_11 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_5 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.626   570  1619 W InputReader: Device virtio_input_multi_touch_2 is associated with display ADISPLAY_ID_NONE.
04-09 15:12:28.680   308  1286 D goldfish-address-space: claimShared: Ask to claim region [0x3eb058000 0x3eb205000]
04-09 15:12:28.733   308  1286 D goldfish-address-space: claimShared: Ask to claim region [0x3eb205000 0x3eb3b2000]
04-09 15:12:28.742   308   308 D goldfish-address-space: claimShared: Ask to claim region [0x3eb3b2000 0x3eb55f000]
04-09 15:12:28.747  6233  6233 I OpenCPN : invokeDelegateMethod onWindowFocusChanged
04-09 15:12:32.322   209   212 E android.system.suspend@1.0-service: Error opening kernel wakelock stats for: wakeup34: Permission denied
04-09 15:12:32.316   209   209 W Binder:209_2: type=1400 audit(0.0:281): avc: denied { read } for name="wakeup34" dev="sysfs" ino=19031 scontext=u:r:system_suspend:s0 tcontext=u:object_r:sysfs:s0 tclass=dir permissive=0
04-09 15:12:32.327   209   212 E android.system.suspend@1.0-service: Error opening kernel wakelock stats for: wakeup35: Permission denied
[... 37 lines cut ...]
04-09 15:12:43.591  6233  6268 I OpenCPN :   getAndroidLocaleString language ID is: en_US
04-09 15:12:43.656  6340  6340 I crash_dump32: obtaining output fd from tombstoned, type: kDebuggerdTombstone
04-09 15:12:43.657   283   283 I tombstoned: received crash request for pid 6268
04-09 15:12:43.657  6340  6340 I crash_dump32: performing dump of process 6233 (target tid = 6268)
[... cut ...]
04-09 15:12:43.966     0     0 D logd    : logdr: UID=10155 GID=10155 PID=6340 n tail=0 logMask=8 pid=6233 start=0ns timeout=0ns
04-09 15:12:43.968     0     0 D logd    : logdr: UID=10155 GID=10155 PID=6340 n tail=0 logMask=1 pid=6233 start=0ns timeout=0ns
0
```
About all that can be said is that OCPN crashed after ascertaining the language via `getAndroidLocaleString()`. It can be seen that ODraw was instantiated as a plugin and then class itself was instantiated as an object. No other information is readily available.

leamas, 2022-04-09:
The important piece of info here is the stacktrace. Unfortunately, it only contains raw addresses, though. I'm not really comfortable with the clang toolchain -- using gcc these raw addresses can be converted to source lines using addr2line(1). There seems to be a tool  called [ndk-stack](https://developer.android.com/ndk/guides/ndk-stack) which does something similar.

Without source lines it is obvious that this happens in `LoadPlugInDirectory()`. Furthermore, when invoking `wxBitmap::GetWidth()`.

The two possible places are in this snippet  in pluginmanager.cpp around line 1307:
```
  pic->m_version_str = found_version;
        pic->m_short_description = pic->m_pplugin->GetShortDescription();
        pic->m_long_description = pic->m_pplugin->GetLongDescription();
        pic->m_version_major = pic->m_pplugin->GetPlugInVersionMajor();
        pic->m_version_minor = pic->m_pplugin->GetPlugInVersionMinor();
        pic->m_bitmap = pic->m_pplugin->GetPlugInBitmap();
        wxBitmap *pbm = new wxBitmap(pic->m_bitmap->GetSubBitmap(wxRect(
              0, 0, pic->m_bitmap->GetWidth(), pic->m_bitmap->GetHeight())));
                                      ^
        pic->m_bitmap = pbm;

        ret = true;

        if (!pic->m_bEnabled && pic->m_destroy_fn) {
          wxBitmap *pbm = new wxBitmap(pic->m_bitmap->GetSubBitmap(wxRect(
              0, 0, pic->m_bitmap->GetWidth(), pic->m_bitmap->GetHeight())));
                                      ^
          pic->m_bitmap = pbm;
          pic->m_destroy_fn(pic->m_pplugin);
          pic->m_destroy_fn = NULL;
          pic->m_pplugin = NULL;
          pic->m_bInitState = false;
          if (pic->m_library.IsLoaded()) pic->m_library.Unload();
        }
```
Which basically boils down to that  `pic->m_bitmap` probably is 0.

which points to `LoadPlugIn()`
which points to `m_library.Load()`
which points  to  [wxDynamicLibrary::Load()](https://docs.wxwidgets.org/3.0/classwx_dynamic_library.html#a8fc30053e38d44cfe0b2252a16598c44)

Two conclusions:
   - pluginmanager could be a more conservative, testing if `m_bitmap is != 0` before accessing it (as done in other places).
   - Something in this plugin makes the bitmap unloadable.

jongough, 2022-04-09:
I think there may be a typo in the bitmap name for png files. Investigating.
Should android plugin bitmaps be png, svg or mui_svg?

nohal, 2022-04-09:
The crash on plugins providing no or broken bitmaps is addressed in f1cd36764 (I have accidentaly hit the same thing yesterday as well)
`opencpn_plugin::GetPlugInBitmap()` returns a pointer to `wxBitmap`, to get that bitmap I would say nowadays the most reasonable way is to provide a SVG file and call `GetBitmapFromSVGFile()` API to generate the bitmap.

jongough, 2022-04-09:
I have switch to using svg icons which has got rid of that crash. But it appears that using wxDialog::Destroy() also causes a crash on android. 

So, the issue still exists, there is no real, available, debugging environment usable by plugin developers. This makes identifying and fixing issues raised in the 'Prod' environment very difficult and time consuming with most of it happening with guesswork.

bdbcat, 2022-04-10:
Jon...
"But it appears that using wxDialog::Destroy() also causes a crash on android."
Which dialog?  If it is a top-level dialog within a plugin, why not do an immediate "delete" on the dialog object?  If the plugin is exiting, there is really no telling how the race conditions will turn out.  The loaded library could be unloaded and reallocated before wxWidgets gets around to finishing the Destroy() method by flushing the dialog's event queue and formally deleting the object.  Especially unpredictable on Android, since there are two event queues running (wxWidgets and QT), in different threads.  It is, by design, a bit fragile.

And, yes, btw, the issue of Android debugging still exists, and will continue to do so.  Android native code debugging is very poorly supported, in comparison to Java side resources. But that should not prevent us from trying to solve, by whatever means available, the bugs we encounter.

jongough, 2022-04-10:
Looking further into this there seems to be an anomaly in the actions between disabling a plugin when the  DeInit and destroy_pi get called and shutting down the application via the 'X' in the list of running applications which appears to do nothing with the plugins (is it really shutdown) and the force stop of the application which also seems not to do any cleanup (certainly none of the destructors in the plugin are called.

I wonder if this is the cause of some of the 'strange' behavior?