OpenCPN/OpenCPN #2608: OpenCPN 5.6.1 beta crashes on exit

mgrouch, 2022-04-10:
```

ASSERT INFO:
/usr/include/wx-3.0/wx/datetime.h(852): assert "IsValid() && dt.IsValid()" failed in operator<=(): invalid wxDateTime

BACKTRACE:
[1] wxDateTime::operator<=(wxDateTime const&) const
[2] ActiveTrack::DoExtendDaily()
[3] wxMBConvUTF16LE::~wxMBConvUTF16LE()
[4] std::enable_if<std::__and_<std::__not_<std::__is_tuple_like<bool> >, std::is_move_constructible<bool>, std::is_move_assignable<bool> >::value, void>::type std::swap<bool>(bool&, bool&)
[5] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[6] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[7] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[8] wxEvtHandler::TryHereOnly(wxEvent&)
[9] wxEvtHandler::DoTryChain(wxEvent&)
[10] wxEvtHandler::ProcessEvent(wxEvent&)
[11] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[12] wxWindowBase::Close(bool)

```

It's on Raspberry pi, Buster, armhf

bdbcat, 2022-04-11:
Is this crash consistent and reproducible, or occasional, or one time only?

mgrouch, 2022-04-11:
@bdbcat 

it is consistent. Happens every time. 
On 5.6.1-beta and on 5.6.0. 

Users can’t upgrade from 5.2.4 because of it. 
and on 5.2.4 they can’t download o-charts. 
Is there o-charts_pi for armhf buster for OpenCPN 5.2.4?

thanks

nohal, 2022-04-11:
It seems like an assert. There should be no asserts generated from release builds. Is what you are running a package provided by us or your own build?

mgrouch, 2022-04-11:
It pops up window on 5.6.0 release.

nohal, 2022-04-11:
Yes, that is what asserts do, in debug builds. Is what you are running a package provided by us or your own build?

mgrouch, 2022-04-11:
I’m running .deb package provided by OpenCPN team

bdbcat, 2022-04-11:
"There should be no asserts generated from release builds."
There is some bad code in wxWidgets that leaks ASSERTS on wxDateTime faults, even in Release mode.
Needs an upstream Bug Report/PR, I think.

mgrouch, 2022-04-11:
Looks like wxWidgets are pretty close to releasing 3.2 version.

transmitterdan, 2022-04-11:
> I’m running .deb package provided by OpenCPN team

Can you tell us exactly how/where you got the .deb package?

I have run this on RPi and never seen this ASSERT.

bdbcat, 2022-04-11:
"Users can’t upgrade from 5.2.4 because of it.
and on 5.2.4 they can’t download o-charts."
Do you mean that every user is affected?

"Is there o-charts_pi for armhf buster for OpenCPN 5.2.4?"
No.

mgrouch, 2022-04-11:
> > I’m running .deb package provided by OpenCPN team
> 
> Can you tell us exactly how/where you got the .deb package?
> 
> I have run this on RPi and never seen this ASSERT.

`sudo apt install opencpn=5.6.0~ubuntu18.04.1
`
installs from 

```
Get:1 http://ppa.launchpad.net/opencpn/opencpn/ubuntu bionic/main armhf opencpn armhf 5.6.0~ubuntu18.04.1 [10.8 MB]
Fetched 10.8 MB in 1s (10.5 MB/s)   

```

Same error with popup window on exit (with 5.6.0 release)

leamas, 2022-04-11:
All gtk3 builds forces using -NDEBUG  due to #1500. Which means that this should not  happen on almost any linux build.

However: buster is using gtk2, so the fix for gtk3 in CMakeLists.txt does not enforce -DNDEBUG here. Could the PPA  buster version possibly thus be built without -DNDEBUG? Build logs should tell. 

Out of curiosity: is the situation the same using the official package? Build logs for it is [here](https://buildd.debian.org/status/fetch.php?pkg=opencpn&arch=armhf&ver=5.6.0%2Bdfsg0-1%7Ebpo10%2B1&stamp=1641999566&raw=0); it basically uses `-g -O2 -DNDEBUG` as default flags, so at least this should not trigger any asserts. To test it:

- Uninstall opencpn.
- Remove the PPA from /etc/apt/sources.list (or possibly some file in /etc/apt/sources.list.d)
- Enable the backports-sloppy repos as described [here](https://backports.debian.org/Instructions/)
- Re-install opencpn

mgrouch, 2022-04-11:
The one from sloppy (5.6.0) crashes on startup

```

sudo apt install opencpn/buster-backports-sloppy

Selected version '5.6.0+dfsg0-1~bpo10+1' (Debian Backports:oldstable-backports-sloppy [armhf]) for 'opencpn'

opencpn 
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
 SENC: nCPU: 4    m_max_jobs :3
pure virtual method called
terminate called without an active exception
Aborted

23:11:21.629 MESSAGE chart1.cpp:1852 

________

23:11:21.629    INFO logger.cpp:130 ------- OpenCPN version 5.6.0 restarted at 2022-04-11 -------

23:11:21.629    INFO logger.cpp:130 Using loglevel INFO
23:11:21.629 MESSAGE chart1.cpp:1876 wxWidgets version: wxWidgets 3.0.4 Linux 64 bit wxGTK
23:11:21.661 MESSAGE chart1.cpp:1881 OS Version reports as:  5.10
23:11:21.661 MESSAGE chart1.cpp:1883 MemoryStatus:  mem_total: 7760 mb,  mem_initial: 211 mb
23:11:21.661 MESSAGE chart1.cpp:1897 OCPN_OSDetail:  armhf ; Raspbian GNU/Linux ; 10 ; raspbian ; debian
23:11:21.661 MESSAGE chart1.cpp:1904 SData_Locn is /usr/share/opencpn/
23:11:21.662 MESSAGE chart1.cpp:1919 PrivateDataDir is /home/user/.opencpn
23:11:21.662 MESSAGE chart1.cpp:1974 Using existing Config_File: /home/user/.opencpn/opencpn.conf
23:11:21.687 MESSAGE navutil.cpp:1248 winPluginDir, read from ini file: 
23:11:21.688 MESSAGE navutil.cpp:1295 Setting Viewpoint Lat/Lon 50.0434, -52.7241
23:11:21.688 MESSAGE navutil.cpp:1332 Setting Ownship Lat/Lon 40.9297, -74.1266
23:11:21.690 MESSAGE navutil.cpp:3183 Switch InlandEcdis mode Off
23:11:21.697 MESSAGE styles.cpp:866 Styles loading from /usr/share/opencpn/uidata/styles.xml
23:11:21.702 MESSAGE styles.cpp:843 No styles found at: /home/user/
23:11:21.703 MESSAGE styles.cpp:843 No styles found at: /home/user/.opencpn/
23:11:21.739 MESSAGE styles.cpp:800 Using SVG Icons
23:11:21.739 MESSAGE chart1.cpp:2034 Detected display size (horizontal): 338 mm
23:11:21.740 MESSAGE OCPNPlatform.cpp:913 Getting DefaultSystemLocale...
23:11:21.740 MESSAGE chart1.cpp:2070 System default Language:  en_GB
23:11:21.740 MESSAGE chart1.cpp:2073 Config file language:  
23:11:21.740 MESSAGE chart1.cpp:2078 Adjusted App language:  
23:11:21.740 MESSAGE OCPNPlatform.cpp:994 ChangeLocale: Language load for:  
23:11:21.760 MESSAGE OCPNPlatform.cpp:1047 ChangeLocale: Locale Init OK for:  
23:11:21.760 MESSAGE OCPNPlatform.cpp:1066 Loading catalog for opencpn core.
23:11:21.780 MESSAGE OCPNPlatform.cpp:913 Getting DefaultSystemLocale...
23:11:21.780 MESSAGE chart1.cpp:2085 Opencpn language set to:  
23:11:21.780 MESSAGE ConfigMgr.cpp:631 Loading Configs catalog: /home/user/.opencpn/Configs/configs.xml
23:11:21.783 MESSAGE chart1.cpp:2306 Creating MyFrame...size(1280, 766)  position(0, 34)
23:11:21.784 MESSAGE chart1.cpp:2889  **** Connect stuff
23:11:23.409 MESSAGE chcanv.cpp:959 Creating glChartCanvas
23:11:23.527 MESSAGE chcanv.cpp:2374 Metrics:  m_display_size_mm: 338     g_Platform->getDisplaySize():  1280:800   
23:11:23.527 MESSAGE chcanv.cpp:2379 wxDisplaySize(): 1280 800
23:11:23.527 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/share/opencpn/tcdata/harmonics-dwf-20210110-free.tcd
23:11:23.527 MESSAGE tcmgr.cpp:673    Error loading Tide/Currect data source /usr/share/opencpn/tcdata/harmonics-dwf-20210110-free.tcd Error Code: TC_FILE_NOT_FOUND
23:11:23.527 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: /usr/share/opencpn/tcdata/harmonics-dwf-20210110-free.tcd
23:11:23.528 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONICS_NO_US.IDX
23:11:23.581 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONICS_NO_US.IDX
23:11:23.632 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONICS_NO_US.IDX
23:11:23.683 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONICS_NO_US.IDX
23:11:23.734 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONICS_NO_US.IDX
23:11:23.785 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONICS_NO_US.IDX
23:11:23.836 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONICS_NO_US.IDX
23:11:23.886 MESSAGE TCDataSource.cpp:50 Loading Tide/Current data source: /usr/share/opencpn/tcdata/HARMONICS_NO_US.IDX
23:11:24.037 MESSAGE chartsymbols.cpp:1181 ChartSymbols loaded from /usr/share/opencpn/s57data/chartsymbols.xml
23:11:24.134 MESSAGE chart1.cpp:1328 Using s57data in /usr/share/opencpn/s57data
23:11:24.328 MESSAGE s52plib.cpp:518 Core s52plib:  ppmm: 3.78698 rv_scale_factor: 1.40833  calc_display_size_mm: 338
[... 163 lines cut ...]
23:11:27.377 MESSAGE pluginmanager.cpp:1183 Skipping not enabled candidate.
23:11:27.378 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libtwocan_plugin_pi.so
23:11:27.378 MESSAGE pluginmanager.cpp:1183 Skipping not enabled candidate.
23:11:27.378 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libwindvane_pi.so
23:11:27.378 MESSAGE pluginmanager.cpp:1183 Skipping not enabled candidate.
23:11:27.378 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libvdr_pi.so
23:11:27.378 MESSAGE pluginmanager.cpp:1183 Skipping not enabled candidate.
[... cut ...]
23:11:27.379 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libdashboard_tactics_pi.so
23:11:27.379 MESSAGE pluginmanager.cpp:1183 Skipping not enabled candidate.
23:11:27.379 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libdashboard_pi.so
23:11:27.379 MESSAGE pluginmanager.cpp:1183 Skipping not enabled candidate.
23:11:27.379 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libchartdldr_pi.so
23:11:27.379 MESSAGE pluginmanager.cpp:1183 Skipping not enabled candidate.

```

mgrouch, 2022-04-11:
actually it run after I deleted ~/.opencpn

but it's fresh install then not upgrade. And still crashes going into plugins tab in settings

nohal, 2022-04-11:
Are you sure it crashes even when you disable all the many thirdparty plugins? If a plugin is what crashes it, please open issue in the respective project.

nohal, 2022-04-11:
> but it's fresh install then not upgrade. And still crashes going into plugins tab in settings

This seems a lot like a faulty plugin, not a bug in OpenCPN core. Please try to open issues only for bugs in the core application and the 4 plugins distributed with it. We do not develop the 3rd party plugins and can't fix them.

mgrouch, 2022-04-11:
yes I remove ~/.opencpn. So no plugins

user@bbn:~ $ rm -rf .opencpn
user@bbn:~ $ opencpn 
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4

(opencpn:17512): Gtk-CRITICAL **: 23:25:40.259: gtk_window_resize: assertion 'height > 0' failed

(opencpn:17512): Gtk-CRITICAL **: 23:25:47.931: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar

(opencpn:17512): Pango-CRITICAL **: 23:25:48.679: pango_layout_set_width: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.679: pango_layout_set_alignment: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.680: pango_layout_set_text: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.681: pango_layout_set_width: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.681: pango_layout_set_alignment: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.682: pango_layout_set_text: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.683: pango_layout_set_width: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.684: pango_layout_set_alignment: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.685: pango_layout_set_text: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.686: pango_layout_set_width: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.686: pango_layout_set_alignment: assertion 'layout != NULL' failed

(opencpn:17512): Pango-CRITICAL **: 23:25:48.687: pango_layout_set_text: assertion 'layout != NULL' failed
Segmentation fault
user@bbn:~ $ cat .opencpn/opencpn.log 
23:25:37.658 MESSAGE chart1.cpp:1852 

________

23:25:37.659    INFO logger.cpp:130 ------- OpenCPN version 5.6.0 restarted at 2022-04-11 -------

23:25:37.659    INFO logger.cpp:130 Using loglevel INFO
23:25:37.659 MESSAGE chart1.cpp:1876 wxWidgets version: wxWidgets 3.0.4 Linux 64 bit wxGTK
23:25:37.677 MESSAGE chart1.cpp:1881 OS Version reports as:  5.10
23:25:37.677 MESSAGE chart1.cpp:1883 MemoryStatus:  mem_total: 7760 mb,  mem_initial: 178 mb
23:25:37.677 MESSAGE chart1.cpp:1897 OCPN_OSDetail:  armhf ; Raspbian GNU/Linux ; 10 ; raspbian ; debian
23:25:37.677 MESSAGE chart1.cpp:1904 SData_Locn is /usr/share/opencpn/
23:25:37.677 MESSAGE chart1.cpp:1919 PrivateDataDir is /home/user/.opencpn
23:25:37.678 MESSAGE chart1.cpp:1978 Creating new Config_File: /home/user/.opencpn/opencpn.conf
23:25:37.678 MESSAGE navutil.cpp:1248 winPluginDir, read from ini file: 
23:25:37.678 MESSAGE navutil.cpp:3183 Switch InlandEcdis mode Off
23:25:37.686 MESSAGE styles.cpp:866 Styles loading from /usr/share/opencpn/uidata/styles.xml
23:25:37.691 MESSAGE styles.cpp:843 No styles found at: /home/user/
23:25:37.692 MESSAGE styles.cpp:843 No styles found at: /home/user/.opencpn/
23:25:37.726 MESSAGE styles.cpp:800 Using SVG Icons
23:25:37.727 MESSAGE chart1.cpp:2034 Detected display size (horizontal): 338 mm
23:25:37.727 MESSAGE OCPNPlatform.cpp:913 Getting DefaultSystemLocale...
23:25:37.727 MESSAGE chart1.cpp:2070 System default Language:  en_GB
23:25:37.727 MESSAGE chart1.cpp:2073 Config file language:  
23:25:37.727 MESSAGE chart1.cpp:2078 Adjusted App language:  
23:25:37.727 MESSAGE OCPNPlatform.cpp:994 ChangeLocale: Language load for:  
23:25:37.747 MESSAGE OCPNPlatform.cpp:1047 ChangeLocale: Locale Init OK for:  
23:25:37.747 MESSAGE OCPNPlatform.cpp:1066 Loading catalog for opencpn core.
23:25:37.767 MESSAGE OCPNPlatform.cpp:913 Getting DefaultSystemLocale...
23:25:37.767 MESSAGE chart1.cpp:2085 Opencpn language set to:  
23:25:37.767 MESSAGE ConfigMgr.cpp:610 Creating new Configs catalog: /home/user/.opencpn/Configs/configs.xml
23:25:37.767 MESSAGE ConfigMgr.cpp:631 Loading Configs catalog: /home/user/.opencpn/Configs/configs.xml
[... 621 lines cut ...]
23:25:51.302 MESSAGE OCPNPlatform.cpp:1061 Loading catalog for:  opencpn-chartdldr_pi
23:25:51.322 MESSAGE OCPNPlatform.cpp:1066 Loading catalog for opencpn core.
23:25:51.342 MESSAGE OCPNPlatform.cpp:913 Getting DefaultSystemLocale...
23:25:51.342 MESSAGE pluginmanager.cpp:1055 Loading plugins from dir: /usr/lib/opencpn
23:25:51.342 MESSAGE pluginmanager.cpp:1085 PlugInManager searching for PlugIns in location /usr/lib/opencpn
23:25:51.343 MESSAGE pluginmanager.cpp:1123 Found 19 candidates
23:25:51.343 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libaisradar_pi.so
23:25:51.343 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libotcurrent_pi.so
23:25:51.343 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libgrib_pi.so
23:25:51.343 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libgpsodometer_pi.so
23:25:51.343 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libpolar_pi.so
23:25:51.343 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libengine_dashboard_pi.so
23:25:51.344 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libsquiddio_pi.so
23:25:51.344 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libtwocan_plugin_pi.so
23:25:51.344 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libwindvane_pi.so
[... cut ...]
23:25:51.344 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libwmm_pi.so
23:25:51.344 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libstatusbar_pi.so
23:25:51.344 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libdashboard_tactics_pi.so
23:25:51.344 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libdashboard_pi.so
23:25:51.345 MESSAGE pluginmanager.cpp:1132 Checking plugin candidate: /usr/lib/opencpn/libchartdldr_pi.so
user@bbn:~ $

leamas, 2022-04-11:
It's not enough to remove ~/.opencpn, as you can see it also loads from /usr/lib/opencpn/ if it find anything there.

Not saying that this is the case, but it *might* be that it is loading some plugin from /usr/lib/opencpn and crashes while doing it.

Using the --loglevel debug option might reveal something more.If you do, please just submit the last 20 lines or so.

transmitterdan, 2022-04-11:
When submitting log files it's easier on us if these are attachments rather than pasted text. If pasted text is all one can manage at least tag them as code with the triple back-tick strings so they don't wrap.

mgrouch, 2022-04-11:
Removed all plugin directories. Went into download manager, updated catalogue
installed one plugin and OpenCPN just hang (didn't crash)

```
sudo rm -rf ~/.opencpn /home/user/.local/lib/opencpn/ /usr/lib/opencpn

opencpn -l=debug
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4
libGL error: failed to create dri screen
libGL error: failed to load driver: vc4

(opencpn:19671): Gtk-CRITICAL **: 00:13:49.263: gtk_window_resize: assertion 'height > 0' failed

(opencpn:19671): Gtk-CRITICAL **: 00:13:58.708: gtk_box_gadget_distribute: assertion 'size >= 0' failed in GtkScrollbar

(opencpn:19671): Pango-CRITICAL **: 00:13:59.405: pango_layout_set_width: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.405: pango_layout_set_alignment: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.405: pango_layout_set_text: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.406: pango_layout_set_width: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.406: pango_layout_set_alignment: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.406: pango_layout_set_text: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.407: pango_layout_set_width: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.407: pango_layout_set_alignment: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.407: pango_layout_set_text: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.408: pango_layout_set_width: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.408: pango_layout_set_alignment: assertion 'layout != NULL' failed

(opencpn:19671): Pango-CRITICAL **: 00:13:59.408: pango_layout_set_text: assertion 'layout != NULL' failed
```
[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/8468345/opencpn.log)

mgrouch, 2022-04-11:
Now it just hangs when I try to go into plugins tab

mgrouch, 2022-04-12:
And also

```
00:19:54.708   ERROR file.cpp:264 can't open file '/home/user/.opencpn/mmsitoname.csv' (error 2: No such file or directory)
00:20:07.783   ERROR pluginmanager.cpp:1891 Could not open file "opencpn" for reading with errno = 2.

ls -l /home/user/.opencpn/mmsitoname.csv
-rw-r--r-- 1 user user 60 Apr 12 00:26 /home/user/.opencpn/mmsitoname.csv

```
There file is there though

nohal, 2022-04-12:
The file was obviously created later. Can you explain how these log messages relate to the subject of this issue? I'm again getting somewhat lost in what you talk about.

mgrouch, 2022-04-12:
I'm trying to say that neither version 5.6.0 nor 5.6.1-beta-2 do not work on armhf buster.
PPA versions pop up error on exit.
sloppy versions don't even work with Plugin download manager on armhf buster.

transmitterdan, 2022-04-12:
> And also
> 
> ```
> 00:19:54.708   ERROR file.cpp:264 can't open file '/home/user/.opencpn/mmsitoname.csv' (error 2: No such file or directory)
> 00:20:07.783   ERROR pluginmanager.cpp:1891 Could not open file "opencpn" for reading with errno = 2.
> 
> ls -l /home/user/.opencpn/mmsitoname.csv
> -rw-r--r-- 1 user user 60 Apr 12 00:26 /home/user/.opencpn/mmsitoname.csv
> ```
> 
> The file is there though

The mmsitoname.csv file will be created automatically if it cannot be opened.  This is normal behavior and unlikely to be related to any crash or hang issue.

mgrouch, 2022-04-12:
I got that. What is the last message 

" Could not open file "opencpn" for reading with errno = 2."

This is the case when it crashes going into Plugin Manager.

The case where it says something about mutex is when it is hanging going into plugin manager

Thanks

mgrouch, 2022-04-12:
> Can we please get back to a system where you install only the latest OpenCPN 5.6.1 without miriad of obviously incompatible plugins of unknown versions and origin and try to find out what triggers the subject of this issue, which is an assert on extension of daily track during shutdown? I really do not understand where this drifted, but it is absolutely sure we are not getting any closer to a solution.

That is the case where someone asked me to attach navobj.xml and I did. It fully reproducible to me with PPA version
5.6.1-beta-2 on rpi buster armhf

nohal, 2022-04-12:
> That is the case where someone asked me to attach navobj.xml and I did. It fully reproducible to me with PPA version
5.6.1-beta-2 on rpi buster armhf. All the timestamps look fine to me in that file.

Yes, that someone was me and I am not able to reproduce the issue even with that file, so now please try to guide me through your configuration, starting from vanilla config delivered after installation of the package, adding your customizations one by one until the problem appears.

bdbcat, 2022-04-12:
OK, back on point.  I got it.
The provided navobj.xml file has an improperly formed track point.

```
      <trkpt lat="40.929742333" lon="-74.127831167">
        <time>2022-04-02T09:58:39Z</time>
      </trkpt>
      <trkpt lat="40.929757500" lon="-74.127741167" />

```
You can see from the snippet above that the first point (lat = 40.......2333) is OK.  But the next point (lat=40.....7500) contains no <time> element.  The undefined time string eventually finds its way to an invalid wxDateTime.
How did that invalid track point get written to navobj.xml?  Unknown at this point.  To be tracked down.
But the safe solution now is to simply drop that track point on navobj.xml loading, and carry on.  That fixes this instant fault.  I'll do that in code now.

Dave

bdbcat, 2022-04-12:
Pavel...
Set Breakpoint here:     m_timestring = NULL;

```
void TrackPoint::SetCreateTime(wxString ts) {
  delete[] m_timestring;
  if (ts.Length()) {
    m_timestring = new char[ts.Length() + 1];
    strcpy(m_timestring, ts.mb_str());
  } else
    m_timestring = NULL;
}

```

bdbcat, 2022-04-12:
Close, please?

mgrouch, 2022-04-12:
I see you did sanitise the data on load. However error happens on exit. So if this value get's assigned to null at runtime will it still generate an error on exit?

mgrouch, 2022-04-12:
That's probably why many users had this 'disable plugins dialog' on next start-up even if they closed OpenCPN properly.

LennartG-Sve, 2022-04-12:
mgrouch, another reason for the 'disable plugin dialog' you mention could be the config file not updated properly. See Flyspray #2883 filed yesterday. System hangs, and eventually closes, when clicking the upper right X. Related?

mgrouch, 2022-04-12:
> mgrouch, another reason for the 'disable plugin dialog' you mention could be the config file not updated properly. See Flyspray #2883 filed yesterday. System hangs, and eventually closes, when clicking the upper right X. Related?

Needs to be tested.