OpenCPN/OpenCPN #2609: Rudder angle gauge on dashboard is too big

mgrouch, 2022-04-10:
Rudder angle gauge on dashboard is too big.

Why is it bigger than all others? It shows just one simple angle reading.

Thanks

mgrouch, 2022-04-10:
Thta's on 5.6.1-beta
opencpn_5.6.1+6858+4747344d8_ubuntu18.04.1_armhf.deb

bdbcat, 2022-04-10:
Could we see a screenshot, please?

mgrouch, 2022-04-10:
[image]

Also why plugin toolbar covers engine dashboard?

Thanks

bdbcat, 2022-04-10:
Thanks.
No further GUI changes contemplated for O562.  This issue is deferred

Hakansv, 2022-04-10:
re: Also why plugin toolbar covers engine dashboard?
Because otherwise you shouldn't be able to enter options when dashboards are there.
But if you grab the "hamburger" with the mouse you can move the toolbar panel.