OpenCPN/OpenCPN #2612: debian armhf buster 'backports-sloppy' build against gtx3 errors with Plugin Manager

mgrouch, 2022-04-12:
There are 2 cases

1. Hanging after a download
2. Crashes on going into Plugin Manager settings tab

More details and logs here:

https://github.com/OpenCPN/OpenCPN/issues/2611

Thanks

Hakansv, 2022-04-12:
Related or not.
On my my homebrewed present master on RPI4 Buster armhf O crash when opening Pluginmanager. Not valid picture format. 
A clean "first" use of O. 
See log

[opencpn.log](https://github.com/OpenCPN/OpenCPN/files/8470032/opencpn.log)

leamas, 2022-04-12:
Need to fire up a buster Rpi. It will be a RPi-3, but should be sufficient.

mgrouch, 2022-04-12:
> @Hakansv: Are you saying that it displays a message box "Not valid picture format"? Cannot find anything about picture in the log.
```
WARNING image.cpp:2703 Unknown image data format.
```

mgrouch, 2022-04-12:
Hanging after plugin download case:

```
00:14:22.810   DEBUG threadpsx.cpp:340 pthread_mutex_[timed]lock(): mutex not initialized
00:14:22.810   DEBUG threadpsx.cpp:397 pthread_mutex_unlock(): mutex not initialized.
00:14:22.812   DEBUG threadpsx.cpp:340 pthread_mutex_[timed]lock(): mutex not initialized
00:14:22.812   DEBUG threadpsx.cpp:397 pthread_mutex_unlock(): mutex not initialized.
00:14:22.814   DEBUG threadpsx.cpp:340 pthread_mutex_[timed]lock(): mutex not initialized
00:14:22.814   DEBUG threadpsx.cpp:397 pthread_mutex_unlock(): mutex not initialized.
00:14:22.816   DEBUG threadpsx.cpp:340 pthread_mutex_[timed]lock(): mutex not initialized
00:14:22.816   DEBUG threadpsx.cpp:397 pthread_mutex_unlock(): mutex not initialized.
```

leamas, 2022-04-12:
OTOH, what I do see is that at least shipdriver is linked against gtk2. This reveals some basic errors: we need to handle the gtk3 builds on armhf in the same way as on ubuntu i. e.,  on buster we need separate plugin builds for gtk2 vs gtk3. At a minimum, we should not load plugins linked to gtk2 on a host linked to gtk3.

The loading of gtk2-linked plugins is only an issue on buster, from bullseye everything is gtk3. Which means that this need to be patched in the sloppy build. I'll look into that, and possibly talk to @bdbcat about making a similar patch in the PPA builds. 

There is also a need to update the plugin build chains since it seems that buster will be around on armhf for some time.

@mgrouch: what is your os, exactly? Have you updated it? Have you anything in _/usr/local_,  _~/.opencpn_, _/usr/lib/opencpn_ or _~/.local_ which messes with this?

mgrouch, 2022-04-12:
@leamas 

```
image.cpp:2703 Unknown image data format.
```
Also observed independently by @Hakansv

leamas, 2022-04-12:
It's not that I don't trust you or Håkan. It's just that I can't reproduce it. Have you any idea what the difference could be?

mgrouch, 2022-04-12:
> It's not that I don't trust you or Håkan. It's just that I can't reproduce it. Have you any idea what the difference could be?

@leamas 

Try giving it this opencpn.conf

https://raw.githubusercontent.com/bareboat-necessities/lysmarine_gen/buster/install-scripts/6-desktop/files/opencpn.conf-highres-bbn

I’m not sure what is different. I’m on latest pi4 buster. OpenGL enabled. 

thanks

leamas, 2022-04-12:
> Try giving it this opencpn.conf

will do, stay tuned.

> I’m not sure what is different. I’m on latest pi4 buster. OpenGL enabled.

Vanilla buster or raspbian/buster? Have you updated?

Hakansv, 2022-04-12:
> Hakansv: Are you saying that it displays a message box "Not valid picture format"? Cannot find anything about picture in the log.
@leamas No, no display box, O is immediately killed after click "Plugins".  gdb says only "received SIGFAULT"

leamas, 2022-04-12:
@Hakansv : it might be about plugins linked to gtk2 which are loaded -- this can't work since the main program is using gtk3. I need to make new packages. Stay tuned.

Hakansv, 2022-04-12:
@leamas 
Could be, but plugins enabled in config, like vmm and chartfinder are actually working?
But I also didn't get dependencies well organized on Buster. Could for example be the base for the icons? If you find a proper list I think it would help others a lot.
We could of course let Buster go for Bullseye (32) but I do like Anydesk and it's not yet well linked to that environment.
Thanks
Håkan

leamas, 2022-04-12:
> Could be, but plugins enabled in config, like vmm and chartfinder are actually working?

These are built together with main program, so no problems there.

As for dependencies, any attempt to compile a list will inevitably fail. We need a working control file for buster.  control files are preferred since they can handle alternatives so much better than a plain list.

buster is still 'oldstable' in debian, and we thus need to support for still some time. It's a pain, but a fact.

leamas, 2022-04-12:
@Hakansv @mgrouch : I have created new packages which rightfully does not load any gtk2-linked plugins. Could you test with these packages (both required) so we can see if the issues you have reported are still with us?

The safe route for testing is to download the packages to some place and then:

    $ sudo apt remove opencpn opencpn-data
    $ sudo apt install ./opencpn_5.6.0+dfsg0-1~bpo10+2_armhf.deb ./opencpn-data_5.6.0+dfsg0-1~bpo10+2_all.deb

The packages are available at
  - https://dl.cloudsmith.io/public/alec-leamas/opencpn/deb/debian/pool/buster/main/o/op/opencpn-data_5.6.0+dfsg0-1~bpo10+2/opencpn-data_5.6.0+dfsg0-1~bpo10+2_all.deb
  - https://dl.cloudsmith.io/public/alec-leamas/opencpn/deb/debian/pool/buster/main/o/op/opencpn_5.6.0+dfsg0-1~bpo10+2/opencpn_5.6.0+dfsg0-1~bpo10+2_armhf.deb

leamas, 2022-04-12:
@Hakansv : The dependencies on Buster are different if using gtk3 vs gtk2.  For gtk3,  the file _ci/control_ could be used with the following patch:
```
diff --git a/ci/control b/ci/control
index 074fc5ef..8aa1e2fa 100644
--- a/ci/control
+++ b/ci/control
@@ -22,7 +22,7 @@ Build-Depends: debhelper (>= 9),
  base-files (>=11) | libwxgtk3.0-dev,
  base-files (>=11) | libwxgtk3.0-0v5 | libwxgtk3.0-0,
  libwxgtk3.0-gtk3-dev | base-files (<< 10),
- base-files (>=11) | libwxgtk-webview3.0-dev,
+ libwxgtk-webview3.0-dev | base-files,
  libwxgtk-webview3.0-gtk3-dev | base-files (<< 10),
  libwxsvg-dev | base-files (<< 11),
  portaudio19-dev
```

gtk2 is a completely different beast, I'm not even sure it's possible to build it on a Debian 10 host. BBL

mgrouch, 2022-04-12:
Will Ubuntu PPA buster still be gtk2 and load gtk2 linked plugins via plugin manager?
And sloppy buster will be gtk3 and load gtk3 plugins via plugin manager?

thanks

jongough, 2022-04-13:
How about some consistency in the naming that does not keep on requiring rebuilds of plugins? We seem to be having multiple problems with gtk2/3 and widgets 3.0.2/3.1.2/3.1.5 and the inconsistent use of target names. If the gtk version and widgets version was used in all plugin targets then we would not have this issue. It would also make maintaining the build/deployment simpler.

Hakansv, 2022-04-13:
Alec..
I'm not home for some days and on a terrible connection so maybe not much activity from my side. But to compare these dependencies are used.  Anything explaining the icon failures?
```
apt-cache depends opencpn
opencpn
  Beroende av: libarchive13
  Beroende av: libbz2-1.0
  Beroende av: libc6
  Beroende av: libcairo2
  Beroende av: libcurl4
  Beroende av: libelf1
  Beroende av: libexif12
  Beroende av: libexpat1
  Beroende av: libgcc1
  Beroende av: libgl1
  Beroende av: libglib2.0-0
  Beroende av: liblz4-1
  Beroende av: liblzma5
  Beroende av: libpango-1.0-0
  Beroende av: libpangocairo-1.0-0
  Beroende av: libsqlite3-0
  Beroende av: libstdc++6
  Beroende av: libtinyxml2.6.2v5
  Beroende av: libudev1
  Beroende av: libusb-1.0-0
  Beroende av: libwxbase3.0-0v5
  Beroende av: libwxgtk3.0-0v5
  Beroende av: libx11-6
  Beroende av: zlib1g
  Beroende av: wx3.0-i18n
  Står i konflikt med: <opencpn-plugin-chartdldr>
  Står i konflikt med: <opencpn-plugin-wmm>
  Gör sönder: opencpn-gshhs-crude
  Gör sönder: opencpn-tcdata
  Rekommenderar: xcalib
  Rekommenderar: xdg-utils
  Rekommenderar: opencpn-doc
  Rekommenderar: <opencpn-sglock-amd64>
  Rekommenderar: opencpn-sglock-arm32
  Ersätter: <opencpn-data>
  Ersätter: opencpn-gshhs-crude
  Ersätter: <opencpn-plugin-chartdldr>
    opencpn
  Ersätter: <opencpn-plugin-wmm>
    opencpn
  Ersätter: opencpn-tcdata
pi@RPi4:~/builds/OpenCPN/build $ 
```

leamas, 2022-04-13:
@Hakansv  @mgrouch I have been disconnected over tha day, my ISP had some problems.

I have revised my thinking, and will try to patch the sloppy package to use gtk2, the same as the PPA. It's some tricky, but probably solvable. So, please stay tuned.

leamas, 2022-04-13:
@Hakansv : I see nothing strange in those dependencies, besides the Swedish translation. In particular  breaks -> Gör sönder... took some while to understand. 

Besides that, there is nothing wrong here. We need to fix the linkage before further tests, the linkage could basically lead to anything.

leamas, 2022-04-13:
@jongough : debian-gtk3-armhf is a new ABI. There is no need to rebuild anything, existing ABIs are not affected. The question to add  builds for this new ABI is still open.

leamas, 2022-04-13:
@mgrouch: That question is something I'm working with now.

Hakansv, 2022-04-13:
Alec..
Yes the SE translation is like some early Google. And I haven't so far convinced the RPi to use English as system language but staying in Sweden using a SE keyboard. I'll try again. These SE notes in the command window are not nice.

leamas, 2022-04-13:
> I haven't so far convinced the RPi to use English as system language but staying in Sweden using a SE keyboard.

Works for me. In the initial setup there was a checkbox for using English after selecting the keyboard. But where it is now is another question.

mgrouch, 2022-04-14:
I feel hanging after a plugin download might be related to plugins being compiled with gtk2. while main 'sloppy' app is gtk3
The icons issue still seems there

jongough, 2022-04-14:
If plugin manager needs gtk version and wx version for any ABI then it should, for consistency, be applied to all ABI's. This would avoid the crashing as then plugin manager can check the plugin ABI against the OCPN ABI and as all the information is included there should not be an issue with changes causing plugins to fail.

We are now supporting 2 versions of gtk and 3 versions of wxwidgets in 17 build environments. Each change to the ABI needs a new version number in the builds to help identify what version is being used in case a new ABI, or associated build changes, cause issues on other platforms.

Currently it would appear that linux, windows, macOS and android all handle widgets differently at the event level. This is causing issues trying to get plugins working on all environments consistently when installing/reinstalling/enabling/disabling plugins (it depends on how many and what types of events are handled by the plugin). So knowing what version of the build on what platform helps workout what is going on, particularly with andriod which has a very poor debugging environment for OCPN.

> @jongough : debian-gtk3-armhf is a new ABI. There is no need to rebuild anything, existing ABIs are not affected. The question to add builds for this new ABI is still open.

leamas, 2022-04-14:
@mgrouch:

> I feel hanging after a plugin download might be related to plugins being compiled with gtk2. while main 'sloppy' app is gtk3
> The icons issue still seems there

You're most likely right. I'm working on this, time-constrained and with some surprises down the road. Please stay tuned.

leamas, 2022-04-14:
@jongough 

> So knowing what version of the build on what platform helps 

The target platform is known at build time. There is an unfortunate exception in current code (#2492) -- we are working to get rid of this according to the [roadmap](https://github.com/OpenCPN/OpenCPN/discussions/2502#discussioncomment-1904180).

leamas, 2022-04-14:
Here are at least two independent problems, one with the sloppy backport official package and one with the testplugin toolchain. The root cause is that the the state of the Debian 10/Ubuntu 18 armhf builds has been unclear. 

However, as a fact: The ubuntu-18 armhf build which is sort of  de-facto spec is linked against gtk2. We should all adapt to this.

The sloppy build is linked to gtk3. This is a problem which should be fixed. I have new test builds here, and would appreciate testing. These test builds are linked against gtk2. Test notes:
  - Any gtk3-based plugin will cause an  immediate crash of various kinds. Just remove anything under _~/.local/lib/opencpn_ to get  a fresh start
  - Download : [opencpn-data](https://dl.cloudsmith.io/public/alec-leamas/opencpn/deb/debian/pool/buster/main/o/op/opencpn-data_5.6.0+dfsg0-1~bpo10+2/opencpn-data_5.6.0+dfsg0-1~bpo10+2_all.deb) and [opencpn](https://dl.cloudsmith.io/public/alec-leamas/opencpn/deb/debian/pool/buster/main/o/op/opencpn_5.6.0+dfsg0-1~bpo10+2/opencpn_5.6.0+dfsg0-1~bpo10+2_armhf.deb) to some place.
  - Uninstall using `sudo apt remove opencpn opencpn-data`
  - Install using  `sudo apt install ./opencpn-data_5.6.0+dfsg0-1~bpo10+2_all.deb ./opencpn_5.6.0+dfsg0-1~bpo10+2_armhf.deb`

Note that all plugins does not work, see below. Plugins that does work includes radar and shipdriver.

leamas, 2022-04-14:
The tesplugin toolchain seems linked to gtk3. For example:
```
pi@raspberrypi:~/OpenCPN $ ldd ~/.local/lib/opencpn/libtestplugin_pi.so  | grep wx
        libwx_baseu-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0 (0xb6c20000)
        libwx_gtk3u_core-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_gtk3u_core-3.0.so.0 (0xb6784000)
        libwx_baseu_net-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_baseu_net-3.0.so.0 (0xb673c000)
        libwx_baseu_xml-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_baseu_xml-3.0.so.0 (0xb671f000)
        libwx_gtk3u_html-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_gtk3u_html-3.0.so.0 (0xb6670000)
        libwx_gtk3u_adv-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_gtk3u_adv-3.0.so.0 (0xb6505000)
        libwx_gtk3u_aui-3.0.so.0 => /lib/arm-linux-gnueabihf/libwx_gtk3u_aui-3.0.so.0 (0xb6485000)
```

This will cause problems on ´the PPA build (which is gtk2, correct) but, unfortunately, does work with the current sloppy build. 

Still, it's a bug which should be fixed.

leamas, 2022-04-14:
Downstream Debian bug: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1009701

Hakansv, 2022-04-15:
@leamas 
Following this:
> However, as a fact: The ubuntu-18 armhf build which is sort of de-facto spec is linked against gtk2. We should all adapt to this.

I built using: cmake -DOCPN_TARGET_TUPLE:STRING="ubuntu-armhf;18.04;armhf" ../
But no difference regarding the O crash when enter into Option->Plugins. Exactly the same log entries as the logfile posted [here](https://github.com/OpenCPN/OpenCPN/issues/2612#issuecomment-1096143637)

I'll try the deb linked above tomorrow. 
Håkan

leamas, 2022-04-16:
@hakansv. If you are running into the same troubles as I did (which I think), this all about plugins and plugin compatibility. The plugin compatibility is  assumed to be OK in the PPA package and the release candidate for a new sloppy package [above](https://github.com/OpenCPN/OpenCPN/issues/2612#issuecomment-1099132669).  Things to do:

  - Remove all files under _~/.local/lib/opencpn_
  - Remove possible installed packages opencpn-data and opencpn.
  - Build according to instructions [above](https://github.com/OpenCPN/OpenCPN/issues/2612#issuecomment-1097198857) and install that.
  - Or, install the revised sloppy packages.

I'm unsure about the webview stuff. If it chokes on it, just remove it form the control file. 

You might want to check the package if you build it using (in the  build directory): 

    $ ldd ./opencpn | grep wx.

This will display the wxWidgets libraries linked to opencpn. If the name contains _gtk2u_ it's gtk2, if there is is _gtk3u_ it's gtk3. In this case it should be gtk2/gtk2u

Hakansv, 2022-04-16:
For info after: cmake -DOCPN_TARGET_TUPLE:STRING="ubuntu-armhf;18.04;armhf" ../

```
pi@RPi4:~/builds/OpenCPN/build $ ldd ./opencpn | grep wx
	libwx_gtk2u_gl-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_gl-3.0.so.0 (0xb6aff000)
	libwx_baseu_net-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_baseu_net-3.0.so.0 (0xb6ab7000)
	libwx_baseu_xml-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_baseu_xml-3.0.so.0 (0xb6a9a000)
	libwx_gtk2u_html-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_html-3.0.so.0 (0xb69eb000)
	libwx_gtk2u_adv-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_adv-3.0.so.0 (0xb687c000)
	libwx_gtk2u_aui-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_aui-3.0.so.0 (0xb67fa000)
	libwx_gtk2u_core-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_gtk2u_core-3.0.so.0 (0xb6350000)
	libwx_baseu-3.0.so.0 => /usr/lib/arm-linux-gnueabihf/libwx_baseu-3.0.so.0 (0xb6110000)
pi@RPi4:~/builds/OpenCPN/build $ 

```

leamas, 2022-04-16:
Looks good. Also means that you have the build dependencies in place since before, which means the the control file etc. isn't reguired.

This time, that is.

Hakansv, 2022-04-16:
I'll have a look but since all inbound plugins are built in the same process as OCPN itself the base would be the same?

leamas, 2022-04-16:
No: https://github.com/jongough/testplugin_pi/issues/255

rgleason, 2022-04-16:
I don't fully understand what is going on here, but I don't like the sounds of it.

Hakansv, 2022-04-16:
For a test I checked out commit Tag: Version 5.6.0 and build. No problem opening "Plugins". The log is clean and no icon errors.
So I assume the "Plugins" crash is due to some code change in OCPN after that.
Now it's impractical to bisect all these steps from 5.6.0 to today on a RPi so maybe someone could narrow it down where to find the causing code?

bdbcat, 2022-04-17:
Hakan...
Late comer here.
Please describe build process that creates crashing on O561.

bdbcat, 2022-04-17:
Rick...
Let me try to describe what is going on.  Please correct my "Executive Summary" if I am oversimplifying things, as is my way.
1.  Ubuntu18/Debian 10/armhf normally provides gtk2.
2. Our plugins for Ubuntu18/Debian10/armhf are built for gtk2, and they work fine.
3. Debian 10/"sloppy-backports" upgraded from gtk2 to gtk3.  We think that is a error, and Alec filed a Debian bug saying so.
4. Our plugins built for gtk2 will not work on gtk3, so cause crashes when used with Debian 10/"sloppy-backports".
5. So, we should immediately remove Debian10/gtk2 builds from our master catalog, to avoid user problems.
6. And then we add Debian10/gtk3 plugin builds to the plugin toolchain, for "sloppy".
7. And we add back Debian10/gtk2 plugins, with revised TARGET in the metadata.

Steps 6 and 7 are the only new builds required.  And the timeframe is not particularly urgent, but should be done in due course.  We will hear from (the few) users of Debian 10/armhf, and gauge the urgency from that.  I predict small waves.  Evidence: Not a single user has complained about o-charts_pi on Debianb10/armhf platform.

All in all, this is normal maintenance.  Who would have predicted that "sloppy backports" would have changed the gtk version from 2 to 3?  I cannot imagine that OCPN is the only project caught by surprise at this change. Really, we (devs) are not required nor expected to accurately predict the future.  So, we must do our best, and react to events of the day.

jongough, 2022-04-17:
I have just looked at the xsd in plugins from master and beta and there is no mention of gtk on debian builds for armhf. There is a ubuntu-gtk3-armhf and ubuntu-armhf. Where is the lastest, authoritative, ocpn-plugins.xsd?

jongough, 2022-04-17:
There is a 'deprecated' element in the xsd called 'build-gtk' which 'may' have helped with some of this. Should it be reinstated and used? Should there be a similar element for wxWidgets?

bdbcat, 2022-04-17:
jon...
In master, I see:
`<xs:enumeration value = "debian-armhf"/>
<xs:enumeration value = "debian-gtk3-armhf"/>`

??

bdbcat, 2022-04-17:
"There is a 'deprecated' element in the xsd called 'build-gtk' which 'may' have helped with some of this. Should it be reinstated and used? Should there be a similar element for wxWidgets?"

No.  This would break existing OCPN releases.  OCPN knows how to parse the existing elements only.
That is the nature of legacy code.  In later days, one can always figure out how we "might have done this better".  Bottom line: we need to build compatibly against the current PIM design, which means careful specification of targets.

jongough, 2022-04-17:
So why is the change only in master and not beta and alpha? This makes building and testing very difficult and frustrating.
Why would using a new element break existing releases. Can be added as an additional check if it contains a non-zero length string. It would then be optional until the next release.

We are building incompatibilities all over the place, making assumptions about what is and isn't in a version of an operating system, adding extra text to the target string which is platform and os specific as to whether it contains elements or not. The assumptions are made because it is 'obvious' that the default is xxxx. It is not at all obvious to me what is and isn't in and os and as to whether the target has to specify the 'obvious' or not. XML/XSD is supposed to help with clearly defining information, we seem to be ignoring the benefits whilst accepting the shortcomings. 

This are now 19 different environments/versions within OCPN 5.6.1 plugins. Many of which are not being tested at all until they arrive in production and cause issues like this. I can only test in Ubuntu 18.04,20.0, mswindows 10, macOS catalina, android 8 (my phone), android 10 (via android studio). I can only debug effectively in Ubuntu and Windows, the rest is trial and error which takes considerable time and is proving to be not very effective. We also have to support gtk3,/gtk2 and wxWidgets 3.0.2/3.1.2/3.1.5, and in the later versions of widgets they are deprecating functions that have no alternatives in earlier versions.

We need to stop, take stock, rationalise, and move forward with a supportable set of environments with people that can and will test effectively. Currently it would appear that OCPN master is a test environment not a prod environment and I am not sure why. The latest releases are getting more and more difficult to support in this way. Android being a classic example.

I just don't have time to support this current process.

leamas, 2022-04-17:
@Hakansv @bdbcat:

I have rebuilt, installed and started  OpenCPN from current master at 8f0a262f1 using

    $ cmake -DOCPN_TARGET_TUPLE:STRING="ubuntu-armhf;18.04;armhf" -DOCPN_FORCE_GTK3=OFF ..
    $ make -sj4
    $ sudo make install
    $ sudo apt remove opencpn-data
    $ hash -r

I don't see any particular warnings, everything looks sound at a glance. Using an rpi-4 with 8GB, sadly with 32-bit Raspbian which means that only 4GB is accessible.

@Hakansv: Can you reproduce your problems without any plugins i. e., after removing all files in _~/.local/lib/opencpn_?

leamas, 2022-04-17:
@bdbcat 
> correct me try to describe what is going on. Please correct my "Executive Summary" if I am oversimplifying things, 

The sloppy backport using gtk3 is one part of the problems here. Another is the fact that the plugins for Debian 10/armhf from testplugin was using gtk3 instead of gtk2: https://github.com/jongough/testplugin_pi/issues/255.

This caused interesting testing issues since the faulty testplugin plugins actually worked on sloppy, but this should now be sorted out.

The third, possible issue is then Håkan's. If it is an issue it's by definition unrelated, since it's about current master rather than the 5.6.0 packages.

bdbcat, 2022-04-17:
"So why is the change only in master and not beta and alpha?"
All you have to do is ask. I think you are the only builder using beta and alpha at the moment.

Adding metadata that is "optional until the next release" does not solve our instant problem.  We must support the installed base of OCPN 5.2+.  We know from experience that users are slow to upgrade OCPN core.  We should not require an update to use a specific set of plugins, if it can be at all avoided.

I think of the current scheme this way:  If OCPN is built with an API spec of "banana-cream-pie", and a plugin is built with the same TARGET, then the plugin will be judged compatible, and be loaded.  It really is simple string comparison that controls this.  The XSD spec is simply to validate the plugin build process.

OCPN master branch at HEAD is NOT a prod environment.  It is our development sandbox.  The tagged releases are the prod points.

Android has always been a special case.  The codebase is littered with "#ifdef __OCPN__ANDROID__", and I expect it will be indefinitely.  I have found that it mostly can always be made to work, with some T&E.  It is popular in the market...

Rick may be successful in organizing a trackable plugin test matrix.  It is a worthy goal, and I support it.  But in the meantime , I suggest that you should test the environments that you can, and release everything else to plugin master catalog.  Having a broken plugin in the catalog for a particular platform is not the end of the world.  Also, It is certainly OK for plugins to be in the catalog with platform specific versions.

Finally, if you would like to propose specific PRs to add elements to the metadata, please do so.  They will, as I say, be ignored by OCPN core today.  But maybe in future they could be useful.

bdbcat, 2022-04-17:
XSD updated in Beta and Alpha catalog branches.

rgleason, 2022-04-17:
`This caused interesting testing issues since the faulty testplugin plugins actually worked on sloppy, but this should now be sorted out.`

Thanks for the Executive Summary, additional explanations and "human" sounding dialog in English. I am starting to understand, but am still confused why  testplugin template using gtk3 which actually worked on "sloppy" should be removed so that a new one using gtk2 can be prepared?   ...and then add a gtk3 version?  I know I must have this a bit confused.

Also what the heck is ShipDriver doing through all this?  Wouldn't we have a higher success rate if ShipDriver and Testplugin were equally informed about what is needed?

I do think a comprehensive volunteer test program for the templates (just prior to release) would really help.

bdbcat, 2022-04-17:
"am still confused why testplugin template using gtk3 which actually worked on "sloppy" should be removed so that a new one using gtk2 can be prepared? ...and then add a gtk3 version?"

testplugin built using gtk3 worked, but only on "sloppy".  Did not work on OOB debian10/armhf, since that uses gtk2.  So we must remove it, and then add it back with specific target "debian-gtk3-armhf" for "sloppy".
We also have to rebuild and publish it for gtk2, thereby matching the "debian-armhf" metadata target.

rgleason, 2022-04-17:
Wow, its pretty complicated. So you want me to remove debian 10/armhf   

Which of these is it that needs removing now?  (about 16-17 plugins)
armhf debian buster         (debian 10 I think)
armhf debian bullseye      (debian 11 I think)

This also leads me to the question, is this the list of builds that I should use while looking for and assigning testers?  Can it be shortened somehow reliably?
https://github.com/OpenCPN/plugins/issues/660#issue-1205065025

rgleason, 2022-04-17:
I am also going to make what I view as a simple request, but for some reason it may not be.
Why can't we all use the same naming conventions for the config.yml builds and the metaurls and tarballs, it would really simplify a sea of swirling conventions?

bdbcat, 2022-04-17:
Rick...
Which of these is it that needs removing now? (about 16-17 plugins)
armhf debian buster (debian 10 I think)
armhf debian bullseye (debian 11 I think)

1. Please check and remove any debian/10 (buster) plugin that uses gtk3, and has metadata target "debian-armhf".  These are the ones that will try to load, and probably crash.  To identify them, you will need to look at the build logs and see what version of wxWidgets they are linked against.
2. debian/11 (bullseye) is fine, no action needed.

rgleason, 2022-04-17:
Just to be sure...   aisradar_pi-1.2.21.0-debian-x86_64-10-buster.xml   would not be removed?

rgleason, 2022-04-17:
I did remove debian armhf 10  several days ago.  16 plugins
https://github.com/OpenCPN/plugins/commit/9d14e9b73677bd90f4e18cc92774b75f91defcd7

leamas, 2022-04-17:
>  So we must remove it, and then add it back with specific target "debian-gtk3-armhf" for "sloppy".

No.This is what's happening:

  - I have submitted a new build of sloppy using gtk2. It's currently waiting for some Debian developer for review, blessing and uploading. With some luck, it should not take so long time. The build is available on [mentors](https://mentors.debian.net/package/opencpn/)
  - The testplugin is already fixed so it uses gtk2.
  - The ABI debian-gtk3-armhf is thus not used, defined only for symmetry (and because I rushed away a little early, sorry).

In the end, the basic message is simple: On Debian 10/Ubuntu 18/armhf we always use gtk2. 

Note that Debian 10/Ubuntu 18 are the only versions which actually offers a choice, older versions are gtk2 only and newer versions are gtk3 only.

bdbcat, 2022-04-17:
Just to be sure...
aisradar_pi-1.2.21.0-debian-x86_64-10-buster.xml would not be removed?
celestial_navigation_pi-2.3.13.0-debian-x86_64-10-buster.xml not removed?

Correct.  This tempest is about armhf only.

bdbcat, 2022-04-17:
"I have submitted a new build of sloppy using gtk2. It's currently waiting for some Debian developer for review, blessing and uploading. With some luck, it should not take so long time."
You have confidence that this will be approved?

[30 more replies]

closed 2022-04-23 by mgrouch