OpenCPN/OpenCPN #2614: RFE: Plugin API - allow plugins access to the Show Text button for Squiddio

rgleason, 2022-04-14:
Squiddio when using Opencpn Waypoints has no way to turn off/on the Show Text. for those waypoints.
It is always on, so the redraw is very slow and the screen very cluttered.

(When Squiddio uses ODraw marks it works a lot better, but we are never sure that ODraw is in the environment.)

See https://github.com/jongough/testplugin_pi/issues/253

rgleason, 2022-04-14:
@bdbcat   We understand that changes to the API must occur at specific times and that we just had such a change. Mauro has offered to make a PR to change the API for this purpose, but he is intrepid and might appreciate a couple of suggestions from you on how he can make engineer this PR.  Thanks
@mauroc  Mauro this improvement will be very welcome to users, I believe.

rgleason, 2023-02-28:
I hope it is not too late in the game....

antipole2, 2023-03-03:
API 17 allows display of name to be turned on or off by a plug-in.
Easily done by the JavaScript plug-in so could be done by any other plug-in.

rgleason, 2023-03-03:
Thank you Tony.  I did not realize this had been added so long ago. I will advise Mauro Calvi (Squiddio)  and close this.

closed 2023-03-03 by rgleason