OpenCPN/OpenCPN #2617: Inconsistent sounding sizing between ENC and oeSENC charts

nohal, 2022-04-17:
On the same Retina system with the same settings the sounding sizes are not consistent across vector chart types

oeSENC:
[image]

ENC:
[image]

bcn2, 2022-04-18:
Needs perhaps some more investigation - quilted oeSENC(oesu so actual ones)  and plain S-57 in the Skagrerak - same size
[image]
[image]

bdbcat, 2022-04-20:
This affects only MacOS with Retina display, available on MacBook only.
Corrected by o-charts_pi Version 1.0.15.0, now in master catalog.
Thanks, Pavel.

nohal, 2022-04-20:
Confirmed

closed 2022-04-20 by nohal