OpenCPN/OpenCPN #2618: View submarie cables

Ambrozius, 2022-04-20:
the opeCPN in version 5.6 will not show the underwater cables.
in version 5 it was ok
see the attach file
[image]
[image]

rgleason, 2022-04-20:
What have you done, if anything to turn on underground cables?

Hamburger in lower left. Pick User Display. Then in options > charts > vector charts 

Then pick cable submarine. You have control.

rgleason, 2022-04-20:
Please close this post and post such "questions" in cruisers forum.

rgleason, 2022-04-20:
Sorry, after seeing the screenshots, I guess this is a real issue.

bdbcat, 2022-04-21:
Submarine cables are enabled by Hamburger->Display Category (All or UserStandard) and CheckBox Anchoring Info.
Pix attached.

[image]

CarCode, 2022-04-21:
Yet another obfuscation. Does it never stopp?

leamas, 2022-04-21:
@CarCode  https://github.com/OpenCPN/OpenCPN/issues/2307#issuecomment-909226052, https://github.com/OpenCPN/OpenCPN/issues/2553#issuecomment-1033727729

rgleason, 2022-04-21:
See https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:faq#how_to_show_cables_underground_and_anchoring_information

bcn2, 2022-04-21:
If it is clear, please close the issue.

rgleason, 2022-04-22:
@Ambrozius   It is customary that the OP original person closes the posting.  Please close this now.  Thanks.

closed 2022-04-22 by Ambrozius