OpenCPN/OpenCPN #2620: Dashboard Tactics plugin fails on latest OpenCPN 5.6.2 beta

mgrouch, 2022-04-22:
RPI Buster, opencpn                                5.6.2~ubuntu18.04.1 

```

ASSERT INFO:
../src/aui/framemanager.cpp(1009): assert "Assert failure" failed in AddPane(): A pane with that name already exists in the manager!

BACKTRACE:
[1] wxAuiManager::AddPane(wxWindow*, wxAuiPaneInfo const&)
[2] wxAuiManager::AddPane(wxWindow*, wxAuiPaneInfo const&, wxPoint const&)
[3] PlugInManager::UpdatePlugIns()
[4] PluginPanel::SetEnabled(bool)
[5] PluginPanel::OnPluginEnableToggle(wxCommandEvent&)
[6] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[7] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[8] wxEvtHandler::SearchDynamicEventTable(wxEvent&)
[9] wxEvtHandler::TryHereOnly(wxEvent&)
[10] wxEvtHandler::ProcessEventLocally(wxEvent&)
[11] wxEvtHandler::ProcessEvent(wxEvent&)
[12] wxEvtHandler::SafelyProcessEvent(wxEvent&)

```

rgleason, 2022-04-22:
@mgrouch   Off-topic
Will you be a volunteer plugin tester for RPI Buster,  ubuntu18.04.1?  (I assume "armhf ubuntu buster")
See  https://www.cruisersforum.com/forums/f134/volunteer-plugin-testers-needed-263327.html
Need OS, name, Cruiser Forum name.
Thanks.

Thanks, I will try to get someone who can address that issue.

mgrouch, 2022-04-22:
> @mgrouch Off-topic Will you be a volunteer plugin tester for RPI Buster, ubuntu18.04.1? (I assume "armhf ubuntu buster") See [...]

Yes. I’m in fact testing them already. 
yes rpi buster armhf

rgleason, 2022-04-22:
Great.  It would be good if you could confirm the ones tested that do appear to work ok as well. Thanks!

mgrouch, 2022-04-23:
Ive tested all and reported all issues I’ve found.

mgrouch, 2022-04-23:
Are you killing gtk3 ‘sloppy’ version for buster because you do not want to maintain buster(bionic) gtk2 plugins and gtk3 plugins at same time?

nohal, 2022-04-23:
We do not release anything targetting "sloppy" or "buster" from this project, hence we can't drop it. If something we produce accidentaly works or worked with those things, it was certainly unintentional. Packaging OpenCPN for Debian is a Debian related effort, managed the Debian way using the Debian tools and workflows.
As far as I can tell, everything you talk about recently is related to running debug builds, which simply did not do what you, or anybody else, would expect from a release build on an end user system.

mgrouch, 2022-04-23:
All that matters for the end users is working code. 
In my understanding OpenCPN provides CI scripts in ci directory of the project for various platforms. So in a sense the way OpenCPN is built on various platforms is controlled by core OpenCPN dev team. And it’s a good thing. 

failing asserts is a good thing in testing phase. If assert fails it could mean the program will fail as well.

nohal, 2022-04-23:
Please stop this leading nowhere discussion and close the issues opened for asserts in third party software unrelated to OpenCPN.

closed 2022-04-23 by mgrouch