OpenCPN/OpenCPN #2628: OpenCPN by flatpak does not start in headless systems

sailoog, 2022-04-26:
Several users reported the same issue and I have also been able to reproduce it.

Raspberry Pi 4
Raspberry Pi OS 64bit
OpenCPN (flatpak) Version 5.6.2-0+b75be76 Build 2022-04-22
No monitor connected. Using the default RealVNC server and RealVNC viewer
output:

```
myboat@raspberrypi:~ $ flatpak run org.opencpn.OpenCPN
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.front.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_direct.c:2014:(snd1_pcm_direct_parse_open_conf) The field ipc_gid must be a valid group (create group audio)
18:54:57: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
18:54:57: Debug: Device: 1: sysdefault
18:54:57: Debug: Device: 2: pulse
18:54:57: Debug: Device: 3: speex
18:54:57: Debug: Device: 4: upmix
18:54:57: Debug: Device: 5: vdownmix
18:54:57: Debug: Device: 6: default
18:54:57: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
18:54:57: Debug: Device: 1: sysdefault
18:54:57: Debug: Device: 2: pulse
18:54:57: Debug: Device: 3: speex
18:54:57: Debug: Device: 4: upmix
18:54:57: Debug: Device: 5: vdownmix
18:54:57: Debug: Device: 6: default
18:54:57: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
18:54:57: Debug: Device: 1: sysdefault
18:54:57: Debug: Device: 2: pulse
18:54:57: Debug: Device: 3: speex
18:54:57: Debug: Device: 4: upmix
18:54:57: Debug: Device: 5: vdownmix
18:54:57: Debug: Device: 6: default
18:54:57: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
18:54:57: Debug: Device: 1: sysdefault
18:54:57: Debug: Device: 2: pulse
18:54:57: Debug: Device: 3: speex
18:54:57: Debug: Device: 4: upmix
18:54:57: Debug: Device: 5: vdownmix
18:54:57: Debug: Device: 6: default

(opencpn:2): Gdk-ERROR **: 18:54:58.144: The program 'opencpn' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadAccess (attempt to access private resource denied)'.
  (Details: serial 380 error_code 10 request_code 130 (MIT-SHM) minor_code 1)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the GDK_SYNCHRONIZE environment
   variable to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

```

leamas, 2022-04-26:
Perhaps not really about Flatpak?  I have no Raspbian-64 at hand, but instead  a setup with
  - Rpi-4
  - Fedora 35
  - tigervnc-server
  - OpenCPN 5.6.2 using Flatpak.
  - remmina viewer on my desktop box.

Using this setup OpenCPN seems to at least start up without problems.

FWIW, using tigervnc also works on my rpi-3, but that is traditional .deb packages  so no surprise.

My first guess is that this is about either RealVNC or some more basic problems with  Raspbian/Bullseye/aarch64. Making a test with tigervnc instead of RealVNC should give some interesting input.

leamas, 2022-04-26:
Another possible reason might be the flatpak version. Fedora 35 comes with 1.12.5, Bullseye with  1.10.7. If a test with tigervnc doesn't seem to be conclusive, another possibility is to install flatpak 1.12.7 from bullseye/backports and see if it makes a difference

sailoog, 2022-04-27:
I will try to test that ASAP and report, thanks.

leamas, 2022-05-01:
hm... perhaps this bug will be with us for some time before it can be resolved. In this case I suggest that the title is updated to reflect that this is about headless Flatpak on Raspbian rather than headless Flatpak as such --- after all, Flatpak is actually working on some headless configurations.

sailoog, 2022-05-04:
After some tests I see that I am also getting the same error in non headless systems and randomly. I do not have a pattern yet and the next test will be with a fresh Raspberry OS 32 and 64bit. Title changed.

leamas, 2022-05-04:
This things somewhat reminds me of early troubles we had with Wayland. The first question then becomes if Raspbian is running Wayland or X11. Do

    $ env | grep DISPL
    WAYLAND_DISPLAY=wayland-0
    GNOME_SETUP_DISPLAY=:1
    DISPLAY=:0

If you get output like above, you are on Wayland. However, if there is only DISPLAY:0 you are on X11.

The problem is that the default  window manager on Raspbian is XFCE, and it does not support Wayland. Found [this thread](https://forums.raspberrypi.com/viewtopic.php?t=313685), perhaps it should make sense to try with a modern Ubuntu version instead?

My setup (and, I presume, pretty much anyone using a modern OS) is using Wayland, in my case with GNOME which is default on both Ubuntu and Fedora. GNOME is somewhat resource-hungry, but works fine on a RPI-4, preferably 64 bit so you can use all of the memory.

transmitterdan, 2022-05-04:
I don't think Raspbian runs Wayland.  I may be wrong about that.  Regardless, it's a bridge too far IMO to force Raspbian users to use a non-default WM.  XFCE on X11 systems works ok for native OpenCPN (non-flatpak) on all Ubuntu platforms I have tested (Raspbian on Pi3 up through Ubuntu 21 on Pi4).

leamas, 2022-05-04:
@transmitterdan : indeed. Trying other OS.es would be about trying to triage the problem. I have seen successful use of all kind of platforms, but here and now we only have Fedora...

sailoog, 2022-05-04:
I get only DISPLAY=:0

From Raspberry news (https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/):

One more thing…

Some of you may have heard of Wayland, a proposed replacement for the X Window System which has underpinned most Unix desktop environments for several decades now. Wayland has various advantages over X, notably security and performance, but it is still fairly new technology and hence still under development. A couple of Linux distros now run on top of Wayland, but it hasn’t yet been widely adopted – that said, it is looking as if Wayland is likely to be the future of desktop Linux.

When we released the Bullseye version of Raspberry Pi OS last year, we started using mutter as the default window manager instead of openbox, and one of the reasons for this was that mutter supports the Wayland protocols. In this release, we are making it possible to run the desktop on top of Wayland as an experimental configuration for people who are interested in trying it.

Before going any further, please note – we absolutely do not recommend most people run on top of Wayland yet! This is experimental code and there are many features which are not yet supported under Wayland. (To name a few – taking screenshots, the screen magnifier, any remote desktop application, the screen resolution setting tool; we hope to get some of these working in due course, but for now, they don’t.)

Also, this is not a pure Wayland implementation of the desktop. There are several features of X – notably communication between applications – which are not supported at all in Wayland, which is purely a graphics protocol. The desktop relies on some of these features of X, and so this version of the desktop is a sort of half-way house. The mutter window manager runs as a true Wayland application, but everything else runs under XWayland, which is basically an implementation of X which uses Wayland to draw the graphics. This means that some of the advantages of Wayland will not be available in this version of the desktop.

Nonetheless, after the use of mutter in the Bullseye release, this is another major step in the direction of Wayland, and people who are interested are welcome to try it. (But please don’t complain to us that some feature you depend on doesn’t work under Wayland – as above, we already know that a bunch of stuff doesn’t work, which is why we’ve warned you in advance!)

Switching to Wayland is easy. Launch the raspi-config application from a terminal window using

sudo raspi-config

and under the Advanced Options menu, just select the Wayland option – enable it, and reboot. You shouldn’t notice any obvious difference, but if you want to check if you are really running on Wayland, open a terminal and do

echo $XDG_SESSION_TYPE

Under X, this will return “x11”; under Wayland it will return “wayland”. If you want to go back to X, use the same option in raspi-config to disable Wayland. (Using Wayland shouldn’t change anything permanently on your system, but we do recommend taking a backup before you enable anything experimental like this, just in case.)

For now, Wayland is more something for the curious to play with than anything of interest to most users. But if you are curious, do have a play!
How do I get it?

The new image is available for download from the usual place: [our Downloads page](https://www.raspberrypi.org/downloads/).

To update an existing image, use the usual terminal command:

sudo apt update
sudo apt full-upgrade

The rename-user script will automatically be installed on a desktop image when you update, but it will need to be manually installed on a Lite image, using:

sudo apt install userconf-pi

If you want to install the experimental Wayland support, you will also need to do:

sudo apt install rpi-wayland

As ever, all feedback is welcome, so please leave your comments below.

leamas, 2022-05-04:
Wayland is the default setup in both Debian, Ubuntu, Fedora, Arch  and what not. To say that it is "not widely adopted" is thus plain wrong. If anything is experimental here it is the use of Wayland on Raspbian.

It would be interesting to see if it makes a difference, even it it definitely isn't that mature.

On a sidenote this also tells us why Raspbian probably never will get the same universal role on 64-bits system as it has on the 32-bit ones -- it's just too long behind to offer anything useful to users who already are  running Ubuntu, Fedora and other mainstream distributions.

sailoog, 2022-05-04:
You are probably right, I have just enabled wayland and it bricked my SD.

Said that I have not that impression at all from the OpenPlotter 3 testers. Raspberry OS 64bit is much more used in Raspberry 4 than any other distribution including Ubuntu. I am quite sure that it will have the same numbers than the 32bit version. No feedback yet about OpenPlotter 3 running in non Raspberry OS systems.

leamas, 2022-05-04:
Well, that's OpenPlotter. The world is bigger than that. 

OTOH, we need to understand this better. I have a spare (small) SD card, will try to make some more testing, perhaps tomorrow.  Hopefully I can reproduce your problem

sailoog, 2022-05-04:
Of course, it seems than the world is even bigger than OpenCPN :)

Testing with fresh installation to try to see a pattern.

leamas, 2022-05-05:
I have now a fully updated Raspbian/64 running on a RPi-4/8GB.  Basically, WORKSFORME.

On  a clean system, made a --user installation as described in the [manual](https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:getting_started:opencpn_installation:flatpak). Starting OpenCPN from the menu (under Graphics). Installed  some random plugins including o-charts. And it just works.

Tried to provoke it by running as root, running via vnc (tigervnc), running the vnc server as root,  having both the hardware and vnc display active. Still works.

What I have not done is installing RealVNC. I'm skeptical to closed source software which is not compiled together with the rest of the system, and avoids it if at all possible.

sailoog, 2022-05-05:
I have just done the same test and it crash at startup from the first moment. I will focus my investigation on RealVNC then.

leamas, 2022-05-05:
OK. If you want to try tigervnc instead:
  
    $ sudo apt install tigervnc-standalone-server
    $ tigervncserver

Lot's of documentation out there on how to configure the systemd service. I have not been successful trying to do that. Just starting the server liek this works fine, though.

leamas, 2022-05-05:
Still wondering what this could be about, Made a try with a RPi-3/1GB, just moved the SD card. Still works in the sense that it starts  without problems, no more testing done (actually a bit surprised that  started on  just 1 GB).

So this is is about software. We need to compare our stacks in detail to sort this out, I guess.

sailoog, 2022-05-05:
I have a pattern.

Raspberry 4/8GB
latest Raspberry OS 64bit
RealVNC server

When I boot with no monitor connected and try to run opencpn from a RealVNC client, it crashes.
If I connect a monitor on the fly during the same session, it doesn't crash and maintains this behavior even if I disconnect the HDMI monitor. After booting without the motor connected, it crashes again.

This is the output when crashing and then when it runs, pay attention to the list of detected devices:

myboat@raspberrypi:~ $ flatpak run org.opencpn.OpenCPN
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.front.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_direct.c:2014:(snd1_pcm_direct_parse_open_conf) The field ipc_gid must be a valid group (create group audio)
18:23:30: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
18:23:30: Debug: Device: 1: sysdefault
18:23:30: Debug: Device: 2: pulse
18:23:30: Debug: Device: 3: speex
18:23:30: Debug: Device: 4: upmix
18:23:30: Debug: Device: 5: vdownmix
18:23:30: Debug: Device: 6: default
18:23:30: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
18:23:30: Debug: Device: 1: sysdefault
18:23:30: Debug: Device: 2: pulse
18:23:30: Debug: Device: 3: speex
18:23:30: Debug: Device: 4: upmix
18:23:30: Debug: Device: 5: vdownmix
18:23:30: Debug: Device: 6: default
18:23:30: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
18:23:30: Debug: Device: 1: sysdefault
18:23:30: Debug: Device: 2: pulse
18:23:30: Debug: Device: 3: speex
18:23:30: Debug: Device: 4: upmix
18:23:30: Debug: Device: 5: vdownmix
18:23:30: Debug: Device: 6: default
[... 74 lines cut ...]
18:24:24: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
18:24:24: Debug: Device: 1: vc4-hdmi-0: MAI PCM i2s-hifi-0 (hw:1,0)
18:24:24: Debug: Device: 2: sysdefault
18:24:24: Debug: Device: 3: pulse
18:24:24: Debug: Device: 4: speex
18:24:24: Debug: Device: 5: upmix
18:24:24: Debug: Device: 6: vdownmix
18:24:24: Debug: Device: 7: default
18:24:24: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
18:24:24: Debug: Device: 1: vc4-hdmi-0: MAI PCM i2s-hifi-0 (hw:1,0)
18:24:24: Debug: Device: 2: sysdefault
18:24:24: Debug: Device: 3: pulse
18:24:24: Debug: Device: 4: speex
18:24:24: Debug: Device: 5: upmix
18:24:24: Debug: Device: 6: vdownmix
18:24:24: Debug: Device: 7: default
*** BUG ***
In pixman_region32_init_rect: Invalid rectangle passed
Set a breakpoint on '_pixman_log_error' to debug

----------------------------------------------------------------------------------------------------------------------

Enabling hdmi_force_hotplug in config.txt does not help.
Next test: with tigervnc
Have you tried without a monitor connected?

leamas, 2022-05-05:
> Have you tried without a monitor connected?

Yes. The last test on RPi-3 was also without monitor.

leamas, 2022-05-05:
Just to something similar top compare to (also last Raspbian/64, updated as of today):
  - Moved my SD card to my RPi-4/8GB.
  - Power on
  - Disconnect monitor (attached last time).
  - Log in using ssh
  - Start tigervncserver
  - Connect using Remmina.

OpenCPN starts.

sailoog, 2022-05-05:
Well we have a winner, actually a looser. Tested with TightVNC server everything works even without monitor connected so it seems that RealVNC server is causing this and this is a big problem because it comes installed by default in Raspberry OS and it is integrated in Raspberry GUI.

leamas, 2022-05-05:
Why not tigervnc?

leamas, 2022-05-05:
>  this is a big problem because it comes installed by default in Raspberry OS and it is integrated in Raspberry GUI.

What kind of integration are you referring to here? In a headless system you must start the VNC server at boot time...

sailoog, 2022-05-05:
I mean this integration:

[image]

There is where you enable and disable the RealVNC server in the Raspberry GUI. This GUI is not ready to use any other program. I know you disagree but I think that the official Raspberry OS 64bit is the first option for any owner of a Raspberry 3 and 4 specially for newbies. Instructing them to uninstall RealVNC server, forget that setting and install and configure another program will be a pain.

leamas, 2022-05-05:
>  Instructing them to uninstall RealVNC server, forget that setting and install and configure another program will be a pain.

Yes. It might be necessary to pack this into something simple, a clickable action which does all the housekeeping: uninstalls RealVNC server, installs TighVNC or tigervnc and configures the VNC server so it starts at boot. OTOH, doing this is not *taht* complicated and probably less work than making instructins and handle all those who does not read them.

sailoog, 2022-05-05:
I guess you are thinking of openplotter. Yes, not complicated at all, that is the goal, make things easy for newbies, but this solution has multiple side effects and the most important is that you are breaking a default feature of the host. Beside this, as you said the world is bigger than openplotter and there they will be alone. I will sleep on it and explore all options, thanks.

leamas, 2022-05-05:
OK. over and out :)

leamas, 2022-05-07:
With my opencpn hat on, this looks like a downstream Raspbian bug. This is filed as https://bugs.launchpad.net/raspbian/+bug/1972041.

It could be argued that this should be documented in the user manual. However, besides that it's not much we can do in the opencpn project about this.

There might be  a need to file a new bug against OpenPlotter as well.

This  bug (#2628) could IMHO be closed or kept open blocked on the [Raspbian bug](https://bugs.launchpad.net/raspbian/+bug/1972041)

sailoog, 2022-05-07:
With my OpenPlotter hat on. We had a discussion among OP devs and testers and this is the summary:

- We will not add any feature in OpenPlotter 3 to stop/uninstall RealVNC server in Raspberry OS and install/configure an alternative server in this OS and other possible hosts. Too much intrusive for Raspberry OS and too much complex to maintain for other hosts (ubuntu, debian, mint...) and offer a good support. Virtual desktops should be guaranteed by hosts.
- If this can not be fixed by Raspberry OS or OpenCPN we should live with it adding a warning in docs proposing alternatives to RealVNC or OpenCPN for headless/flatpak systems. Actually we already include AvNav in OpenPlotter which is the most obvious solution for headless systems instead of OpenCPN.
- Obviously this made start again the endless discussion about flatpak as requirement for arm64. Having OpenCPN versions for arm64 in debian, debian backport and PPA sources with no plugins does not make sense at all and it is really confusing. I do not want to start this discussion here again but with the official Raspberry OS 64bit stable in the wild the scenario has changed a lot so we will try to gather usage statistics so that OpenCPN has more info to reconsider fixing this situation.

I am going to close this issue but I am afraid it will appear again sooner or later.

closed 2022-05-07 by sailoog

sailoog, 2023-03-16:
@leamas 

FYI. After lots of boots it seems than this is fixed in O571, no crashes now in bullseye. It works with the experimental Wayland backend enabled or disabled. I do not know where the fix came from, but congrats anyway :)