OpenCPN/OpenCPN #2631: CI workflow to check the pull requests needed

nohal, 2022-04-27:
We really need the CI workflow to check the pull requests back in place.
As it does not seem CircleCI is delivering it any time soon and Travis is gone, we probably have to do it using Github Actions.

leamas, 2022-04-27:
@nohal: I assume you have seen this (?) https://discuss.circleci.com/t/only-build-pull-requests-not-every-branch/200/6?utm_source=google&utm_medium=sem&utm_campaign=sem-google-dg--emea-en-dsa-maxConv-auth-brand&utm_term=g_-_c__dsa_&utm_content=&gclid=Cj0KCQjw06OTBhC_ARIsAAU1yOXJU_2tuYxnbrXkDANLf77K47b0p2hAgyaqXjhoidlHWA__y8dvXYsaArn8EALw_wcB

nohal, 2022-04-27:
So it is possible to do it there already? Why don't we have it then?

nohal, 2022-04-27:
Anyway, Github Actions seem much better for it...

leamas, 2022-04-27:
> So it is possible to do it there already? Why don't we have it then?

The problem is that when it comes to those things "we" basically means you and Dave i. e., those who have permissions on the repo.  

Great to see you have some time to spend here again!

PR #2632 (2022-04-27): Implement Github Actions CI and issue templates
Fixes #2631
commit 53aaedba (2022-04-27): Implement Github Actions CI and issue templates

Fixes #2631
closed 2022-04-27 by nohal