OpenCPN/OpenCPN #2633: RFC: Drop Windows XP support

nohal, 2022-04-27:
We have been the last actively maintained piece of software I'm aware of still supporting XP for several years now and I feel it is not worth the pain to try to keep it for the next release which will be using wxWidgets 3.2

The `v141_xp` toolchain was also dropped from current VisualStudio, so insisting on it blocks using the up to date tools and discourages possible contributors from participating in the project.

In case XP support is still needed going forward, we can get it using MINGW in a way that is probably at most the same painful.

In case I don't hear any strong reasons against it here, I will implement this after wxWidgets 3.2.0 is available mid-2022.

Hakansv, 2022-04-27:
I agree but we've to collaborate this with o-charts_pi and O-charts.org still selling licenses valid for a year to XP-users. "We" have discussed this topic half a year ago so not a big issue as long we prepare it in a proper way I suppose. 

Håkan

nohal, 2022-04-27:
We will effectively not drop the XP support until the next "big" release, which is end of 2022/beginning of 2023. And 5.6.x will still be usable for the current users even after that, so I don't expect this to be a major problem, but of course it has to be taken into account.

leamas, 2022-04-27:
> I. Does Appveyor have a VS2019 image?

Yes: https://www.appveyor.com/docs/windows-images-software/

closed 2023-02-04 by nohal