OpenCPN/OpenCPN #2637: CI:  ubuntu/latest: new version available

leamas, 2022-04-28:
**Describe the bug**

The current CI jobs runs a build labeled ubuntu/latest. This builds focal/20.04. However, these days jammy/22.04 is released and should be used instead in this cycle.

leamas, 2022-04-28:
Not entirely clear whether this is a bug or an RFE, agreed.

nohal, 2022-04-28:
We depend on https://github.com/actions/virtual-environments here, it will take them some time to move "latest" to jammy. Or we have to use Docker inside the VE.

leamas, 2022-04-28:
Right, there is no hurry. Let's keep bug open until they come up with an updated image.

leamas, 2022-04-29:
hm... at a second thought: ubuntu/latest is indeed about the github action script which eventually will be updated.

However, this does not apply to the regular circle CI builds. Here, the version is hardcoded and we will need to replace the focal image with a jammy one manually.

bgercken, 2022-08-03:
So... I was setting up a new dev VM to build/develop OpenCPN on and decided to use the jammy/22.04 release of Ubuntu.

Following the steps in the INSTALL I executed `sudo mk-build-deps -i -r ci/control`.

This effectively down graded my machine to 20.04 downgrading the majority of the core packages (and removing X for some reason). 

Fun. No worries at least now I know why. Cheers.

leamas, 2022-08-03:
> Fun. No worries at least now I know why. Cheers.

Perhaps not ;)

Long story short I cannot reproduce this. Attaching my log in an updated Jammy VM. Furthermore, if you look into _ci/control_ you will see that it's basically just some packages to install with almost no upper bounds on versions (there are some conditions of the kibnd "at least version x.y", but should not cause anything like this.

So, until proven wrong I think it is or was something with your initial setup which caused this, probably combined with approving suggested changes you perhaps should not have accepted.

[install.log.gz](https://github.com/OpenCPN/OpenCPN/files/9255473/install.log.gz)

bgercken, 2022-08-03:
Yes - I agree - I did not pay attention to what it said it would remove. I am a few minutes away from the exact spot - so in the interest of figuring it out - I will use the same step to see what I get. (I was just going to add the packages by hand but this will be worth proving to myself that I was not seeing things.)  :-)

bgercken, 2022-08-03:
OK - it worked perfectly that time. 

Two things that I did different this time were:

1. The last time before running the commands I had installed cmake by hand (which installed all of its dependencies) and g++.
2. I also had installed OpenGL by hand.

I can't see how either of those would cause an issue. 

Sorry for taking up your time as it worked fine this go around.  I will try to recreate it again in another instance but doubt it will fail - too weird. 

Thanks.

bgercken, 2022-08-04:
In the "for what it is worth" category - I was able to repeat the issue. Resulted in packages being removed such that X no longer started and the network settings were erased. (After the VM is rebooted - it is no longer on the network and I had to access it via VNC.)

This was the warning prior to typing `y` to mk-build-deps - so yes definitely packages were removed:

"... 23 upgraded, 122 newly installed, 56 to remove and 191 not upgraded."

Note that I did not `apt update ; apt upgrade` prior to running the commands.

History:

    1  sudo apt-get update
    2  sudo apt-get install openssh-server git
    3  ip a
    4  sudo vi /etc/sudoers.d/wgercken
    5  sudo apt-get install cmake
    6  sudo apt-get install libglu1-mesa-dev freeglut3-dev mesa-common-dev
    7  sudo apt-get install g++
    8  vi main.cpp
    9  g++ main.cpp -o firstOpenGlApp -lglut -lGLU -lGL
    10  mkdir -p src/github/OpenCPN/
    11  cd src/github/OpenCPN/
    12  git clone  https://github.com/OpenCPN/OpenCPN.git
    13  cd OpenCPN/
    14  git fetch
    15  git pull
    16  history
    17  history > /tmp/install.log
    18  sudo apt install devscripts equivs | tee -a /tmp/install.log
    19  sudo mk-build-deps -i -r ci/control | tee -a /tmp/install.log
    20  cd
    21  history >> /tmp/install.log

I made a couple of notes inside the install.log file to orient where step 18 starts. Really not sure why `mk-build-deps` made the decisions that it did. The install.log file only includes command output starting at step 18.

[install.log.gz](https://github.com/OpenCPN/OpenCPN/files/9256350/install.log.gz)
[image]

leamas, 2022-08-04:
Something is fishy with your setup. The stuff removed seems to be dependencies of default-dbus-user-session which, if I get this right, is a virtual package provided by dbus-user-session. 

Basically, this boils down to "what is the difference between your system and mine?" . Well, not only mine, you are not the only one out there building OpenCPN -- if this was a general problem we should have had more reports. So, questions:

  - Is this an upgraded Focal box or a clean install?
  - I have dbus-user-session installed, have you?
  - Have you changed anything in /etc/apt/sources.list et. al.?
  - What happens if you first do `apt upgrade`?

bgercken, 2022-08-04:
Good Morning, first thank you for looking into this for me, I appreciate your time and attention.

At this point this is not a problem as I was already able to re-install performing the steps in the correct order and everything worked fine.

Here are the answers to your questions:

1. This was a fresh install using: ubuntu-22.04-desktop-amd64.iso
2. In the bad VM `dbus-user-session` is one of the packages that was marked for removal so it is missing when I look for it after the fact. In a new working VM it is installed as expected: `dbus-user-session/jammy,now 1.12.20-2ubuntu4 amd64 [installed]`.
3. The history above was exactly all of the steps performed in the VM - no changes were made to the base install.
4.  If I do things in the correct order without steps 5 through 9 above everything installs correctly. 

The only additional step I had to perform was to complete the OpenGL install to add the GLU libs.

Yes - your BTW foot note was the real root of the cause - since I am dealing with VMs I tend to be more experimental when installing as you can reinstall in short order. This time I should have listened to the warning.

Again, thank you very much for your time. I still have the virtual disk for the bad install if you need me to look at anything - otherwise I am good to go and appreciate the support!

leamas, 2023-02-04:
we have updated the CI builds. Closing

closed 2023-02-04 by leamas