OpenCPN/OpenCPN #2643: Duplicate text boxes being rendered

MSBGit, 2022-05-03:
**Describe the bug**
When mousing over the piano keys, two identical (except for font size) text boxes are rendered, cluttering the display.

**To Reproduce**
Steps to reproduce the behavior:
1. Mouse over a piano key
2. Observe the two text boxes that pop up (unless you are too far over to the right side of the display in which case you will see only a partial second text box being rendered or no second text box).

**Expected behavior**
Only one text box is rendered.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information if applicable):**
 - OS: macOS Monterey 12.3.1
 - Version OpenCPN 5.6.2

**Additional context**
I am running on an M1 MacBook Pro with two 4K displays connected.
[image]
[image]

nohal, 2022-05-03:
It's been like this for as long as I can remember, just the positioning is completely different and not annoying for me
[image]

MSBGit, 2022-05-03:
@nohal do you see the duplicate text boxes on your setup?

nohal, 2022-05-03:
As you can see on my screenshot I do and have for years. Just they are positioned completely dofferently than for you, which very likely has to do with you system with multiple HiDPI dosplays connected.

MSBGit, 2022-05-03:
Your screenshot doesn't show multiple identical text boxes, it just shows the one. Did you mean to take a larger screenshot to post here?

Anyway, I think this is a bug, and may be due to OpenCPN just not handling high DPI displays as you say (and this might be causing other UI bugs on 4K displays on the Mac that I am reporting). I am reporting these because I really want to use OpenCPN on my 4K display on my Mac so am hoping the dev team will take a look at better supporting these types of displays.

nohal, 2022-05-03:
Look to the upper left corner of that textbox on my screenshot, I'm sure if you concentrate, you will actually see there is the second copy of it hidden behind and only slightly visible, hence me saying the positioning is different on different system (in my case a Retina Macbook without any external monitors).
The root cause of all the issues you are reporting is problems with HiDPI support, mostly in wxWidgets itself, but also in our own code working those wxWidgets issues around, especially in environments with mixed HiDPI and lowDPI displays, even worse when those systems are Apple and some of the displays are Retina. We do know abou these and they are unfortunately extremely hard to address without access to such systems.

MSBGit, 2022-05-03:
OK I see the other window in your screenshot, it's not nearly as obvious as what I see on my 4K display.

I'm happy to help testing any changes the team wants to make to support HiDPI and multiple/mixed displays.

nohal, 2023-02-02:
Should be fixed, please reopen if still present with https://dl.cloudsmith.io/public/david-register/opencpn-unstable/raw/files/OpenCPN_5.7.1.pkg

closed 2023-02-02 by nohal