OpenCPN/OpenCPN #2644: Text in Options windows doesn't render properly

MSBGit, 2022-05-03:
**Describe the bug**
Text in various widgets in the options window is not correctly rendered in  the widgets in that window. It looks like the text is a bit too large for the size of the widget that the text is being rendered in. This occurs in all of the tabs in the _Options_ window.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the _Options_ window (the gear).
2. Observe the text rendering issue.

**Expected behavior**
The text is properly rendered inside the associated widget. In the case of the _Sample_ text, that word should be properly vertically aligned with the _Choose Font..._ widget.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: macOS Monterey 12.3.1
 - Version: OpenCPN 5.6.2
 - Hardware: M1 MacBook Pro
 - Two 4K external displays connected to the MBP

nohal, 2022-05-03:
There is probably nothing we can do about this, seems like a wxWidgets limitation, especially with multiple monitors.

MSBGit, 2022-05-03:
@nohal is there a recommended platform/OS to run OpenCPN on that doesn't exhibit these strange UI issues?

nohal, 2022-05-03:
Any, without multiple monitors with different DPI mixed in a single system, as far as I can tell (As someone not having any system with multiple monitors with different DPI and not personally experiencing your issues).

MSBGit, 2022-05-03:
Thanks. There are other cross platform open source projects that have been successful at supporting this type of display configuration (KiCad, VLC for example) so it's technically possible. Is it on the roadmap for OpenCPN to do some work in this area, or is my use case pretty niche and not worth putting effort into supporting?

nohal, 2022-05-03:
Your use case is pretty niche, but it is worth putting effort into supporting. Will I do it personally? No, not without access to a multi monitor system, because without access to such a system doing anything about it os virtuallly imposisble.