OpenCPN/OpenCPN #2645: Chart sections missing from chart display

MSBGit, 2022-05-03:
**Describe the bug**
Parts of the chart that should be visible are not visible.

**To Reproduce**
Steps to reproduce the behavior:
Unfortunately I can't determine a reliable and consistent way to reproduce this issue. It occurs when clicking around on the piano keys.

**Expected behavior**
The chart should be completely rendered mont he screen without sections missing.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: macOS Monterey 12.3.1
- Version: OpenCPN 5.6.2
- Hardware: M1 MacBook Pro
- Two 4K external displays connected to the MBP

**Additional context**
In the screenshot, I have a track loaded and the track is displayed into the area where there is no chart rendered. If I zoom around that section of the chart will properly display at some point, but again the operations of the chart function is very unclear to me and I can not understand why I wouldn't see sections of a chart that I have loaded into OpenCPN.

nohal, 2022-05-03:
Which chart exactly is displayed?

nohal, 2022-05-03:
If you are actually asking why you don't see the chart cell to the north of the one being displayed as well, it is because you have disabled quilting and are in single chart mode, to switch back, press `q` or use the hamburger icon in the lower right corner to invoke the chart canvas options menu and turn quilting back on.

rgleason, 2022-05-03:
Added to FAQ
https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:faq#why_are_chart_sections_missing_from_the_display

OP should close this now, if Pavel has provided an answer.

MSBGit, 2022-05-03:
Thanks that resolves this. My mistake, I usually leave chart quilting on. The FAQ section on this is clear.

closed 2022-05-03 by MSBGit