OpenCPN/OpenCPN #2646: chart1.cpp has some merge cruft

transmitterdan, 2022-05-03:
Around line 7900:

```

<<<<<<< HEAD
    //    Search the no-show array
    bool b_is_in_noshow = false;
    for( unsigned int i = 0; i < g_quilt_noshow_index_array.size(); i++ ) {
        if( g_quilt_noshow_index_array[i] == selected_dbIndex ) // chart is in the noshow list
=======
    else
        if( index < pCurrentStack->nEntry ) {
            if( ( ChartData->GetCSChartType( pCurrentStack, index ) == CHART_TYPE_KAP )
                    || ( ChartData->GetCSChartType( pCurrentStack, index ) == CHART_TYPE_GEO )
                    || ( ChartData->GetCSChartType( pCurrentStack, index ) == CHART_TYPE_PLUGIN ) ) {
                ChartBase *new_pThumbChart = ChartData->OpenChartFromStack( pCurrentStack, index );
                if( new_pThumbChart )         // chart opened ok
>>>>>>> multicanvas
```

nohal, 2022-05-03:
Been there since MUI merge in 2018, will better leave it for @bdbcat

transmitterdan, 2022-05-03:
It's old, ok.

nohal, 2022-05-03:
I would not close it, it is there for a long time, but still is an issue. The only question is whether the correct solution it to just remove the part doing nothing or the logic needs a bigger fix, at which Dave has the best chance to see without some more extensive study.

bdbcat, 2022-05-03:
As you can see, this cruft is in a larger block that is "#ifdef 0".  That block relates to chart thumbnails, which we no longer support  since O5.0.  Thumbnails are really only useful in single-chart mode.  And no-one has mentioned (or even noticed) their absence.  So, I propose to remove all references to thumbnails, if no-one objects.
If we really want them, the code should be refactored into ChartCanvas, since we now can have two distinct instances of ChartCanvas, each with piano bar and associated thumbnails.  That is the reason for the TODO comment at MyFrame::SetChartThumbnail(int index)
Thoughts?
Dave

leamas, 2022-05-03:
There is a similar situation in _src/datastream.cpp_ and _src/glChartCanvas.cpp_ : Large `#ifdef 0`  blocks containing git markers i. e., not fully rebased.

IMHO, the habit of leaving deleted code in place in the src should be  a temporary work state that should be removed, if not earlier so when we approach release. After all, it's still available in git.

leamas, 2022-05-03:
> If we really want them, the code should be refactored into ChartCanvas

Whether this should be done or not I still see no reason to keep the `ifdef 0` block. I think it should be removed, preferably in a single commit with a descent description  so it could be found.

Left in code could be a short comment that the thumbnail code is removed, and perhaps about the possible re-factoring into ChartCanvas. That is enough to easily find it in the git history.

leamas, 2022-05-03:
BTW: 

    $ git grep -l '#if 0' include 
    include/GL/gl_private.h
    include/iENCToolbar.h
    include/s52s57.h
    include/viewport.h
    $  git grep -l '#if 0' src | wc -l
    46

i. e., there is one or more `#if 0` in 46 source files and also some headers. Time to clean up?

bdbcat, 2022-05-03:
Yes, time.
Most (all?) of these are mine, something of an _aide-mémoire_
I'll work on that, slowly.

leamas, 2022-05-03:
Right. Now, this bug has been partly hijacked by me, bringing in other files than _chart1.cpp_ which is the core issue here. Let's keep it that way and close bug as soon as _chart1.cpp_ is taken care of. If there is a need to track the other files it could be done in a separate bug

commit 36c54f1b (2022-05-05): Remove deprecated chart thumbnail management in MyFrame.  Closes #2646
closed 2022-05-05 by bdbcat