OpenCPN/OpenCPN #2647: Different chart detail displayed in different chart sections

MSBGit, 2022-05-03:
**Describe the bug**
Different parts of the displayed chart display different levels of detail.

**To Reproduce**
Steps to reproduce the behavior:
I don't know how to reliably and consistently reproduce this behavior, but it occurs often. Clicking randomly on the piano keys seems to cause it to occur and also causes it to not occur (which means the chart displays the same level of detail throughout the complete displayed chart).

**Expected behavior**
The complete displayed chart displays the same level of detail.

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
- OS: macOS Monterey 12.3.1
- Version: OpenCPN 5.6.2
- Hardware: M1 MacBook Pro
- Two 4K external displays connected to the MBP

**Additional context**
I don't understand if this is a bug or a design feature. If it's a design feature, how is this useful to the user?

nohal, 2022-05-03:
This is a design feature and why is this usefull to the user you will see as soon as you go to the toolbox, move the vector slider for chart zoom/scale weighting to the right and will be in an area with many charts and on a machine you did not buy in the past 12 month for 2000USD.

closed 2022-05-03 by nohal

MSBGit, 2022-05-03:
Got it. I can see on a machine like the RPI that this makes sense. That is my use case not he boat, I am only using my MBP at home to do waypoint and track/route work.