OpenCPN/OpenCPN #2653: RFE: Dashboard Instruments need Graphic improvements and integration.

rgleason, 2022-05-03:
**Is your feature request related to a problem? Please describe.**
The Dashboard Instruments look very dated and with better design will occupy less screen space.

**Describe the solution you'd like**
A graphic interface designer to improve the Dashboard Instrument displays.
This would improve OpenCPN's overall "look" and "utility" considerably.

**Describe alternatives you've considered**
Users are different regarding needs, perhaps there should be alternative styling "skins" that can be selected.

**Additional context**
There must be some good examples of this that can be used.

rgleason, 2022-05-05:
There is hope. Pavel has some work in progress on this (before I even posted, unknowing)!
https://github.com/nohal/dashboardsk_pi/issues

bc38, 2022-05-27:
At the moment of developing a new dashboard, I want to propose a concept for the design of tools in the form of widgets.
The overall style retains the new CPN graphic style
The appearance of the tools is based on the latest modern trends and rules of OpenBridge.
See PDF file
If you like something, we can continue
waiting for your comments

[CPN_New_dashboard_concept.pdf](https://github.com/OpenCPN/OpenCPN/files/8784706/CPN_New_dashboard_concept.pdf)

rgleason, 2022-05-27:
I think this is an big improvement. I am sure there are some hurdles and special situations. How flexible is it to allow some user adjustments? How is this being done?

ozolli, 2022-05-30:
> The appearance of the tools is based on the latest modern trends and rules of OpenBridge.
> See PDF file
> If you like something, we can continue

This is very nice!
I like the B&G SailSteer like dial on the top right corner.

rgleason, 2022-05-30:
[How OpenBridge seeks to improve maritime workplaces](https://www.marine-pilots.com/articles/17173-how-openbridge-seeks-to-improve-maritime-workplaces)
 Prof. Kjetil Nordby Institute of Design - The Oslo School of Architecture and Design
[Major upgrade of "OpenBridge Design Guideline" released
](https://www.marine-pilots.com/articles/121198-major-upgrade-of-openbridge-design-guideline-released#comments)

http://openbridge.no/index.html#Content

ozolli, 2022-05-30:
> Are these terms considered "Opensource"? I think so.
> https://openbridge-ds.webflow.io/

The Apache License is Opensource.

rgleason, 2022-05-30:
I registered (free) and answered the questions, then was able to look around. There is a figma interface that is used to contribute, and it is on GitLab  https://gitlab.com/openbridge/openbridge-web-components
[image]

rgleason, 2022-05-30:
Unfortunately the development activity stopped in Jan 2022

rgleason, 2022-12-17:
Pavel [Nohal] has created a new plugin **DashboardSK**  https://opencpn-manuals.github.io/main/dashboardsk/0.1/index.html
which is based on [SignalK Specification](https://signalk.org/specification/1.7.0/doc/index.html)  allowing greater flexibility in configuration and  [Alerts Alarms and Notifications](https://signalk.org/specification/1.7.0/doc/notifications.html)  also [Histograms](https://opencpn-manuals.github.io/main/dashboardsk/0.1/simplehistograminstrument.html).

Thank you Pavel!

rgleason, 2022-12-17:
Perhaps it will evenutally develop into something like openbridge as [bc38 posted](https://github.com/OpenCPN/OpenCPN/issues/2653#issuecomment-1139349321)
[CPN_New_dashboard_concept.pdf](https://github.com/OpenCPN/OpenCPN/files/8784706/CPN_New_dashboard_concept.pdf)

rgleason, 2022-12-17:
Also, Thank you to Bdbcat, Leamas and all the others who worked on signalK!

closed 2022-12-17 by rgleason