OpenCPN/OpenCPN #2657: [Translations] Debian build chain diagnostics

leamas, 2022-05-04:
**Describe the bug**
The tr_TR Turkish local contains two extremely long lines with the following diagnostics from Debian build chain.
```
plugins/wmm_pi/po/tr_TR.po line 33 is 518 characters long (>512)
po/opencpn_tr_TR.po line 3383 is 521 characters long (>512)
```
Also, further checks available in https://i18n.debian.org/l10n-pkg-status/o/opencpn.html:
```
gettext: po/opencpn_hi_IN.po:7013: 'msgid' and 'msgstr' entries do not both end with '\n'
msgfmt: found 1 fatal error
gettext: po/opencpn_ko_KR.po:4820: 'msgid' and 'msgstr' entries do not both end with '\n'
msgfmt: found 1 fatal error
gettext: po/opencpn_th_TH.po:41: 'msgid' and 'msgstr' entries do not both begin with '\n'
msgfmt: found 2 fatal errors
```

All of these seems like simple oversights in the translation process.

**To Reproduce**

Rebuild Debian package using something like

    $ git clone https://gitlab.com/leamas/opencpn.git
    $ sudo mk-build-deps -i 
    $ git clean -fxd
    $ debuild -us -uc

And/or check package status as of above.

**Expected behavior**
Clean translation

**Additional context**
Admittedly, this is nit-picking.

Hakansv, 2022-05-04:
The below notes I've corrected in Crowdin.  (To my best understanding; non-Latin characters are not easy but this was all about "new line" symbols incorporated in the text to translate) 
So next Crowdin build and po-files download will hopefully solve this issue
Håkan

>> OK >> po/opencpn_tr_TR.po line 3383 is 521 characters long (>512)  

>>OK>> plugins/wmm_pi/po/tr_TR.po line 33 is 518 characters long (>512)

>>OK>gettext: po/opencpn_hi_IN.po:7013: 'msgid' and 'msgstr' entries do not both end with '\n'
msgfmt: found 1 fatal error

>>OK>>gettext: po/opencpn_ko_KR.po:4820: 'msgid' and 'msgstr' entries do not both end with '\n'
msgfmt: found 1 fatal error

>>OK>>gettext: po/opencpn_th_TH.po:41: 'msgid' and 'msgstr' entries do not both begin with '\n'
msgfmt: found 2 fatal errors

leamas, 2022-05-04:
Great! I presume that this includes breaking the long line in _plugins/wmm_pi/po/tr_TR.po_ into shorter ones?

leamas, 2022-05-04:
Whatever. There will always be these kinds of diagnostics. Let's close this one, possibly open a new after next po-files download.

@Hakansv : thanks!

closed 2022-05-04 by leamas

Hakansv, 2022-05-04:
Long lines...
Yes there was some "misspelled" backslash-n i corrected.

leamas, 2022-05-04:
There was also a line which just was missing `\n`  in _plugins/wmm_pi/po/tr_TR.po_. I', pretty sure one need to split that single string into several smaller, one for each line.  And add a trailing `\n` to each and every, this is just how translations are done.

Could be wrong, for sure.

Hakansv, 2022-05-04:
I did a correction to "plugins/wmm_pi/po/tr_TR.po line 33" Is that the same you mentioned now or is there another one?

leamas, 2022-05-04:
yes, that's  the one