OpenCPN/OpenCPN #2661: RFE: Prioritization of plugin overlays, allow plugins to render on the very top of the canvas.

nohal, 2022-05-04:
Currently the plugin overlays are drawn on top of the chart, but under some objects managed ny the core application (AIS targets, ownship, Depth unit indicator). This is perfectly OK for most of the currently existing overlays like radar or grib, but becomes disruptive for overlays like dashboard instruments.
We should offer an option for a plugin to render overlay on the very top layer.
[image]

bdbcat, 2022-05-05:
What are those instruments on your miniature screenshot?

nohal, 2022-05-05:
https://github.com/nohal/dashboardsk_pi - new dashboard plugin in the making

bdbcat, 2022-05-05:
What API are you using to render the new instruments?
Any suggestions on how to modify the API to your needs?

nohal, 2022-05-05:
The standard overlay callbacks - `RenderOverlayMultiCanvas` and `RenderGLOverlayMultiCanvas`

I think that the solution could perhaps look something like this

```c++

// Some predefined arbitrary values, void spaces allow sane plugin devs to arrange priorities between them
enum PI_OVERLAY_PRIO {
  PI_OVERLAY_PRIO_ON_TOP = 1,
  PI_OVERLAY_PRIO_BELLOW_VESSELS = 32,
  PI_OVERLAY_PRIO_BELLOW_NAVOBJECTS = 64,
  PI_OVERLAY_PRIO_OLD_API = 192,
  PI_OVERLAY_PRIO_BELLOW_ALL = 255
}

std::multimap<uint8_t, opencpn_plugin*> g_plugins_with_overlay;

// Plugins looking for callbacks register themselves using this API function
// The older plugins with WANT_*OVERLAY_CALLBACK would be registered when instantiated with PI_OVERLAY_PRIO_OLD_API priority and rendered in the same place as now
void RegisterPluginOverlay(opencpn_plugin* plugin, uint8_t prio) {
  g_plugins_with_overlay.emplace(std::make_pair(prio, plugin));
}

// Called anywhere we want the plugins to be able to insert overlay with corresponding priority range
void RenderOverlays(uint8_t prio_from, uint8_t prio_to) {
  for (const auto & plugin : g_plugins_with_overlay) {
    if (plugin.first >= prio_from && plugin.first =< prio_to) {
      ((opencpn_plugin*)plugin.second)-> RenderGLOverlayMultiCanvas(...); // Here perhaps adding a new parameter sending back the prio value would allow a single plugin to provide multiple different overlays at different layers
    }
  }
}
```

bdbcat, 2022-05-06:
OK in theory.  By NAVOBJS, we mean routes, tracks, and marks, I assume.
Thinking...

leamas, 2022-05-09:
Notes on implementation sketch above:

`enum PI_OVERLAY_PRIO`: C++11 has class Enum, here perhaps something like `enum class  PI_OVERLAY_PRIO { ON_TOP = 1,  BELOW_VESSELS = 32 , ...};` Enum classes come with type checking, lesser namespace impact etc.

`void RegisterPluginOverlay`and  `void RenderOverlays()` could use the event mechanism pending in #2607. This basically boils down to that each plugin defines an event, uses a Listener to wait for events and registers a callback executed on that event. See [observable.h](https://github.com/OpenCPN/OpenCPN/blob/482640b2c47c5f643c8bfbd93c27c027b3f517de/include/observable.h), around line 100 for details.

Code that want to trigger plugin actions just  uses  `notify()`. 

There are overall advantages using a common mechanism for this kind of stuff. In particular, the events code  handles the situation when a plugin goes out of scope (reloaded, uninstalled, etc.) and leaves a dangling reference in  `g_plugins_with_overlay`.

nohal, 2022-05-09:
Enum classes are perfect in most cases, but in this particular one using it would make us lose that trick allowing plugins to arange the preferences without the core even knowing about it that using `uin_t` priority allows.
And switching to the new mechanism based on events would not be compatible with the current plugins.

I would say we probably do want to revise the plugin API and will probably hardly find a better moment than the global switch to wx3.2, but it will be a slightly bigger operation than I want to cover in this issue.

leamas, 2022-05-09:
Fair enough, agreed.

closed 2022-11-30 by nohal