OpenCPN/OpenCPN #2672: 5.6.2 macOS 12.3. Random "not responsive" after Screen Recording permission action

jeronimo, 2022-05-07:
**Describe the bug**
Gets unresponsive after doing something with Screen Recording permissions (tried to deny, allow or allow only on itself), and why is it needed anyway?

**Screenshots**
[image]

**Desktop (please complete the following information if applicable):**
 - OS: macOS
 - Version 12.3

**Additional context**
```
Date/Time:        2022-05-07 07:14:10.436 -0600
End time:         2022-05-07 07:30:50.653 -0600
OS Version:       macOS 12.3 (Build 21E230)
Architecture:     arm64e
Report Version:   35
Incident Identifier: 168595A9-E9B0-47C1-8787-5C847192CBFF
Share With Devs:  Yes

Data Source:      Stackshots
Shared Cache:     00EDBAF8-9281-3F26-B94F-E6A315E3A659 slid base address 0x1a39c8000, slide 0x239c8000
Shared Cache:     CB70EAD7-3266-30B7-BECD-BD97E7998A2A slid base address 0x1a59cc000, slide 0x259cc000
Shared Cache:     71741BEE-0ABC-3E06-B329-770CAC959B0C slid base address 0x7ff80ebfc000, slide 0xebfc000

Command:          OpenCPN
Path:             /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:       org.opencpn
Version:          5.6.2 ()
Architecture:     x86_64 (translated)
Parent:           launchd [1]
PID:              65696
Time Since Fork:  1015s

Event:            hang
Duration:         1000.22s
Duration Sampled: 4.30s (process was unresponsive for 996 seconds before sampling)
Steps:            43 (100ms sampling interval)

Hardware model:   MacBookPro18,3
Active cpus:      8
HW page size:     16384
VM page size:     16384

Time Awake Since Boot: 3369206s
Time Since Wake:  30394s

Fan speed:        2310 rpm
Total CPU Time:   5.445s (14.8G cycles, 84.4G instructions, 0.18c/i)
Advisory levels:  Battery -> 3, User -> 2, ThermalPressure -> 0, Combined -> 2
Free disk space:  27.54 GB/460.43 GB, low space threshold 3072 MB

--------------------------------------------------
Timeline format: stacks are sorted chronologically
Use -i and -heavy to re-report with count sorting
--------------------------------------------------

Heaviest stack for the main thread of the target process:
  42  start + 462 (dyld + 21790) [0x205d0c51e]
  42  main + 20 (OpenCPN + 868500) [0x104e49094]
  42  wxEntry(int&, wchar_t**) + 56 (libwx_baseu-3.1.5.0.0.dylib + 412984) [0x10fc4bd38]
  42  wxApp::OnRun() + 26 (libwx_osx_cocoau_core-3.1.5.0.0.dylib + 349162) [0x1100e83ea]
  42  wxAppConsoleBase::MainLoop() + 116 (libwx_baseu-3.1.5.0.0.dylib + 41572) [0x10fbf1264]
  42  wxEventLoopBase::Run() + 85 (libwx_baseu-3.1.5.0.0.dylib + 206325) [0x10fc195f5]
  42  wxCFEventLoop::DoRun() + 29 (libwx_baseu-3.1.5.0.0.dylib + 880525) [0x10fcbdf8d]
  42  wxGUIEventLoop::OSXDoRun() + 186 (libwx_osx_cocoau_core-3.1.5.0.0.dylib + 769386) [0x11014ed6a]
  42  -[NSApplication run] + 586 (AppKit + 193337) [0x7ff811930339]
  42  -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394 (AppKit + 248970) [0x7ff81193dc8a]
  42  _DPSNextEvent + 927 (AppKit + 255437) [0x7ff81193f5cd]
  42  _BlockUntilNextEventMatchingListInModeWithFilter + 70 (HIToolbox + 194533) [0x7ff8180d77e5]
[... 375 lines cut ...]
  <thread QoS default (requested default), process unclamped, process received importance donation from distnoted [518], process received importance donation from 1Password Extension Helper [85239] [unique pid 283984], IO tier 0>
  43  thread_start + 8 (libsystem_pthread.dylib + 8332) [0x1a3ced08c] 1-43
    43  _pthread_start + 148 (libsystem_pthread.dylib + 29292) [0x1a3cf226c] 1-43
      43  __NSThread__start__ + 808 (Foundation + 378652) [0x1a4c9171c] 1-43
        43  ??? (CFNetwork + 2533708) [0x1a8ba594c] 1-43
          43  CFRunLoopRunSpecific + 600 (CoreFoundation + 533648) [0x1a3dbc490] 1-43
            43  __CFRunLoopRun + 1180 (CoreFoundation + 536780) [0x1a3dbd0cc] 1-43
              43  __CFRunLoopServiceMachPort + 372 (CoreFoundation + 543772) [0x1a3dbec1c] 1-43
                43  mach_msg_trap + 8 (libsystem_kernel.dylib + 6352) [0x1a3cb48d0] 1-43
                 *43  ??? (kernel.release.t6000 + 224476) [0xfffffe00078d2cdc] 1-43

  Binary Images:
           0x102da8000 -                ???  com.agilebits.onepassword7 7.9.4 (70904001) <39395861-BE51-346E-93B4-BF111CBCD526>              /Applications/1Password 7.app/Contents/MacOS/1Password 7
           0x103490000 -        0x1034effff  com.microsoft.appcenter.crashes 1 (1.0)     <E2FEECE9-84AC-3D5E-957F-A9510CFB84ED>              /Applications/1Password 7.app/Contents/Frameworks/AppCenterCrashes.framework/Versions/A/AppCenterCrashes
           0x103544000 -        0x1035bffff  dyld (955)                                  <FBB89662-E6F2-3434-B542-F75185AC5E74>              /usr/lib/dyld
           0x1a3cb3000 -        0x1a3ceafff  libsystem_kernel.dylib (8020.101.4)         <1D7B3B8E-75A1-34EA-AA52-9F7C23155C55>              /usr/lib/system/libsystem_kernel.dylib
           0x1a3ceb000 -        0x1a3cf7fff  libsystem_pthread.dylib (486.100.11)        <CEE8BC77-6923-34D9-89A3-6F8F7279605E>              /usr/lib/system/libsystem_pthread.dylib
           0x1a3d3a000 -        0x1a427ffff  com.apple.CoreFoundation 6.9 (1858.112)     <8E1809FF-F5B1-3D9E-BE15-F2E7B1D8CE51>              /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
           0x1a4c35000 -        0x1a5025fff  com.apple.Foundation 6.9 (1858.112)         <589677DD-19B7-3F93-B2B6-EED7117729CA>              /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation
           0x1a68da000 -        0x1a7791fff  com.apple.AppKit 6.9 (2113.40.126)          <465B9570-6646-31EA-A082-AB0DFA538966>              /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
           0x1a893b000 -        0x1a8deefff  com.apple.CFNetwork 1331.0.7 (1331.0.7)     <CFDB5CCD-A5E9-3E08-AD4F-087F420DB6A0>              /System/Library/Frameworks/CFNetwork.framework/Versions/A/CFNetwork
           0x1ac9a9000 -        0x1accdcfff  com.apple.HIToolbox 2.1.1 (1113.9)          <4A1D1F30-C79D-3502-B51F-9792ADBD8E06>              /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox
   *0xfffffe000789c000 - 0xfffffe000812bfff  kernel.release.t6000 (8020.101.4)           <178AA913-FCA9-33FD-A81C-DF08315F458D>__TEXT_EXEC   /System/Library/Kernels/kernel.release.t6000

............................

nohal, 2022-05-07:
Are you able to reproduce this in a reliable way? FWIW I have never seen it freeze after responding to that macOS crap, regardless of the response.

jeronimo, 2022-05-07:
No, can't do that in reliable way, but maybe the issue is not with the permission as it was running fine but after a while 
[image]
Had a 7 day crossing and it got slow 3 or 4 times so I just restarted it, but after reaching Galapagos it started to hang, started to guess that maybe AIS targets are overloading it, but I am sure there are way more places in the world with way more traffic, so I am confused what might be the cause. Continuing to monitor and trying to figure out reliable way to reproduce it.

nohal, 2022-05-07:
How does your connection setup look? It is pretty well possible to couase this if there is a loop causing data to return back to the input. As far as normal AIS processing is concerned, I was testing with some~20k targets from AisHub some time ago and the impact on overall performance was sure visible, but not fatal, so what is there in VHF range in Galapagos should probably be fine.
Unless we have some really big issue with running on Apple Silicon, but that I unfortunatelly can't check right now as I only have access to an Intel Mac.

jeronimo, 2022-05-07:
SignalK is showing 144 targets in 130nm range. Have disabled that connection for now and will enable it again to see if it is the cause.

[image]

jeronimo, 2022-05-09:
By the memory usage I would guess that there is a memory leak on calculating CPA target warnings on "Warn if CPA is less than..." but I can't reproduce it straight away, once it ran the whole night without issues, and sometimes it is not memory, but CPU usage. 
And once it got "Not Responding" even when CPA warning was disabled so it got even more confusing and I started to think that the issue is with setting Priority 1 on SignalK connection.
[image]
Sorry for being so vague. You guys are doing amazing job.

jeronimo, 2022-05-09:
Failed again on CPU without CPA warnings enabled. It might be there are 2 separate issues...

[image]

```
Date/Time:        2022-05-08 20:20:02.465 -0600
End time:         2022-05-08 21:03:31.553 -0600
OS Version:       macOS 12.3 (Build 21E230)
Architecture:     arm64e
Report Version:   35
Incident Identifier: 16849E10-6210-4FBE-8466-E2B0E8D3CE5E
Share With Devs:  Yes

Data Source:      Stackshots
Shared Cache:     00EDBAF8-9281-3F26-B94F-E6A315E3A659 slid base address 0x1c2d58000, slide 0x42d58000
Shared Cache:     CB70EAD7-3266-30B7-BECD-BD97E7998A2A slid base address 0x1a14dc000, slide 0x214dc000
Shared Cache:     71741BEE-0ABC-3E06-B329-770CAC959B0C slid base address 0x7ff808bac000, slide 0x8bac000

Command:          OpenCPN
Path:             /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:       org.opencpn
Version:          5.6.2 ()
Architecture:     x86_64 (translated)
Parent:           launchd [1]
PID:              23574
Time Since Fork:  8992s

Event:            hang
Duration:         2609.09s
Duration Sampled: 4.30s (process was unresponsive for 2605 seconds before sampling)
Steps:            43 (100ms sampling interval)

Hardware model:   MacBookPro18,3
Active cpus:      8
HW page size:     16384
VM page size:     16384

Time Awake Since Boot: 124623s
Time Since Wake:  119254s

Fan speed:        0 rpm
Total CPU Time:   7.982s (20.0G cycles, 24.5G instructions, 0.82c/i)
Advisory levels:  Battery -> 3, User -> 2, ThermalPressure -> 0, Combined -> 2
Free disk space:  26.12 GB/460.43 GB, low space threshold 3072 MB

--------------------------------------------------
Timeline format: stacks are sorted chronologically
Use -i and -heavy to re-report with count sorting
--------------------------------------------------

Heaviest stack for the main thread of the target process:
  43  start + 462 (dyld + 21790) [0x201bb151e]
  43  main + 20 (OpenCPN + 868500) [0x100ba3094]
  43  wxEntry(int&, wchar_t**) + 56 (libwx_baseu-3.1.5.0.0.dylib + 412984) [0x10b9a5d38]
  43  wxApp::OnRun() + 26 (libwx_osx_cocoau_core-3.1.5.0.0.dylib + 349162) [0x10be423ea]
  43  wxAppConsoleBase::MainLoop() + 116 (libwx_baseu-3.1.5.0.0.dylib + 41572) [0x10b94b264]
  43  wxEventLoopBase::Run() + 85 (libwx_baseu-3.1.5.0.0.dylib + 206325) [0x10b9735f5]
  43  wxCFEventLoop::DoRun() + 29 (libwx_baseu-3.1.5.0.0.dylib + 880525) [0x10ba17f8d]
  43  wxGUIEventLoop::OSXDoRun() + 186 (libwx_osx_cocoau_core-3.1.5.0.0.dylib + 769386) [0x10bea8d6a]
  43  -[NSApplication run] + 586 (AppKit + 193337) [0x7ff80b8e0339]
  43  -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394 (AppKit + 248970) [0x7ff80b8edc8a]
  43  _DPSNextEvent + 927 (AppKit + 255437) [0x7ff80b8ef5cd]
  43  _BlockUntilNextEventMatchingListInModeWithFilter + 70 (HIToolbox + 194533) [0x7ff8120877e5]
  43  ReceiveNextEventCommon + 594 (HIToolbox + 195146) [0x7ff812087a4a]
  43  RunCurrentEventLoopInMode + 292 (HIToolbox + 195814) [0x7ff812087ce6]
  43  CFRunLoopRunSpecific + 562 (CoreFoundation + 514180) [0x7ff808f91884]
  43  __CFRunLoopRun + 2014 (CoreFoundation + 517922) [0x7ff808f92722]
  43  __CFRunLoopDoTimers + 307 (CoreFoundation + 622283) [0x7ff808fabecb]
  43  __CFRunLoopDoTimer + 923 (CoreFoundation + 623463) [0x7ff808fac367]
  43  __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20 (CoreFoundation + 624758) [0x7ff808fac876]
[... 104 lines cut ...]

  Thread 0xaa308    1 sample (3)    priority 31 (base 31)    cpu time <0.001s (938.8K cycles, 728.4K instructions, 1.29c/i)
  <Thread in another stack for 2 samples, thread QoS default (requested default), process unclamped, process received importance donation from WindowServer [373], IO tier 0>
  1  <truncated backtrace> 3
    1  ??? (runtime + 141332) [0x7ff7ffedb814] 3
     *1  ??? (kernel.release.t6000 + 704160) [0xfffffe0007947ea0] (suspended) 3

  Thread 0xaa6a7    1 sample (3)    priority 31 (base 31)    cpu time <0.001s (892.3K cycles, 643.3K instructions, 1.39c/i)
  <Thread in another stack for 2 samples, thread QoS default (requested default), process unclamped, process received importance donation from WindowServer [373], IO tier 0>
  1  <truncated backtrace> 3
    1  ??? (runtime + 141332) [0x7ff7ffedb814] 3
     *1  ??? (kernel.release.t6000 + 704160) [0xfffffe0007947ea0] (suspended) 3

  Binary Images:
           0x100acf000 -        0x1014b3fff  org.opencpn 5.6.2 ()                      <D8EA116A-F40D-3C3D-9031-C400DCBA0686>              /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
           0x10b941000 -        0x10ba9cfff  libwx_baseu-3.1.5.0.0.dylib (0)           <CFE63B63-2551-30EC-82BA-F8C41427E117>              /Applications/OpenCPN.app/Contents/Frameworks/libwx_baseu-3.1.5.0.0.dylib
           0x10bded000 -        0x10c2a4fff  libwx_osx_cocoau_core-3.1.5.0.0.dylib (0) <E12A2E94-51BE-3B46-A312-B25E10EA7FC2>              /Applications/OpenCPN.app/Contents/Frameworks/libwx_osx_cocoau_core-3.1.5.0.0.dylib
           0x11b647000 -        0x11b75afff  libdashboard_tactics_pi.dylib (0)         <FC38137C-0CD0-323A-8B68-97674F3AABEB>              /Users/USER/Library/Application Support/OpenCPN/*/libdashboard_tactics_pi.dylib
           0x201bac000 -        0x201c2ffff  dyld (955)                                <DD9E80DE-FB3B-349B-96A4-46874AD34D11>              /usr/lib/dyld
        0x7ff7ffeb9000 -     0x7ff7ffef8fff  runtime (254.16)                          <24497BE3-A1B4-360A-B90E-CDE9CE78A93B>              /usr/libexec/rosetta/runtime
[... cut ...]
        0x7ff80b8b1000 -     0x7ff80c73efff  com.apple.AppKit 6.9 (2113.40.126)        <36EC7E1B-AB12-377B-B3EB-FFC5BAABEE9F>              /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
        0x7ff812058000 -     0x7ff81234bfff  com.apple.HIToolbox 2.1.1 (1113.9)        <7761967D-A138-33B6-9E3F-5C5420A2EEFF>              /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox
   *0xfffffe000789c000 - 0xfffffe000812bfff  kernel.release.t6000 (8020.101.4)         <178AA913-FCA9-33FD-A81C-DF08315F458D>__TEXT_EXEC   /System/Library/Kernels/kernel.release.t6000

nohal, 2022-05-09:
Try to completely disable the DashboardTactics plugin (and better also all the other 3rd party plugins you might be using).

jeronimo, 2022-05-10:
Update: Disabled all plugins except Dashboard and Grib and it stopped working. Ran whole night without issues without them and as soon as enabled both it stopped working again. Now running just with Grib plugin enabled and will see how it goes.

jeronimo, 2022-05-10:
The only enabled plugin was Grib. But because it was not stopping working before with plugins enabled I would guess the issue is somewhere else...

```
Date/Time:        2022-05-10 15:27:38.760 -0600
End time:         2022-05-10 15:30:22.286 -0600
OS Version:       macOS 12.3 (Build 21E230)
Architecture:     arm64e
Report Version:   35
Incident Identifier: B82D955E-2172-46A3-80B5-BF902CAE70DC
Share With Devs:  Yes

Data Source:      Stackshots
Shared Cache:     00EDBAF8-9281-3F26-B94F-E6A315E3A659 slid base address 0x1c2d58000, slide 0x42d58000
Shared Cache:     CB70EAD7-3266-30B7-BECD-BD97E7998A2A slid base address 0x1a14dc000, slide 0x214dc000
Shared Cache:     71741BEE-0ABC-3E06-B329-770CAC959B0C slid base address 0x7ff808bac000, slide 0x8bac000

Command:          OpenCPN
Path:             /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
Identifier:       org.opencpn
Version:          5.6.2 ()
Architecture:     x86_64 (translated)
Parent:           launchd [1]
PID:              40620
Time Since Fork:  29778s

Event:            hang
Duration:         163.53s
Duration Sampled: 4.40s (process was unresponsive for 159 seconds before sampling)
Steps:            44 (100ms sampling interval)

Hardware model:   MacBookPro18,3
Active cpus:      8
HW page size:     16384
VM page size:     16384

Time Awake Since Boot: 235154s
Time Since Wake:  71098s

Fan speed:        0 rpm
Total CPU Time:   8.152s (19.6G cycles, 78.1G instructions, 0.25c/i)
Advisory levels:  Battery -> 3, User -> 2, ThermalPressure -> 0, Combined -> 2
Free disk space:  25.62 GB/460.43 GB, low space threshold 3072 MB

--------------------------------------------------
Timeline format: stacks are sorted chronologically
Use -i and -heavy to re-report with count sorting
--------------------------------------------------

Heaviest stack for the main thread of the target process:
  43  start + 462 (dyld + 21790) [0x201c3551e]
  43  main + 20 (OpenCPN + 868500) [0x100c66094]
  43  wxEntry(int&, wchar_t**) + 56 (libwx_baseu-3.1.5.0.0.dylib + 412984) [0x10ba68d38]
  43  wxApp::OnRun() + 26 (libwx_osx_cocoau_core-3.1.5.0.0.dylib + 349162) [0x10bf053ea]
  43  wxAppConsoleBase::MainLoop() + 116 (libwx_baseu-3.1.5.0.0.dylib + 41572) [0x10ba0e264]
  43  wxEventLoopBase::Run() + 85 (libwx_baseu-3.1.5.0.0.dylib + 206325) [0x10ba365f5]
  43  wxCFEventLoop::DoRun() + 29 (libwx_baseu-3.1.5.0.0.dylib + 880525) [0x10badaf8d]
  43  wxGUIEventLoop::OSXDoRun() + 186 (libwx_osx_cocoau_core-3.1.5.0.0.dylib + 769386) [0x10bf6bd6a]
  43  -[NSApplication run] + 586 (AppKit + 193337) [0x7ff80b8e0339]
  43  -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394 (AppKit + 248970) [0x7ff80b8edc8a]
  43  _DPSNextEvent + 927 (AppKit + 255437) [0x7ff80b8ef5cd]
  43  _BlockUntilNextEventMatchingListInModeWithFilter + 70 (HIToolbox + 194533) [0x7ff8120877e5]
  43  ReceiveNextEventCommon + 594 (HIToolbox + 195146) [0x7ff812087a4a]
  43  RunCurrentEventLoopInMode + 292 (HIToolbox + 195814) [0x7ff812087ce6]
  43  CFRunLoopRunSpecific + 562 (CoreFoundation + 514180) [0x7ff808f91884]
  43  __CFRunLoopRun + 840 (CoreFoundation + 516748) [0x7ff808f9228c]
  43  __CFRunLoopDoObservers + 543 (CoreFoundation + 519690) [0x7ff808f92e0a]
  43  __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23 (CoreFoundation + 520053) [0x7ff808f92f75]
  43  ___NSRunLoopObserverCreateWithHandler_block_invoke + 41 (AppKit + 9377416) [0x7ff80c1a2688]
  43  __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285 (AppKit + 1746929) [0x7ff80ba5b7f1]
  43  CA::Transaction::commit() + 777 (QuartzCore + 8753) [0x7ff810491231]
[... 167 lines cut ...]
       *43  ??? (kernel.release.t6000 + 5996808) [0xfffffe0007e54108] 1-43

  Thread 0x10ac88    43 samples (1-43)    priority 31 (base 31)    cpu time 0.011s (33.3M cycles, 10.1M instructions, 3.31c/i)
  <process frontmost, thread QoS default (requested default), process unclamped, process received importance donation from WindowServer [373], IO tier 0>
  43  thread_start + 15 (libsystem_pthread.dylib + 8043) [0x7ff808ec8f6b] 1-43
    43  _pthread_start + 125 (libsystem_pthread.dylib + 25825) [0x7ff808ecd4e1] 1-43
      43  wxThreadInternal::PthreadStart(wxThread*) + 990 (libwx_baseu-3.1.5.0.0.dylib + 934542) [0x10bae828e] 1-43
        43  wxThread::CallEntry() + 12 (libwx_baseu-3.1.5.0.0.dylib + 935052) [0x10bae848c] 1-43
          43  WebSocketThread::Entry() + 722 (OpenCPN + 4578162) [0x100fefb72] 1-43
            43  ??? [0x7ff8994a2940] 1-43
             *43  ??? (kernel.release.t6000 + 5996808) [0xfffffe0007e54108] 1-43

  Binary Images:
           0x100b92000 -        0x101576fff  org.opencpn 5.6.2 ()                      <D8EA116A-F40D-3C3D-9031-C400DCBA0686>              /Applications/OpenCPN.app/Contents/MacOS/OpenCPN
           0x10ba04000 -        0x10bb5ffff  libwx_baseu-3.1.5.0.0.dylib (0)           <CFE63B63-2551-30EC-82BA-F8C41427E117>              /Applications/OpenCPN.app/Contents/Frameworks/libwx_baseu-3.1.5.0.0.dylib
           0x10beb0000 -        0x10c367fff  libwx_osx_cocoau_core-3.1.5.0.0.dylib (0) <E12A2E94-51BE-3B46-A312-B25E10EA7FC2>              /Applications/OpenCPN.app/Contents/Frameworks/libwx_osx_cocoau_core-3.1.5.0.0.dylib
           0x201c30000 -        0x201cb3fff  dyld (955)                                <DD9E80DE-FB3B-349B-96A4-46874AD34D11>              /usr/lib/dyld
[... cut ...]
        0x7ff812058000 -     0x7ff81234bfff  com.apple.HIToolbox 2.1.1 (1113.9)        <7761967D-A138-33B6-9E3F-5C5420A2EEFF>              /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox
        0x7ffa21573000 -     0x7ffa216fcfff  GLEngine (19.4.4)                         <A4643D6B-1680-3258-A641-B26B57E5DB00>              /System/Library/Frameworks/OpenGL.framework/Versions/A/Resources/GLEngine.bundle/GLEngine
   *0xfffffe000789c000 - 0xfffffe000812bfff  kernel.release.t6000 (8020.101.4)         <178AA913-FCA9-33FD-A81C-DF08315F458D>__TEXT_EXEC   /System/Library/Kernels/kernel.release.t6000

nohal, 2022-05-10:
We have already seen this place in some of your previous backtraces. Drawing the heading predictor line, trivial OpenGL call. But at east it is a place that is internally really different on M1.

jeronimo, 2022-05-11:
Just another notice that it crashes way quicker when there is only GPS data coming (location, sog and cog) and it stays for way longer when all NMEA instruments are on.

Ok, will submit as many reports as possible to Apple and hopefully they will have a look at it.

nohal, 2022-05-11:
They will not, but you may try :)

jeronimo, 2022-05-12:
Ok, can confirm again that it crashes when there is only GPS data coming - location, sog and cog (in anchorage) and it does not when heading from compass and other data is present.  Maybe it is possible to replicate it with these parameters and it is not just Apple's fault.

nohal, 2022-05-12:
I have GPS data coming constantly, no freezes similar to yours in the last several years, so I highly doubt it is this easy.

jeronimo, 2022-07-08:
I am wandering if there is a way to resolve it quicker as it is becoming an annoying issue (I know, it is just a backup navigation option) with some tracking data lost.

transmitterdan, 2022-07-09:
Is it possible for you to create a voyage data file?  Basically, you have to copy the incoming NMEA stream to a file.  Then you can play it back with the VDR plugin.  Or is this only happening with SignalK?

jeronimo, 2022-07-29:
Yes, but It is just a workaround, right? Every time I leave it with watchdog it crashes without any alarms raised.

bdbcat, 2025-04-26:
Closing, please re-open with more current build of OCPN.

closed 2025-04-26 by bdbcat