OpenCPN/OpenCPN #2681: Significant delay in startup when SignalK data connection is used

nohal, 2022-05-09:
There is a significant delay between the start of the application and the UI becoming usable if there is a SignalK connection configured.

bdbcat, 2022-05-10:
Have you disabled sK Discovery?  This is pretty slow....

Hakansv, 2022-05-10:
I'm not arguing but for info: 
I'm not experience the same. It starts instantly both on O start up and when adding the connection.
My SK is on a RPi4 Buster (Openplotter installation).
Used on RPi, Win10 and Android.

nohal, 2022-05-10:
On macOS it is extremely visible, time to usable toolbar is almost 15s with SignalK connections enabled vs. less than 1s without. And I see similar results also on Linux and Windows.

nohal, 2022-05-10:
And yes, it gets much better with the discovery disabled, but what do we need the discovery for at all on already configured connections? Also, we already visibly start crunshing data and still take several seconds until the Toolbar is displayed and the application actually usable.

Hakansv, 2022-05-10:
I use the discovery only if don't remember the IP address. When found I switch it off. (Off would be default?)

Pavel if you want to compare to trace:
Two tests using Win10 to the same SK server located on a RPi4 over a IP (mobile)network.
(If relevant or not: Data from the SK server: 30 deltas/sec, 49 signalk paths.)

From start with SK connection to Last call ~= 9 seconds
From start with TCP connection to Last call ~= 7 seconds

```
09:56:00.279    INFO logger.cpp:130 ------- OpenCPN version 5.6.2.10 restarted at 2022-05-10 -------
09:56:03.196 MESSAGE datastream.cpp:138 makeDataStream Signal K:signalk.stupan.se:3000
09:56:03.305 MESSAGE SignalKDataStream.cpp:610 Opening Signal K WebSocket client: Signal K:signalk.stupan.se:3000
09:56:03.444 MESSAGE chart1.cpp:6718 OnInitTimer...2
09:56:04.011 MESSAGE SignalKEventHandler.cpp:65 Connected to Signal K server version: 1.42.0
09:56:09.466 MESSAGE chart1.cpp:6992 OnInitTimer...Last Call

09:56:56.352    INFO logger.cpp:130 ------- OpenCPN version 5.6.2.10 restarted at 2022-05-10 -------
09:56:58.625 MESSAGE datastream.cpp:138 makeDataStream TCP:signalk.stupan.se:10112
09:56:58.780 MESSAGE chart1.cpp:6718 OnInitTimer...2
09:56:59.111 MESSAGE NetworkDataStream.cpp:402 TCP NetworkDataStream connection established: TCP:signalk.stupan.se:10112
09:57:03.963 MESSAGE chart1.cpp:6992 OnInitTimer...Last Call
```

closed 2023-02-02 by nohal