OpenCPN/OpenCPN #2693: OpenCPN not able to install / start in Raspberry Pi 4

Corsair-63, 2022-05-11:
**Describe the bug**
RasPi 4 / 8 Gb with x64 Raspberry Pi OS x64.
not connected via SSH, local to monitor.
followed instruction in OpenCPN to install it and followed.
not able to start the program, I had uninstalled it so I have not more info about error but some of them where that one packet was not present and installation won be able to do it, I have been able to install all features but not the program itself that is the one that it seems available but not able to install.

**To Reproduce**
followed web instructions:
[(https://openplotter.readthedocs.io/en/latest/getting_started/installing.html)]
sudo apt update
sudo apt install python3-wxgtk4.0 python3-ujson python3-pyudev whois vlc

Now you have to install the main app openplotter-setting from the .deb file you can download from the Advanced section in downloading chapter.

After downloading the .deb file, you can install it by double click or typing this in a terminal:

downloaded: openplotter-settings_2.9.1-stable_all.deb
open a command window in downloads folder and
sudo dpkg -i openplotter-settings_2.9.1-stable.deb
gone to setup menu and I'm able to install all plugins but not the main program because of error above

**Expected behavior**
start the OpenCpn, but no way.
the icon is present but showing "Invalid desktop entry file '/usr/share/applications/opencpn.desktop'

**Screenshots**
Not available

**Desktop (please complete the following information if applicable):**
 - OS: Raspberry Pi OS x64 Debian
 - Version Bullseye

**Smartphone/Tablet (please complete the following information if applicable):**
N/A

**Additional context**
not configuring software via remote, all done locally with monitor attached to RasPi.
I've seen that has been a problem with x64 but using remote installation, absolutely unable to start the program.

nohal, 2022-05-11:
@sailoog can you help us a bit here? Is the above workflow supposed to work? And if it is what OpenCPN package does it actually try to install?

sailoog, 2022-05-12:
Yes, you are trying to install OpenPlotter v2.x in Bullseye and it is only for Buster. We are working on OpenPlotter v3.x for Bullseye 32bit and 64bit but we are delayed. I will add this info to docs today, sorry for that.

Your options are:

- Installing any of the current OpenPlotter v2.x images (32bit buster). If you have one of the latest Raspberry 4 with the latest firmware version installed, these images will not boot so you need to keep reading.
- You can download the latest Raspberry Pi OS (Legacy) and install OpenPlotter v2.x: https://www.raspberrypi.com/software/operating-systems/
- You can install OpenPlotter v3.x alpha (close to beta) but do not use it in production yet: https://openplotter.readthedocs.io/en/3.x.x/getting_started/downloading.html OpenPlotter v3.x roadmap: https://forum.openmarine.net/showthread.php?tid=3878