OpenCPN/OpenCPN #2694: aarch64: Debian package: No explanation to missing plugins.

leamas, 2022-05-13:
**Describe the bug**
The Debian package is available in backports and thus also possible to install for users on aarch64 systems like Rpi. However, there are no plugins available  since we on aarch64/arm64 only build plugins for Flatpak which causes user confusion.

**To Reproduce**
Steps to reproduce the behavior:
1. On a RPi ´, typically RPi-4, with native Ubuntu, Debian or 64-bit Raspbian: install using for example `apt install opencpn/bulllseye-backports`
2. Go to Options | Plugins
3. No plugins available.

**Expected behavior**
An explanation first time opencpn is started informing user that on this platform Flarpak is the preferred deployment.

**Notes**
This could and should be implemented in the Debian packaging without waiting for next release. The problem was first mentioned in #2628.

leamas, 2022-05-13:
@nohal: Please consider it assigned to me.

sailoog, 2022-05-16:
This might be fine as a temporary solution, but the real question here is what do we need to start building plugins for aarch64/arm64.

As I promised in https://github.com/OpenCPN/OpenCPN/issues/2628 now I have some numbers.

oeSENC, oeRNC (now unified under the plugin o-charts) and S-63 are some of the most used plugins. These plugins need to identify the host to be able to license charts so we can extract some stats from them and this is the result:

```
oeSENC/oeRNC/oesu systems
****************************************************************
Windows:         10457
Linux:           4880 (Raspberry: 3321)
Mac:             2176
Android:         2835
Dongle:          1536
----------------------------------------------------------------
Subtotal:        21884

S-63 systems
****************************************************************
Windows:         1482
Linux:           330 (Raspberry: 160)
Mac:             208
----------------------------------------------------------------
Subtotal:        2020

Total systems:   23904
```
As you can see the amount of Raspberries is really surprising and significant. Nowadays Raspberry models 3 and 4 are 64-bit and model 4 is the most sold. Ubuntu Jammy 64-bit for Raspberry is already stable, the official Raspberry OS 64-bit is also stable. One of these systems will be the logical first choice for Raspberry owners. In this scenario we can say that arm32 will be replaced by arm64 soon if it is not already.

Currently there are OpenCPN versions for aarch64/arm64 in Debian, Debian-backports and Ubuntu PPA repositories with no plugins for them. We get everyday support requests at o-charts asking for missing plugins for these systems. Really annoying and confusing. Flatpak version needs about 400MB of extra data and often there is not enough free space on SD cards. As we saw in https://github.com/OpenCPN/OpenCPN/issues/2628 there are problems with flatpak in headless systems, not because of flatpak but problems that are difficult to solve. Besides this we have multiple reports about a new issue related to displays this time not in headless systems that we are still trying to reproduce.

So, my question is, what is the amount needed to rethink this situation?

transmitterdan, 2022-05-16:
I do not understand resistance to native plugins for RPi most common 64-bit platforms.  I think flatpak is a great idea in theory.  But in practice it has corner cases that arise resulting in difficult compromises.  I agree that we cannot support native plugins for the entire Linux eco-system.  But RPi 4 is so popular we should support at least one or two of the most common OS variants with native plugins.

If we need a native RPi build system for plugins I am happy to supply them. I have a few RPi 4 machines and one or 2 could be devoted to building plugins.  I don't know if Github Actions supports self hosted RPi.  I have made an experimental case for Windows and it works ok.  I will look into whether/how an RPi can be a builder for Github Actions.

leamas, 2022-05-16:
We do build plugins for Raspbian. So far the only real problem is #2628, which is sort of a corner case. 

The fact is that if we started all over today, I think we'd using Flatpak all over the Linux universe. The fact is that we need to build Flatpak anyway, since not everyone us using Debian/Ubuntu (and some Ubuntu users does not use LTS releases, here Flatpak is also the solution). 

>  Flatpak version needs about 400MB of extra data and often there is not enough free space on SD cards

This argument actually makes we wonder what this is all about. OpenCPN users  typically install charts which are measured in GB, so these 400MB can't really be the issue, can they?

> There are problems with flatpak in headless systems

Yes. But they actually have simple work-arounds.

There are different perspectives here.  From the OpenCPN project (which I definitely not speak for) there is a need to keep down the number of plugins we build. The version/platform matrix for Linux is already daunting, and adding still more to this should not be taken lightly. Also note that Flatpak is becoming really popular these days, it's definitely not an odd solution. 

So in the end, this is a strategic decision: how many Linux plugin builds can we handle? To make it more interesting we have the process in  #2502 under way; to add new plugins to this mess under way has it's own  pecularities

leamas, 2022-05-16:
@transmitterdan : As far as I know no plugins are built using Github Actions. At least the two templates we use are mostly using CircleCI  which have native aarch64 builders. So that is no problem.

In the end I think this boils down to if someone make PRs against testplugin and shipdriver, the two templates used for building plugins. It then becomes up to the maintainers of these to decide if they want to add more builds to their matrix.

>  I think flatpak is a great idea in theory. But in practice it has corner cases that arise resulting in difficult compromises.

Besides 400MB of data and #2628, I'm not aware of other issues at this point. Are you? The statement _has corner cases that arise resulting in difficult compromises_ IMHO requires some citation.

sailoog, 2022-05-16:
Having flatpak as the only linux solution? I would buy you that. I am only trying to fix this mess. Look at the title that you yourself have put to this issue.

Do not get me wrong, the 400MB of extra data is just one more argument, the largest chart package at o-charts is 1.4GB and yes, 400MB is a lot for a SD when you are short of free space. 

The work-around that you propose for the headless problem is not simple at all for Linux newbies. Why do you think there are so many raspberry users? most of them are windows users who would never have dared to use linux.

If you do not want to add more plugins to the list, what about replacing the arm32 by arm64?

sailoog, 2022-05-16:
This is another issue reported by 2 users. this happens with and without monitor connected. I have not reported this yet because I am trying to reproduce and understand it:

```
pi@openplotter:~ $ flatpak run org.opencpn.OpenCPN
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.front.0:CARD=0'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4745:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5233:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../oss/pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib ../../usb_stream/pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_direct.c:2014:(snd1_pcm_direct_parse_open_conf) The field ipc_gid must be a valid group (create group audio)
14:13:37: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
14:13:37: Debug: Device: 1: sysdefault
14:13:37: Debug: Device: 2: pulse
14:13:37: Debug: Device: 3: speex
14:13:37: Debug: Device: 4: upmix
14:13:37: Debug: Device: 5: vdownmix
14:13:37: Debug: Device: 6: default
14:13:37: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
14:13:37: Debug: Device: 1: sysdefault
14:13:37: Debug: Device: 2: pulse
14:13:37: Debug: Device: 3: speex
14:13:37: Debug: Device: 4: upmix
14:13:37: Debug: Device: 5: vdownmix
14:13:37: Debug: Device: 6: default
14:13:37: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
14:13:37: Debug: Device: 1: sysdefault
14:13:37: Debug: Device: 2: pulse
14:13:37: Debug: Device: 3: speex
14:13:37: Debug: Device: 4: upmix
14:13:37: Debug: Device: 5: vdownmix
14:13:37: Debug: Device: 6: default
14:13:37: Debug: Device: 0: bcm2835 Headphones: - (hw:0,0)
14:13:37: Debug: Device: 1: sysdefault
14:13:37: Debug: Device: 2: pulse
14:13:37: Debug: Device: 3: speex
14:13:37: Debug: Device: 4: upmix
14:13:37: Debug: Device: 5: vdownmix
14:13:37: Debug: Device: 6: default
Invalid MIT-MAGIC-COOKIE-1 key14:13:37: Error: Unable to initialize GTK+, is DISPLAY set properly?
```

leamas, 2022-05-16:
First of all, this is not about me.  I'm no decision maker, just expressing my opinions. The actual decision makers are the plugin maintainers -- the work with (too?) many plugins is what they do, it's their decision

Anything is too much when you are short of space. Still, on my **development** machine, used to build both flatpak and native packages there is several GB free out of a 16 GB card. In short,  let us drop this.

Of course we could make it easy for end-users to use a different VNC server, it's a question of motivation and resources.  If this indeed is the issue we are talking about it could be solved without adding more plugins, for sure.

Replacing arm32 with arm64? Not a bad idea, actually. Even RPi-3 can run a 64-bit OS, for sure. Still, this would require users to install a new OS. This should happen one day, but is that day now? You know better than anyone else about this, I guess.

leamas, 2022-05-16:
@saillog: That issue probably needs separate bug. Don't hesitate to file it just because you cannot reproduce it, visible bug reports can attract other users with more input.

That said, googling for _flatpak Invalid MIT-MAGIC-COOKIE-1 DISPLAY_ brings several hits to me. It seems to be about misconfigured systems, and at least some of the threads provides possible solutions.

nohal, 2022-05-16:
The other question to consider here is for how long do we want to suffer with wx 3.0 once wx 3.2 is out (Here my personal answer clearly is "Not for a single second longer than absolutely necessary.") Flatpak is a clear answer to this.

On the other hand, as OpenCPN got included in Debian and Ubuntu and one of these is going to be the platform of choice for ARM64 in the foreseable future, we should probably support it. I can easily imagine dropping our support for Ubuntu before 20.04 (And even for that in the next release) to clear the table to make it possible.

And as usually, I will gladly hit the merge button if the RPi community sends a pull request doing what they need, it is kind of hard to believe there is not enough technical skill there to do it given the numbers...

sailoog, 2022-05-16:
> @saillog: That issue probably needs separate bug. Don't hesitate to file it just because you cannot reproduce it, visible bug reports can attract other users with more input.
> 

I have the habit of trying to reproduce and fix issues before reporting instead of acting as a bridge between developers and users. Give me some time.
> 
> And as usually, I will gladly hit the merge button if the RPi community sends a pull request doing what they need, it is kind of hard to believe there is not enough technical skill there to do it given the numbers...

Not sure what you mean here. There is not RPi community organized around OpenCPN. Most of these users are linux newbies. The closest thing to this community would be openplotter and that is what I am trying to do. We do not need a pull request to fix this, we need to take a decision. I remember having a talk with you offering myself to build plugins.

I would agree dropping support for PPA, Ubuntu, Debian and backports in favor of flatpack, anything will be better than this meaningless and complex situation. This is only my opinion wearing sailoog and openplotter hats, not o-charts, we would need to discuss internally for that.

sailoog, 2022-05-16:
> First of all, this is not about me. I'm no decision maker, just expressing my opinions. The actual decision makers are the plugin maintainers -- the work with (too?) many plugins is what they do, [...]

OpenPlotter users know that every new major Debian version they will have to burn a new SD, that is the deal.

leamas, 2022-05-16:
@sailoog :One aspect here is, as I said, that I'm no decision maker. However, this is not  entirely true: I do decide what to do with my own time. Looking at that specific aspect, I don't plan to do any work in this the direction you propose for the moment. This is just personal  priorities, no more and no less. 

One reason is the current state of the plugin universe. We have  #2502 ongoing which means that many plugins are duplicated with both Ubuntu and Debian plugins built. This is planned to settle around next major release, perhaps around the end of 2022.

As Pavel (@nohal) mentioned we also have to take care of the upcoming wxWidgets switch to 3.2. This will cause an ABI break and stir up a lot of dust. To add a new plugin in this situation is just not something I want to work with, it carries all sorts of risks. That is not to say I'm against if other people are motivated and does this work. 

Perhaps Pavel is right, we might eventually need to support this with native plugins since vanilla Debian and/or Ubuntu probably will be the biggest platform on aarch64 in the long run. But I honestly don't think we, at least not I, should try to deal with it until later. As things looks now, 22Q4 seems like a reasonable planning horizon.

This bug is about a missing message. It will be closed if/when  #2702 is merged.  Closing it does not necessarily mean the end of discussion. However, as I see it we should live with the current situation for now, waiting for other plugin issues to be settled first.

sailoog, 2022-05-16:
Understood.

I have found this o-charts systems stats really useful to have an idea about what is going on in the wild. Since the new o-charts plugin we can know at what point the number of arm32 is surpassed by arm64 and the flatpak usage for all linux architectures. Let me know if this can help.

leamas, 2022-05-16:
Any such info be helpful later when we need to re-assess this. But frankly, not before the other plugin issues I mentioned are settled.

Until then, we need to patch the current situation, handling bugs etc.

sailoog, 2022-05-16:
yes, I meant later, do not worry I am not going to insist on this any more.

rgleason, 2022-05-16:
`If we need a native RPi build system for plugins I am happy to supply them. I have a few RPi 4 machines and one or 2 could be devoted to building plugins. I don't know if Github Actions supports self hosted RPi. I have made an experimental case for Windows and it works ok. I will look into whether/how an RPi can be a builder for Github Actions.`

If there is time to work on this I might be very fruitful. Particularly the use of Actions and the experience and examples for our needs, as a possible alternative to circleci which has continued to be very friendly to OpenCPN.

bdbcat, 2022-05-17:
Weighing in with my management hat on...and surely oversimplifying things, as managers are prone to do.
1.  IIRC, we implemented flatpak for ARM64 partly because we could not at that time see clearly which ARM64 O/S solution would become stable, and thus become the _de facto_ choice for new installs on rpi4.  It looks as if that day is now upon us, judging by sailoog's interactions with new users.
2. I am sensitive to plugin dev and and maintainer workloads, even though I maintain two plugins only.  But it seems to me that adding another (sort-of debian compatible) platform to the CCI build matrix is not a bug deal.  Where we have the most problems is non-debian-like platforms (Mac, flatpak, Win). I further assume the generally used process of publishing to catalogs is also scripted, to some extent.  At least it is for mine.  So again, not a big deal.  Finally,  I cannot see any realistic reduction in workload by removing ARMHF builds, in favor of native ARM64 builds.
3. So, I ask:  "Which one of debian or Ubuntu or RaspiOS should we focus attention on for ARM64/rPI plugins? "  I would rely on opinions gleaned by interaction with the rPI community (sailoog?)  Let's try to decide on one O/S that users will recognize as sensible, and begin moving, however slowly, in that direction.

Finally:
We are developing a reputation as being quickly and logically responsive to real user input, as compared to some other FOSS and commercial projects.  This is one of our strengths.  Let us keep that in mind as we debate.

transmitterdan, 2022-05-17:
Dave,

Agree with all that you said.

I would humbly say that I predict RaspiOS (Debian) and Ubuntu will be the most popular 64-bit distros. Sorry that I cannot predict which will be the most popular among the two. Raspberry sees issues with deprecating RaspiOS and users seem to like Ubuntu.

If a CI workflow can be pulled together it is likely there will not be a lot of difference so supporting both should not be a hardship.

If we decide to only support Raspi or Ubuntu we should consider if we believe there is risk in Canonical dropping Ubuntu. Personally, I think the risk is low (approaching zero). Ubuntu is the most popular distro in the Linux world last time I checked. Canonical supports RPi in a strong way.

My "opinion" is that we consider supporting both Raspi64 and Ubuntu 64-bit with native plugins.

But I would immediately defer to @sailoog opinion on which OS will be most popular.

leamas, 2022-05-17:
It actually doesn't matter that much. When the #2502  process is completed we will build Debian plugins which are aimed to run on both Ubuntu and Raspbian and of course Debian as well.  

So, when the wx3.2 ABI break is done everything is setup for one plugin to rule them all. Or actually two, we will probably support both Debian stable and oldstable.

sailoog, 2022-05-17:
OK, thanks for the trust, I appreciate it.

The last Ubuntu for Raspberry 64-bit (Jammy) is the first one that is stable and after some deep test I can say that it works really good. However, I am sure the most used OS for Raspberry 64-bit machines will be the official Raspberry OS 64-bit and I am afraid openplotter is the culprit of this (in the marine context of course)

From the first moment we realized at o-charts that the Linux ratio was unusual (larger than Mac and Android and half of Windows), if we remove Raspberries from this ratio we get the usual numbers: https://en.wikipedia.org/wiki/Usage_share_of_operating_systems

There is a detail that I have omitted from o-charts statistics, most of the Raspberry installations are openplotters running under Raspberry OS and just few hundreds are non Raspberry OS based systems. I omitted this because often I have to juggle all my hats to be impartial.

There are 2 ways of using openplotter:

A - Burning a SD card with an image based on Raspberry OS with all openplotter apps pre-installed (including OpenCPN).
B - Installing all the openplotter apps manually from deb packages in any Debian derivative and any architecture.

Analyzing the openplotter user profile, we see that most of them are new Linux users, so option A is the first choice. To this amount must be added advanced but lazy Linux users. At openplotter we have an average of 40 unique downloads per day of Option A and an average of 150 unique downloads per day for about 3 months when a new version is released (data from a nextcloud instance). We have no data from option B, we will in next version.

For the upcoming openplotter 3, we will again rely on Raspberry OS for option A (32-bit and 64-bit) because this system offers us really useful tools for building and customizing these images that there are not available for other systems and because the huge community of Raspberry OS users makes easy maintaining and debugging openplotter.

These are the numbers and my impressions, hopefully leamas is right and we do not have to choose a platform but the only sure thing is that arm64 is here to stay.

leamas, 2022-05-17:
hm... we are moving to a point when the ABI break is done and the #2502 process is completed, tentatively around Q4. Is that when can drop the armhf support? 

It is actually a pain  build-wise, requiring slow hardware-emulation in the build host. If could offer 64-bit Debian + Flatpak support on aarch64 it looks reasonable to me.

leamas, 2022-05-18:
@matthijskooijman  This is BTW one of the reasons that Flatpak shines -- it decouples us as upstream from the Ubuntu downstream release cycle. That is, you can install OpenCPN using Flatpak and have full plugin support today.

bdbcat, 2022-05-19:
"I wanted to add that I'm running into this same issue on Ubuntu 22.04, which has recently been released: I see no plugins (other than the four builtin) in the plugins interface"
What $ARCH ?  On x86_64, you should see o-charts_pi now.  If AARCH64, then I agree, there are no third party plugins available yet.  And thus the discussion drift...

leamas, 2022-05-19:
Some hours ago (!) the 5.6.2 backport to jammy was uploaded. This can be installed if you remove the PPA and  uninstall the opencpn and opencpn-data packages. After that, apt update and re-install opencpn.

The backport is patched to accept focal plugins, so it has full support on x86_64/amd64.  On aarch64 Flatpak is the way, it has also full plugin support.

leamas, 2022-05-19:
@matthijskooijman 
> Maybe a new issue should be opened for this?

Perhaps, but it's complicated. Obviously, by the word _support_ we are here talking about plugin support. 

But plugin development is done in separate communities. When a new version of OpenCPN is released, it's up to these communities to create or not to create plugins for various platforms. This is dynamic, and not much could be said in general.

However,  here are two special cases:  the first is aarch64, the core issue in this bug. This is sort of handled in #2702.

The other is the completely empty plugin list on 22.04 which you experienced (we still don't know on which hardware).  First of all, this is already handled in the official backport for amd64. Secondly, this could perhaps be described as a problem in our processes -- after all, the release of Jammy was no surprise and the plugin community could perhaps have been  more prepared. 

Please feel free to open a new issue if you think this needs more discussion.

matthijskooijman, 2022-05-20:
> The other is the completely empty plugin list on 22.04 which you experienced (we still don't know on which hardware).

For completeness, this was on 22.04 x86_64. But again, this is just an example. Even with the backported fix, or more preparation for the 22.04 release, IIUC this same situation will occur for users of other distributions (e.g. slackware, Fedora, etc.), so that's the more generic issue I tried to address (and now suggested a possible fix for, see linked comment below).

> Please feel free to open a new issue if you think this needs more discussion.

From reading this discussion and the code that handles this, there were a number of other points that I wanted to share. I'm not sure if they would be appropriate for this issue, but then I found #2307 which seems to be about the more generic subject of deciding plugin compatibility, so I added my points in https://github.com/OpenCPN/OpenCPN/issues/2307#issuecomment-1132692898

That includes a suggestion on how to more generically handle feedback about unsupported plugins (point 4 in the linked comment).

gregw, 2022-05-30:
I've got the same issue.   
Is this now known/fixed and I should just wait for a release, or is there value in me building from source to test current status.

I've currently got openCPN 5.6.0 on ubuntu:
```
[1357] lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy

[1358] uname -a
Linux pellet 5.15.0-33-generic #34-Ubuntu SMP Wed May 18 13:34:26 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

```

leamas, 2022-05-31:
@gregw : No, this is not the same issue, the subject here is about aarch64 systems while you are on x86_64/amd64.

Anyway, on jammy you can just update opencpn to the latest release 5.6.2+dfsg-1~bpo22.04.1. This has full plugin support using focal plugins.

leamas, 2023-02-04:
The issue is still there:  Users get little feedback when plugins are missing. However, the discussion seems inconclusive.  Converting to a Discussion item until we perhaps can formulate a more targeted issue from here