OpenCPN/OpenCPN #2696: glTextureManager.cpp: Pointer used after delete

leamas, 2022-05-14:
**Describe the bug**
gcc12 emits a warning ` warning: pointer may be used after ‘void operator delete(void*, std::size_t)` for this snippet around line 890 in glTextureManager.cpp:
```
 delete ticket;

  if (g_raster_format != GL_COMPRESSED_RGB_FXT1_3DFX) {
    running_list.DeleteObject(ticket);
    StartTopJob();
  }
```

It seems to me that that the warning is bad, but I have no understanding of this code.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile using gcc12.1.1
2. Watch logs

**Expected behavior**
No warning

transmitterdan, 2022-05-16:
I also don't know if this is a real problem or not.  However, there is definitely a problem somewhere.  I recently caught up with the recent influx of PRs and now cannot run a clean install of OpenCPN without a crash.  This happens on Windows in both Debug and non-Debug builds.  On Debug builds there are more traps that are tripped in the STL code.  I suspect that the recent switch away from wx based containers to STL containers has allowed some very hard to find bugs to creep in.  I am investigating now and will report more.  If I find a true cause i will file a specific issue or PR.  But just wanted everyone to know that I am hunting some fairly serious bugs right now.

How to reproduce:

1. Uninstall O
2. Delete all existing chart files, SENC cache and the like.  Try to make the machine as "virgin" as possible.
3. Install O built from current master
4. Use chart downloader to install some charts (ENC, RNC & MBTILES0.
5. Move the chart to a location where some of these charts exist and let O build up the SENC cache.  50% of the time O crashes with a bad pointer exception.

leamas, 2022-05-17:
@transmitterdan : seems like a separate issue (?) Have you any way to make a stacktrace from the crash?

transmitterdan, 2022-05-17:
It could be a separate issue. The crash dump does not help in finding the root cause. The heap is corrupted long before the crash.

leamas, 2022-05-17:
Well, as a starter I guess we should fix the issue here, just to get rid of the uncertainty. Busy doing other things now, have any chance to look into it?

Besides that, the only way I can think of attacking it making a bisect. Which, given that it's not fully reproducible and messy to set up, is painful.

transmitterdan, 2022-05-17:
There is a heap-check framework in the MSW build.  Running that should quickly find the issue.

I will look into this over next couple days.

leamas, 2022-05-19:
Did a 20-minutes check of the code here.  There is clearly more about the lifespan of this object, I will definitely not be able to cover all details. However,  it seems to me that this could be sorted out by converting the ticket (which is a `JobTicket*`)  to a `unique_ptr`. I guess the compiler should catch most of  the problems after such a change.

Later, time permitting, making a note now.

transmitterdan, 2022-05-26:
@leamas/@bdbcat

I believe the crashing I have seen is due to this commit: 6771c95c1d2d2df53619e7677b331fb9b26b3c1e

If I build from master and do a virgin install of O with no pre-existing SENC cache, then O always crashes at some point whilst zooming in and out when ENC charts are being processed into SENC cache by multiple threads.  When I revert this commit the crashes do not happen.  I think there is some threading bug introduced by this commit.  I don't think it is necessarily related to the particular issue that @leamas had identified here but not 100% certain of that.  But I have tested on x86 64-bit, arm 64 bit and x86 windows 32 bit all built from master branch and all of them can be made to crash.  Reverting this commit fixes the crash.

bdbcat, 2022-05-27:
Dan...
I can well believe this.  Mutex management, and especially debugging thereof, is a non-trivial thing.
Let us revert this commit, and move on to more visible improvements to the code base and UX.

Thanks for your diligence on this.
Dave

leamas, 2022-05-27:
MOre notes on original issue (sorry for keeping my diary here):
```
bool JobTicket::DoJob(const wxRect &rect)
 pt->DoJob();
 OCPN_CompressionThreadEvent Nevent(wxEVT_OCPN_COMPRESSIONTHREAD, 0);
Nevent.SetTicket(pt);
void SetTicket(JobTicket* ticket) { m_ticket = ticket; }
```
i. e., a pointer is assigned a boolean. This code stinks.

PR #2985 (2023-05-03): glTextureManager: Don't delete pointer before last use (#2696)
This is a simple point  fix for #2696. I have had  ambitions to do something more about this code which generally certainly could be improved. However, no time, defaulting to the obvious, simple fix.

Closes: #2696
closed 2023-05-03 by bdbcat