OpenCPN/OpenCPN #2704: HARMONICS_NO_US.IDX, and HARMONICS_NO_US Tables Duplicates and Issues

jfwgourlay, 2022-05-20:
### **_There are 249 duplicate stations in HARMONICS_NO_US.IDX with the (2) after them.
### **_Most, but not all, are in British Columbia.
### **_See headings below for more info…_**
________________________________________
Search "(2)" (249 hits in 1 file of 1 searched)
  C:\Users\John\Desktop\TC\HARMONICS_NO_US.IDX (249 hits)
	Line  268: TNAP:Can:BC: -127.8167 51.6 -8:0 Addenbroke Island, British Columbia (2)
	Line  296: TNAP:Can:BC: -126.9333 50.5833 -8:0 Alert Bay, British Columbia (2)
	Line  299: TNAP:Can:BC: -129.5 55.4667 -8:0 Alice Arm, British Columbia (2)
	Line  326: TNAA:Can:Newf: -53.9833 47.3 -3:30 Argentia, Newfoundland (2)
	Line  331: TNAP:Can:BC: -132.3833 53.1167 -8:0 Armentieres Channel, British Columbia (2)
	Line  372: TNAP:Can:BC: -125.1333 48.8333 -8:0 Bamfield, British Columbia (2)
	Line  379: TNAP:Can:BC: -129.1167 53.0833 -8:0 Barnard Harbour, British Columbia (2)
	Line  405: TNAP:Can:BC: -123.6333 48.3333 -8:0 Becher Bay, British Columbia (2)
	Line  410: TNAP:Can:BC: -123.2333 48.7333 -8:0 Bedwell Harbour, British Columbia (2)
	Line  419: TNAP:Can:BC: -128.1333 52.1667 -8:0 Bella Bella, British Columbia (2)
	Line  421: TNAP:Can:BC: -126.8 52.3833 -8:0 Bella Coola, British Columbia (2)
	Line  435: TNAP:Can:BC: -125.1333 50.4 -8:0 Big Bay, British Columbia (2)
	Line  440: TNAP:Can:BC: -125.8667 50.4 -8:0 Billygoat Bay, British Columbia (2)
	Line  450: TNAP:Can:BC: -124.1833 49.7167 -8:0 Blind Bay, British Columbia (2)
	Line  452: TNAP:Can:BC: -125.5 50.4167 -8:0 Blind Channel, British Columbia (2)
	Line  454: TNAP:Can:BC: -129.7667 53.15 -8:0 Block Islands, British Columbia (2)
	Line  456: TNAP:Can:BC: -125.3833 50.1167 -8:0 Bloedel, British Columbia (2)
	Line  458: TNAP:Can:BC: -124.6167 49.8 -8:0 Blubber Bay (Powell River Approaches), British Columbia (2)
	Line  460: TAuA:NZ:: 168.35 -46.6 12:0 Bluff, New Zealand (2)
	Line  472: TNAP:Can:BC: -129.2833 52.7333 -8:0 Borrowman Bay, British Columbia (2)
	Line  484: TNAP:Can:BC: -123.4667 48.5833 -8:0 Brentwood Bay, British Columbia (2)
	Line  490: TAuA:Aus:: 153.1667 -27.3667 10:0 Brisbane Bar, Australia (2)
	Line  499: TNAP:Can:BC: -125.3667 50.1667 -8:0 Brown Bay, British Columbia (2)
	Line  503: TNAP:Can:BC: -130.85 54.6167 -8:0 Brundige Inlet, British Columbia (2)
	Line  518: TNAP:Can:BC: -122.8833 49.3667 -8:0 Buntzen Lake, British Columbia (2)
	Line  524: TNAA:Can:NovS: -63.8 45.3 -4:0 Burntcoat Head, Nova Scotia (2)
	Line  527: TNAP:Can:BC: -128.7 53.1667 -8:0 Butedale, British Columbia (2)
	Line  547: TNAP:Can:BC: -125.2333 50.0167 -8:0 Campbell River, British Columbia (2)
	Line  564: TNAA:Can:NovS: -64.35 45.2667 -4:0 Cape Blomidon, Nova Scotia (2)
	Line  598: TNAP:Can:BC: -131.0167 51.9333 -8:0 Cape Saint James, British Columbia (2)
	Line  600: TNAP:Can:BC: -128.4167 50.7833 -8:0 Cape Scott, British Columbia (2)
	Line  624: TNAP:Can:BC: -130.3667 54.2833 -8:0 Casey Cove, British Columbia (2)
	Line  630: TNAP:Can:BC: -126.7167 49.8667 -8:0 Ceepeecee, British Columbia (2)
	Line  640: TNAP:Can:BC: -125.4333 50.3333 -8:0 Chatham Point, British Columbia (2)
	Line  643: TNAP:Can:BC: -123.7 48.9167 -8:0 Chemanius, British Columbia (2)
	Line  662: TNAP:Can:BC: -123.35 48.4 -8:0 Clover Point, British Columbia (2)
	Line  665: TNAP:Can:BC: -127.5833 50.6 -8:0 Coal Harbour, British Columbia (2)
	Line  678: TNAP:Can:BC: -131.1667 52.35 -8:0 Copper Islands, British Columbia (2)
	Line  682: TNAP:Can:BC: -125.4833 50.4333 -8:0 Cordero Islands, British Columbia (2)
	Line  688: TNAP:Can:BC: -123.6167 48.7333 -8:0 Cowichan Bay, British Columbia (2)
	Line  693: TNAP:Can:BC: -122.8833 49.0333 -8:0 Crescent Beach, British Columbia (2)
	Line  697: TNAP:Can:BC: -123.6333 48.85 -8:0 Crofton, British Columbia (2)
	Line  719: TNAP:Can:BC: -130.1667 55.7667 -8:0 Davis River, British Columbia (2)
	Line  721: TNAA:Can:Nun: -58.3833 66.2667 0:0 Davis Strait, Nunavut (2)
	Line  727: TNAP:Can:BC: -122.95 49.3333 -8:0 Deep Cove, British Columbia (2)
	Line  738: TNAP:Can:BC: -124.8167 49.5333 -8:0 Denman Island, British Columbia (2)
	Line  751: TNAP:Can:BC: -123.5667 49.0167 -8:0 Dionisio Point, British Columbia (2)
	Line  763: TNAP:Can:BC: -127.55 51.4833 -8:0 Drainey Inlet, British Columbia (2)
	Line  771: TNAP:Can:BC: -125.3 50.0833 -8:0 Duncan Bay, British Columbia (2)
	Line  790: TNAA:Can:Qbc: -78.05 52.0167 -5:0 Eastmain, Quebec (2)
	Line  800: TNAP:Can:BC: -127.8333 51.25 -8:0 Egg Island, British Columbia (2)
	Line  802: TNAP:Can:BC: -123.9333 49.75 -8:0 Egmont, British Columbia (2)
	Line  820: TNAP:Mex:BCS: -116.6333 31.85 -8:0 Ensenada, Baja California Norte, Mexico (2)
	Line  835: TNAP:Can:BC: -123.4333 48.4333 -8:0 Esquimalt, British Columbia (2)
	Line  850: TNAP:Can:BC: -124.35 49.5 -8:0 False Bay, British Columbia (2)
	Line  852: TNAP:Can:BC: -123.1333 49.2667 -8:0 False Creek, British Columbia (2)
	Line  861: TNAP:Can:BC: -123.3 48.4667 -8:0 Finnerty Cove, British Columbia (2)
	Line  867: TNAA:Can:NovS: -64.0667 45.3833 -4:0 Five Islands, Nova Scotia (2)
	Line  874: TNAP:Can:BC: -125.1667 50.3 -8:0 Florence Cove, British Columbia (2)
	Line  891: TNAP:Can:BC: -124.8167 49.1 -8:0 Franklin River, British Columbia (2)
	Line  899: TNAP:Can:BC: -123.45 48.7667 -8:0 Fulford Harbour, British Columbia (2)
	Line  915: TNAP:Can:BC: -123.5 48.85 -8:0 Ganges Harbour, British Columbia (2)
	Line  927: TNAP:Can:BC: -123.2833 48.8667 -8:0 Georgina Point, British Columbia (2)
	Line  932: TNAP:Can:BC: -123.5 49.4 -8:0 Gibsons, British Columbia (2)
[... 269 lines cut ...]
TNAP:Mex:BCS: -110.3 24.15 -7:0 La Paz, Baja California Sur, Mexico (2)
TNAP:Mex:BCS: -110.345 24.1617 -7:0 La Paz, Baja California Sur, Mexico (3)

TNAP:Mex:BCS: -112.1167 24.7833 -7:0 San Carlos, Baja California Sur, Mexico
TNAP:Mex:BCS: -112.12 24.79 -7:0 San Carlos, Baja California Sur, Mexico (2)
TNAP:Mex:BCS: -112.12 24.79 -7:0 San Carlos, Baja Ca-132.96lifornia Sur, Mexico (3)

TNAP:Can:BC: -123.1167 49.2833 -8:0 Vancouver, British Columbia
TNAP:Can:BC: -123.1167 49.2833 -8:0 Vancouver, British Columbia (2)
TNAP:Can:BC: -123.1167 49.2833 -8:0 Vancouver, British Columbia (3)

TNAP:Can:BC: -123.3667 48.4167 -8:0 Victoria, British Columbia
TNAP:Can:BC: -123.3667 48.4167 -8:0 Victoria, British Columbia (2)
TNAP:Can:BC: -123.3667 48.4167 -8:0 Victoria, British Columbia (3)
 
________________________________________
**_### This is duplicated in HARMONICS_NO_US, but also shows too high a current on one of the stations. Based on CHS tables, the max should be 3 knots…_**
________________________________________

Sansum Narrows
Line 2268: CNAP:Can:BC: -123.55 48.7833 -8:0 Sansum Narrows, British Columbia Current
Line 3385: CNAP:Can:BC: -123.55 48.7833 -8:0 Sansum Narrows, British Columbia Current

Line 3393: cNAA:XX:XX: -123.5500  48.7823 -8:0 Sansum Narrows
^24   0.50 0 -37   0.50 0   1849 0 180 1 Active Pass, British Columbia Current

nohal, 2022-05-20:
Are you aware of any authoritative source of information that is usable to assess which is correct in case of difference in position or any other information allowing us to resolve this? All this is external data so without additional info we are probably pretty limited in what can be done.

jfwgourlay, 2022-05-23:
As a test, I made a backup copy of the files and  deleted all the duplicates. Resolved part of the problem. Would need to go through and try and determine which copy was authoritative to fix this a little more rigorously.

The source for reference and secondary tidal statis in Canada is https://charts.gc.ca/publications/tables-eng.html

One of the harmonics for Sansum Narrows seems to be OK. Just need to determine where the second one is coming from.

I can have a shot at reviewing the ones I have flagged for Canada if that would help. (Give back to the community...) Just need some direction on how to do that effectively without adding too much additional work for the team. Sorting the table geographically might assist in having multiple people work on this.

John